subsurface/subsurface #774: Planner GF high < 35% possible in UI but not accepted, crash for very low values

sfuchs79, 2017-11-04:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Today in the planner the limits for GF low and GF high are for both 1%-150%.
a) I'm not sure if this really makes sense
b) If I decrease GF high below 35% via the combo box arrow buttons this is not reflected in the calculation
c) If I enter very low GF high (1-5%) via keyboard I create a freeze plus crash

### Operating system:

Win 10

### Subsurface version:

Latest Master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

Proposal:
Let's limit GF high (and GF low) directly in the UI file to different values than today.
I have no real proposal yet.
My OSTC seems to allow:
GF low 10...100%
GF high 60...100%

If someone says: "Let's have wider limits for testing/whatever" - Thats ok for me.

Maybe:
GF low 10...150%
GF high 35...150%
???

### Additional information:

### Mentions:

atdotde, 2017-11-06:
Let's first agree that it does not really make sense to use such small values. So it probably would not hurt anybody to set a limit at some intermediate percentage.

That said, I can confirm that the output does not change when using the arrows to go below 25% (but it does once I move the focus away from that scroll box). This seems to be a UI problem that should be further investigated.

The crash could well mean "the deco time is infinite" as in "with such a low gradient factor, even if you saturate at the last stop depth you will not be clear to go to the surface". This we should prevent with an error message like "diverged deco time".  Currently we crash as wait_until goes into infinite recursion doubling the stop time in each step. That should get a limit for the stop time of maybe a few days with an error message.

dirkhh, 2017-11-06:
It seems pretty clear to me that Buehlman is completely untested (and frankly, meaningless) with such log GF values. I'd limit GF high to 50-110% at the most. And I'd bail once a stop exceeds 24h

RickMWalsh, 2017-11-07:
I agree the lower-bound for both GF-Low and GF-High should be increased to something more sensible.  I'd say something in the order of 10% for GF-Low and 40% for GF-High.  As for the upper bound, I think 150% is good.
Try planning an extreme dive in VPM-B, then try approximating it with Buhlmann gradient factors.  You can end up with what would typically be considered a very high GF-High value.  E.g. 60min at 100m on 10/70, with 21/35, 50% and 100% for deco.  With VPM-B +2, I get first stop at 78m, and total runtime of 365min.  The best Buhlmann fit to this profile I can get is with GF 20/148 (sorry @atdotde, the GF fit is doing something wrong - it suggests 81/52) first stop at 78m, total runtime of 364min.  Now, I'm not suggesting you dive this profile.  But some people would, and I think it is helpful to keep the large range in acceptable GF-high, if for no other purpose than to demonstrate that the VPM-B profile is incredibly aggressive.
Now looking at the other extreme, the NEDU deep stop study, which compared a deep stop and a no deep stop profile on air.  I tried matching the not deep stop profile some time ago with gradient factors, and from memory ended up with something like GF 120/70.  This profile was found to be better by the navy researchers, and I think it would be good to keep high GF-low values as acceptable.

sfuchs79, 2017-11-12:
I would say let's close this with minimum effort?!
I followed @RickMWalsh proposal because I tend to say that should be ok for everyone.
PR done...

dirkhh, 2017-11-12:
And I already commented on your PR, trying to make it slightly more effort for YOU and less effort for myself and Willem in the future... I think it's good practice to try to remember to add changes to the ReleaseNotes and where useful to the user manual... :-)

PR #805 (2017-11-13): Planner limit minimum GFlow to 10 and minimum GFhigh to 40

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Planner limit minimum GFlow to 10 and minimum GFhigh to 40

### Changes made:

### Related issues:

Closes #774 

### Additional information:

### Mentions:

@RickMWalsh @atdotde
closed 2017-11-13 by janmulder