subsurface/subsurface #775: Provide Windows installer from the CI environment

dirkhh, 2017-11-05:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

We don't automatically catch it when pull requests break the Windows build. And given the ongoing work in #769 it seems only logical to do the same for a Windows installer

### Additional information:

@sfuchs79 - you said you had some helpful links for how to get the MXE environment initialized?
@neolit123 - any help here would be welcome, of course

### Mentions:

neolit123, 2017-11-08:
@sfuchs79 
hi, i think you've mentioned in the other thread that we have some MXE issues, so this won't work right away. what seems to be the problem and how can i help?

dirkhh, 2017-11-08:
I think the MXE build issues were fixed by https://github.com/Subsurface-divelog/subsurface/commit/76286c4ece36baff3aaabb8a4e1252997d8ad7c4 
I have googled a bit about using MXE in a Travis environment, but I don't think I have found that site with helpful links that @sfuchs79 had mentioned, so this is a ping for him to point us in the right direction... because yes, I want that :-)

sfuchs79, 2017-11-08:
Yeah, I'm here? ;-)
Yes @dirkhh is right: No real issues related to MXE build. All recent small things solved.
What I wanted to highlight is maybe finally even a little bit less that I expected at the beginning:
I found this MXE related info that there are packages available for Travis integration:
http://pkg.mxe.cc/
And there is this example project "battleship" - but this project for sure is very simple compared to Subsurface.

dirkhh, 2017-11-08:
cool - I'll look at that as a starting point... thanks @sfuchs79

neolit123, 2017-11-15:
closing as implemented.

closed 2017-11-15 by neolit123