subsurface/subsurface #776: Presentation error since 4.7.1

fkmuck1, 2017-11-05:
### Describe the issue:

- X Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Since Version 4.7.1 i have a huge presentation error.
1.  I am unable to see the menu bar.
2. All rows are offset by one line (i have to click one line below the field i want to activate)

### Operating system:
Windows 10 Pro, 64 bit, Build 1709
Dell XPS 15 9560 with 4K Monitor

### Subsurface version:
Current 4.7.2
Error since 4.7.1
Official Windows release
Before 4.7.1 i didnt had this problem

### Steps to reproduce:
Maybe to use a 4 K Monitor?

### Current behavior:

### Expected behavior:

### Additional information:
I tried to use compatibilty mode for some other Windows versions  with no result.
I tried different behavior settings with no result (scale 640x480, disable high DPI-scale behavior, fullscreen optimization, run as administratior).
Reinstalled the program already.
Maybe the program has problems with my 4 K Monitor?

English is not my native language. Maybe i can provide more detailed and better informations in german :)

### Mentions:

[image]

[image]

neolit123, 2017-11-05:
that's very strange...
have you tried changing the Windows 10 font scale like so: https://img.gadgethacks.com/img/24/71/63584672105505/0/change-font-size-windows-10.w1456.jpg)
(not sure if your comment about ` disable high DPI-scale behavior` means fonts scale.)

other than that i'm out of ideas. also, i don't see how this issue is only present in `4.7.2`

here is something else to try - NOTE!!! **do at your own risk, this can delete your Subsurface settings if you delete the registry key**
- exit Subsurface
- open the start menu and start `Regedit`
- navigate to HKEY_CURRENT_USER -> Software
- rename `Subsurface` with F2 to something like `Subsurface_`
- start Subsurface
- see if this fixes it

if it doesn't fix it rename `Subsurface_` back to `Subsurface`.

neolit123, 2017-11-05:
> Before 4.7.1 i didn't had this problem

oh, so it's before 4.7.1.

neolit123, 2017-11-05:
you can also try deleting/renaming only this reg. key:
`HKEY_CURRENT_USER -> Software -> Subsurface -> Subsurface -> MainWindow`

fkmuck1, 2017-11-06:
I uninstalled Subsurface again and deleted the registry key 
HKEY_CURRENT_USER -> Software -> Subsurface
But no result. Still missing the menu bar.

Also tryed to play with the scaling you mean but with no result.

The settings i meant is when you right click on subsurface.exe an then go to properties.
I played with all possible options but the menu bar doesnt come back.
[image]

I found an older install file of subsurface on my computer => 4.6.2 , and installed it.
In this version i can see the menu bar !!!
So there must be something new or different in 4.7.1 and 4.7.2 which cause this behavior.
Here you can see the difference.
[image]
[image]

Maybe you can find the reason?

tcanabrava, 2017-11-06:
There's no preferences to set the menubar on subsurface, in the code we
never remove it.
I'll poke here a bit to see what I can find.

ijdod, 2017-11-20:
I'm seeing the same issue since v4.7.1 and still with 4.7.4. Also on a 4K monitor. Windows scaling is at 200%, but none of the suggestions above fix the problem.

neolit123, 2017-11-20:
@ijdod are you sure this happened when you installed 4.7.1 for the first time?
did it work OK with 4.7.0?

ijdod, 2017-11-20:
I went from 4.6.4 (which worked fine) directly to 4.7.1, which had the problem. I have not tried 4.7.0.

neolit123, 2017-11-20:
@ijdod 
i will make an installer for you test in a minute, so that we can find the cause.

neolit123, 2017-11-20:
@ijdod 
i've started our build system to create a test installer here:
https://github.com/Subsurface-divelog/subsurface/pull/842

it contains a small change that might be a fix for the 4k issues.
it would take around 20 minutes for the build setup to finish.
when it's done i will send you another message for you to download it and test it.

thanks

neolit123, 2017-11-20:
@ijdod 

please, download the installer from here:
https://transfer.sh/LwfAn/subsurface-4.7.4-75-gc88cd7834443.exe

you might have to do it quickly before it's gone as it's probably temporary.

you can also backup your existing folder by renaming it to `Subsurface_` and installing this to `Subsurface` once done testing you can restore your original Subsurface installation.

let me know if this fixes the issue?

P.S. this log right here proves that the installer originated from the subsurface CI system and not from me with malicious intents:
https://api.travis-ci.org/jobs/304981817/log.txt?deansi=true

ijdod, 2017-11-20:
Same problem.

neolit123, 2017-11-20:
@ijdod 

thanks for testing.

sadly, i'm out of ideas at this point.
there are something like 1200+ changes (commits) between 4.6.4 and 4.7.4, so i don't know what could have caused that (yet, not many touch the main window).

i've tried reverting a more resent change, but to me this might be a Qt bug (Qt is the framework taht we are using). so a fix might be difficult to obtain any time soon.

ijdod, 2017-11-20:
Thanks for trying. Looks like 4.7.0 wasn't released through the main site, versions there jump from 4.6.4 straight to 4.7.1. This is also where the issues started.

ijdod, 2017-12-09:
Just as confirmation: 4.7.5 didn't fix the problem (didn't expect it to, but still :D)

neolit123, 2017-12-09:
maybe this would be fixed once we move to a new Qt version.
the bug seems to be outside of Subsurface.

vinsanity0, 2018-01-30:
Just a note. I'm also having the same problem on my Thinkpad T470 with Windows 10 and Subsurface 4.7.6. I tried the same things that are above.

Thanks for your hard work guys!! I'll use my MAC until QT fixes it. :-)

MarioGeckler, 2018-03-21:
I have the same problem on a Dell PS 15 9560 with 4K Display

I have tried the following combinarion of resolution and scaling factor with no sucess:
1920x1080/100%
1920x1080/200%
3840x2160/100%
3840x2160/250%

MarioGeckler, 2018-04-11:
[image]

I just tried different screens of Subsurface and discovered, everthing looks fine in the dive planning.

Maybe the issue is not in QT?

neolit123, 2018-04-16:
@MarioGeckler 

i still think it's a Qt issue. i don't think we are doing anything "outside of the box".
maybe there is a way to avoid the issue, but hard to find a way to do that without a developer reproducing it.

MarioGeckler, 2018-04-22:
@neolit123 

I fixed the issue for me.

I had the same issue in Blender, which doesnt use QT at all.

The following steps fixed the issue for me.

- Open Device Manager
- Right Click Display Adapters -> Intel HD Graphics
- Properties -> Driver tab -> Update Driver button
- Use the "List of drivers on my computer" option
- Set to Microsoft Basic Driver. Note, your graphics may now be quite wonky because they aren't using the GPU...
- Install the new driver from Intel.
- Reboot.
- Done!

It looks like the vendor driver for the intel GPU is causing the issue.

[Source](https://developer.blender.org/T51521)

ijdod, 2018-04-22:
The solution described by MarioGeckler solved the problem for me. For my HP laptop an Intel HD video driver update was available through HP.

neolit123, 2018-04-22:
good to hear that a solution was found!
closing this ticket.
thanks.

closed 2018-04-22 by neolit123