subsurface/subsurface #778: OSTC Plus configuration

dadefay, 2017-11-05:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I want to configure my OSTC Plus using Bluetooth, connection works fine and retrieving the computer setup works also. But when I try to send the modifications to the computer, it fails almost each time with the message "error: Failed!" Then it is no more possiblee to read the computer again, you must disconnect / reconnect.

A libdivecomputer log is attached.

There is no problem to donload divelog.

### Operating system:

Windows 10 Home 64-bit.

OSTC Plus.

### Subsurface version:

4.7.2.

Same with 4.7.1.

Official release(s).

N/A.

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

Done.

### Mentions:

glance-, 2017-11-06:
Which firmware are you running on your OSTC Plus?

I haven't spent too much energy on being compatible with older firmware.

That being said, after looking at the log file, it looks like the computer goes mad after sending the first gas:
```INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=77
INFO: Custom Write: size=5, data=1015000100
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=4D
ERROR: Unsupported command. [in /data/src/win/libdivecomputer/src/hw_ostc3.c:217 (hw_ostc3_transfer)]
INFO: Custom Write: size=1, data=60
INFO: Custom Read: size=1, data=4D
ERROR: Unsupported command. [in /data/src/win/libdivecomputer/src/hw_ostc3.c:217 (hw_ostc3_transfer)]
INFO: Custom Write: size=1, data=6A
INFO: Custom Read: size=1, data=4D
ERROR: Unsupported command. [in /data/src/win/libdivecomputer/src/hw_ostc3.c:217 (hw_ostc3_transfer)]
INFO: Custom Write: size=1, data=FF
INFO: Custom Read: size=1, data=4D
```
We send 0x77 , which means, here comes a setting for you, and the computer correctly answers 0x77 as ok, give it to me.
Then we send 0x1015000100 , where first byte, 0x10 means here comes gas1 settings. 0x15 means 21% O2, 0x00 means 0% he, 0x01 means type First, and 0x00 means switch depth 0.
Then the computer correctly answers 0x4D as in, OK, ready for the next command.
When we send 0x77 for the next setting, the computer doesn't answer 0x77 as it should, it answers 0x4D.

Either this is a bug in our bluetooth layer, or this is a bug in the firmware on the device.

Can you test the same thing using the serial emulation in Windows 10 instead? That will tell us if the bug is in the serial layer or in the firmware.

janmulder, 2017-11-13:
I believe that the hardware of a Plus is very similar to an OSTC 3+ (BT/BLE) version.  I checked the settings read/write on my OSTC3+, and that works just fine.

dadefay, 2017-11-13:
Hi Jan,

You're right, those two models are very similar. I've got an OSTC 3+ too 
(hwos 2.26) , and the same problem occurs as with the OSTC Plus.

Regards,
Didier

janmulder, 2017-11-13:
Ok, I'm running hwos 2.26 as well, so the only difference might be the language setting. I have the en+de image, and might have firmware version with French ... should not make any difference ... but it is strange that for me it works, and for you not. Ok, I'm running a Linux box, and you Windows, but the problem seems on a level that Linux/Windows is not relevant.

dadefay, 2017-11-14:
I tried with the en+de image, same problems as with the fr+it...

Notice the problem occurs even if I do no change, i.e. I retrieve the 
setup, then directly save it in the computer, it fails.

Finally I performed the following test (with the OSTC 3+ / fr+it) :
- Reset to factory setup.
- Connection to Subsurface.
- Retrieve the setup.
- Save the setup, with no change, ok.
- Change one parameter (allow alernate GF).
- Try to save the new setup, *it fails.*

I did it a second time (logfile attached) :
- Reset to factory setup.
- Connection to Subsurface.
- Retrieve the setup.
- Save the setup, with no change, ok.
- Change one parameter (allow alernate GF).
- Try to save the new setup, *ok !
*- Change another parameter (MOD displayed).*
* - Try to save the new setup, *it fails.*

Don't know what to think about this... A timing problem ? I will try 
with a different PC as soon as possible.

Regards,
Didier

**

Subsurface: v4.7.2, built with libdivecomputer v0.6.0-devel-Subsurface-branch (e177b28627a81815211abb742cbf97dc7e5330a9)
INFO: Open: name=00:80:25:4A:3B:74
INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
INFO: Timeout: value=3000
INFO: Sleep: value=300
INFO: Purge: direction=3
INFO: Custom Write: size=1, data=BB
INFO: Custom Read: size=1, data=BB
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=60
INFO: Custom Read: size=1, data=60
INFO: Custom Read: size=5, data=001A000000
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=60
INFO: Custom Read: size=1, data=60
INFO: Custom Read: size=5, data=001A000000
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=72
INFO: Custom Read: size=1, data=72
INFO: Custom Write: size=1, data=10
INFO: Custom Read: size=4, data=15000100
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=72
INFO: Custom Read: size=1, data=72
INFO: Custom Write: size=1, data=11
INFO: Custom Read: size=4, data=15000000
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=72
INFO: Custom Read: size=1, data=72
INFO: Custom Write: size=1, data=12
INFO: Custom Read: size=4, data=15000000
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=72
INFO: Custom Read: size=1, data=72
INFO: Custom Write: size=1, data=13
INFO: Custom Read: size=4, data=15000000
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=72
INFO: Custom Read: size=1, data=72
[... 691 lines cut ...]
INFO: Custom Write: size=2, data=451D
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=77
INFO: Custom Write: size=2, data=4665
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=77
INFO: Custom Write: size=2, data=3500
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=77
INFO: Custom Write: size=2, data=4200
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=77
INFO: Custom Write: size=5, data=1015000100
INFO: Custom Read: size=1, data=4D
INFO: Custom Write: size=1, data=77
INFO: Custom Read: size=1, data=4D
ERROR: Unsupported command. [in /data/src/win/libdivecomputer/src/hw_ostc3.c:217 (hw_ostc3_transfer)]
INFO: Custom Write: size=1, data=FF
INFO: Custom Read: size=1, data=4D
ERROR: Unsupported command. [in /data/src/win/libdivecomputer/src/hw_ostc3.c:217 (hw_ostc3_transfer)]
ERROR: Failed to send the command. [in /data/src/win/libdivecomputer/src/hw_ostc3.c:529 (hw_ostc3_device_close)]

janmulder, 2017-11-14:
Just tried 5 changes in one connect session ... all works. Even with an 18% battery that is currently in my OSTC. Very interested to hear your results on a different PC.

dadefay, 2017-11-14:
Hi Jan,

I've installed Subsurface (4.7.4, just downloaded) on a different PC 
(Windows 10 Pro) : no problem, everything works.

So I've upgraded Subsurface on the first PC and then, surprise, it works 
too !

Is there any Bloototh change from 4.4.2 to 4.7.4 ? Nothing on the web 
site. Anyway, it's ok for me now. Thanks for your assistance,

Regards,
Didier

dirkhh, 2017-11-15:
Yes, there are quite a few changes in the Bluetooth support since 4.4.2
That version is two and a half years old... that's why we offer the upgrade reminder when you first install Subsurface.

/D

dadefay, 2017-11-15:
Oups, I'm sorry I wrote 4.4.2 instead of *4.7.2* !

janmulder, 2017-11-15:
To answer the last question: there are no relevant changes between v4.7.2 and v4.7.4 that can explain this strange non-deterministic error (that I am unable to reproduce). 

Closing this for now.

closed 2017-11-15 by janmulder