subsurface/subsurface #779: AppImages don't work on ArchLinux with OpenSSL issue

dirkhh, 2017-11-06:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When starting an AppImage on ArchLinux one gets "Incompatible version of OpenSSL" and neither Facebook nor cloud storage work

### Operating system:

Arch Linux
This works with Ubuntu 17.04

### Subsurface version:

latest master

CI-build AppImage

### Steps to reproduce:

download AppImage from https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous to Arch Linux system
run
### Current behavior:

OpenSSL error
### Expected behavior:

working Facebook integration and working cloud storage

### Additional information:

### Mentions:

@probonopd - this is one issue where I could REALLY use your help

probonopd, 2017-11-06:
(From https://github.com/Subsurface-divelog/subsurface/issues/769#issuecomment-341967367)

The OpenSSL issue is giving me some headache. Not sure yet what causes it, possibly related to https://github.com/probonopd/linuxdeployqt/issues/110. What happens if you unpack the AppImage and manually delete the SSL libs, then run AppRun like this?

```
chmod +x Downloads/Subsurface-*-x86_64.AppImage 
./Downloads/Subsurface-*-x86_64.AppImage --appimage-extract
rm squashfs-root/usr/lib/libssl.so.1.0.0 squashfs-root/usr/lib/libcrypto.so.1.0.0 
./squashfs-root/AppRun 
```

On a related note, I'd be _very_ interested in your opinion @dirkhh about our list of [Desktop Linux Platform Issues](https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues) in which OpenSSL gets a [honorable mention](https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues#openssl).

dirkhh, 2017-11-06:
Oh, I thought I had responded on the comment on 769 but apparently didn't...
removing those libraries does NOT fix things for me on Arch Linux. Same errors.
I agree with you on many of the issues mentioned about desktop Linux. Dependency hell is real (as Subsurface constantly proves). The differences without distinction (in path names, for example) are very annoying. 
I'm not sure how I can best contribute to that wiki page because I a) fundamentally agree and b) frankly know less about this than you do...

dirkhh, 2017-11-06:
Based on mailing list feedback so far (I'll try to remember to update this comment for a while):

**working**:
- Ubuntu 15.10, 17.04, 17.10, Kubuntu 17.04 (at least the 17.xx versions with a ton of gstreamer errors on console)
- Fedora 27

**not working**:
- Arch Linux
- openSUSE Leap 42.3

probonopd, 2017-11-06:
### tl;dr

Please try on Arch Linux (OpenSSL issue) and openSUSE Leap 42.3 (OpenSSL issue):

```
rm -rf ./squashfs-root || true # Remove possible leftovers from earlier experiments
wget "https://github.com/Subsurface-divelog/subsurface/releases/download/continuous/Subsurface-5196b46-x86_64.AppImage"
chmod +x Subsurface-5196b46-x86_64.AppImage
./Subsurface-5196b46-x86_64.AppImage --appimage-extract
sed -i -e 's|1.0.1e|1.0.0\x00|g' squashfs-root/usr/lib/libQt5Network.so.5
./squashfs-root/AppRun
```

Does it run?

---

### Long story

```
wget "https://github.com/Subsurface-divelog/subsurface/releases/download/continuous/Subsurface-5196b46-x86_64.AppImage"
chmod +x Subsurface-5196b46-x86_64.AppImage
./Subsurface-5196b46-x86_64.AppImage --appimage-extract
mv squashfs-root original
./Subsurface-5196b46-x86_64.AppImage --appimage-extract
rm squashfs-root/usr/lib/libssl.so.1.0.0 squashfs-root/usr/lib/libcrypto.so.1.0.0 
mv squashfs-root nossl
```

| Distribution | original | nossl | patched
| --- | --- | --- | --- |
| CentOS-7-x86_64-LiveGNOME-1511.iso | OK | libssl.so.1.0.0: cannot open shared object file: No such file or directory | OK |
| debian-live-8.6.0-amd64-xfce-desktop.iso | OK | OK | OK |
| Fedora-Workstation-Live-x86_64-24-1.2.iso | OK | libssl.so.1.0.0: cannot open shared object file: No such file or directory | OK |
| kubuntu-17.04-desktop-amd64.iso | OK | OK | OK |
| linuxmint-17.3-cinnamon-64bit.iso | OK | OK | OK |
| openSUSE_Leap_Live_KDE.x86_64-42.3-Build17.1.iso | symbol lookup error: /lib64/libssl.so.1.0.0: undefined symbol: OPENSSL_init_library (wrong: is loaded from the base system) | OK | OK |
| ubuntu-14.04.1-desktop-amd64.iso | OK | OK | OK |

`CentOS-7-x86_64-LiveGNOME-1511.iso` has

```
/usr/lib64/libssl3.so
/usr/lib64/libssl.so.10
/usr/lib64/libssl.so.1.0.1e
```

but not `libssl.so.1.0.0`.

Similarly, `CentOS-7-x86_64-LiveGNOME-1511.iso` has

```
/usr/lib64/libcrypto.so.10
/usr/lib64/libcrypto.so.1.0.1e
```

but not `libcrypto.so.1.0.0`.

If I symlink these in place, we get

```
/mnt/AppRun: /mnt/usr/bin/../lib/libssl.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /mnt/usr/bin/../lib/libgit2.so.26)
/mnt/AppRun: /mnt/usr/bin/../lib/libcrypto.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /mnt/usr/bin/../lib/libgit2.so.26)
/mnt/AppRun: /mnt/usr/bin/../lib/libssl.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /mnt/usr/bin/../lib/libcurl.so.4)
/mnt/AppRun: /mnt/usr/bin/../lib/libcrypto.so.1.0.0: version `OPENSSL_1.0.0' not found (required by /mnt/usr/bin/../lib/libcurl.so.4)
```

This seems so suggest that our best option is to bundle a matching version of OpenSSL at this point. But why does this fail on some distributions, such as openSUSE? Here is what I think (thanks to `LD_DEBUG=libs`) is going on:

Something causes `libcrypto.so.1.0.1e` to be searched, and since we do not bundle this version inside the AppImage, the libcrypto from the base system gets loaded instead of the bundled version, which leads to an OpenSSL version mismatch.

It is noteworthy that the Qt being used has the string "1.0.1e" compiled in whereas we have "1.0.0" in our AppImage. Possibly this casuses Qt to load 1.0.1e on systems where this version is available, which results in an OpenSSL version mismatch.

```
me@host:~$ grep -r '1\.0\.1e' original/
Binary file original/usr/lib/libQt5Network.so.5 matches

root@host:/home/me# strings original/usr/lib/libQt5Network.so.5 | grep '1\.0\.1e'
1.0.1e
```

Modifying the bundled `libQt5Network.so.5` makes it accept the bundled version of `libcrypto.so.1.0.0`, rather than trying to load `1.0.1e` from the system:

```
cp -r original patched
sed -i -e 's|1.0.1e|1.0.0\x00|g' patched/usr/lib/libQt5Network.so.5
```

This seems to suggest that the Qt distribution from the Qt Company (most likely compiled on RHEL/CentOS) has a bias toward OpenSSL version 1.0.1e.
Possibly we should contact the Qt company about it, because it is not reasonable to assume that every target system has exactly this version of OpenSSL.

Can we live with that sed hack? If not, I wonder what would happen if we were using Qt from https://launchpad.net/~beineri/+archive/ubuntu/opt-qt591-trusty.

dirkhh, 2017-11-06:
I'm testing this now - and it works perfectly on Arch Linux.
I'll try openSUSE  Leap 42.3 next.
And to answer your question, @probonopd :  I can totally live with that sed hack :-)

closed 2017-11-06 by dirkhh

joshirio, 2018-02-17:
Just wanted to drop a note about Qt 5.10.1:
use `sed -i -e 's|1.0.2k|1.0.0\x00|g' /path/to/myappimg/usr/lib/libQt5Network.so.5`
because it changed with the latest Qt SDK:

```
$ strings libQt5Network.so.5 | grep '1\.0\'.
1.0.2k
OpenSSL version too old, need at least v1.0.2
```

commit f24b6d46 (2017-11-06): Travis AppImage: work around OpenSSL issue

The Qt binaries that we use for the continues build of the AppImage reference
libcrypto.so.1.0.1e, but we bundle libcrypto.so.1.0.0. On distros that have
libcrypto.so.1.0.1e that gets loaded in addition to the one that we bundle
which causes a conflict. More details are explained in the issue below.

Fixes #779