subsurface/subsurface #783: Travis-created AppImage doesn't successfully log in to Facebook

dirkhh, 2017-11-06:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The login page is shown (so this isn't the OpenSSL issue). But then once you enter your credentials and log in, you don't actually get logged in, you're just dumped back at the login screen.
The locally-built version on Linux works just fine, so this is an oddity with the AppImage that needs debugging.

### Operating system:

Tested on Arch Linux and Ubuntu 17.04.

### Subsurface version:

Current master with Travis-built 'official' AppImage
### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@tcanabrava

hhassey, 2017-12-11:
Confirm same behavior in Debian Stretch

neolit123, 2019-02-01:
we are  removing the FB feature:
https://github.com/Subsurface-divelog/subsurface/pull/1958

closed 2019-02-01 by neolit123