subsurface/subsurface #786: Dive locations with no linked dive appear on the map

DennisArreborg, 2017-11-07:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Upon import of GPX fixes to several dives at the same dive spot I have ended up with a number of dive location markers, that are not linked to any dive.

Dispite having slightly different gpx fixes, the dives were at the same spot. So after naming the first dive location accordingly, I applied that dive location to the remaning dives. This has left several unused dive markers on the map
[image]

Whenever I select one of the unused dive spots, the notes tab changes to include the depth and duration infomation, but the selected dive is the same as before.

The extra dive spots are synced to the cloud and appears on other windows machines.

### Operating system:

GPS fixes imported from Android app 2.0(4.6.4.966), Android 7.0 on Samsung A3

PC is Windows 10

### Subsurface version:

Subsurface 4.7.2 official build.

### Steps to reproduce:

### Current behavior:

Dive locations that are not linked to any dive appear on the map.

### Expected behavior:

Only dive locations linked to a dive should appear on map.

### Additional information:

Issue reported initially reported in user forum
https://subsurface-divelog.org/user-forum/?place=topic%2Fsubsurface-divelog%2FaR_169Me5kc%2Fdiscussion

### Mentions:

@neolit123

DennisArreborg, 2017-11-07:
[logbook.zip](https://github.com/Subsurface-divelog/subsurface/files/1451463/logbook.zip)

My logbook is attached... 
There are multiple "unused" divespots around the location of e.g. dives #393-397

DennisArreborg, 2017-11-07:
oops

neolit123, 2017-11-07:
@DennisArreborg 

thanks, we seem to have a backend "breakage" of sorts,
i'm going to ping the developer mailing list as i need some decision making from there.

neolit123, 2017-11-08:
@DennisArreborg

we've decided to discard these unused dive-sites at the saving stage.
so, once you save the log book and then then reload it, these unused flags should be gone.

the fix should be available in the next windows test build.

closed 2017-11-08 by neolit123

DennisArreborg, 2017-11-08:
That's cool!

Do I need to build it myelf? Is the test build available to download somewhere?
I do not have any linux machine running at the moment, so otherwise I will just wait for the next official release

I appreciate your efforts!

neolit123, 2017-11-08:
i will just send you a message when we have a new Windows test build.
thanks.

dirkhh, 2017-11-08:
A new test build that includes this feature is available here:

http://subsurface-divelog.org/downloads/test/subsurface-4.7.2-52-g13b909cf82ef.exe

It would be nice if you could test it and let us know if this fixes the issue for you.

DennisArreborg, 2017-11-08:
> 2-52-g13b909cf82ef.exe

Succes. After I did File>Open Cloud Storage the unliked sites disappeared.

commit fdd152af (2015-01-04): Make sure we do not fall back to edit mode on accept

On Ubuntu 14.04 the edit mode does not exit successfully when applying
the changes. It instead jumps back to edit mode (even though hiding the
option to apply/discard changes again). So let's just have a flag to
prevent faulty behavior.

Fixes #786
PR #788 (2017-11-08): Do not save dive sites that are not used
### Describe the pull request:
- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
As it is not possible to delete dive sites from the logbook, we need to make sure that we never save sites that are not tied to any dive. With this change, unused site that are currently in the logbook will also be removed, so it will also clear up (wrong) historical data.

Supposed to fix #786

### Changes made:
Delete empty sites on  save.

### Mentions:

@neolit123
commit 8d0361ff (2017-11-08): Do not save dive sites that are not used

As it is not possible to delete dive sites from the logbook, we
need to make sure that we never save sites that are not tied to
any dive. With this change, unused site that are currently in
the logbook will also be removed, so it will also clear up
(wrong) historical data.

Supposed to fix #786