subsurface/subsurface #789: OSTC data parsing strange results for first and last sample

sfuchs79, 2017-11-08:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'm still haunting a strange effect around a point where I would like to improve something and now I think I found s.th. even more strange. It could be that I'm completely wrong or this could be the answer to my problem.

The thing is: I have doubts that there is a bug in the OSTC data parsing code of libdivecomputer and/or Subsurface. And I tend to say that this bug is even reflected and visible in some of our reference files for our tests (TestParse).

Please have a look e.g. at the reference file ostc.xml:
https://github.com/Subsurface-divelog/subsurface/blob/master/dives/ostc.xml#L25
In line 25 I tend to say that the CNS value there is absolutely incorrect. How can a dive start with a CNS value of 22% and then only 8 minutes later the CNS is 2%.
22% is by the way the final cns value of the other dive in the same XML file.
One please needs to look at all cns entries for first and last sample of these two dives in the file. In my eyes they don't fit!
And I can reproduce the very same effect when I import real dives from my own real OSTC hardware.

### Operating system:

Windows 10
But doesn't really matter because the issue is also reflected in our tests.

### Subsurface version:

Latest master but issue must be there since decades because the tests were the same since a long time I guess.

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

janmulder, 2017-11-08:
I can confirm the strange cns values on the first and last sample, with OSTC3+ and latest 2.26 firmware. Very, very likely a bug in our (Subsurface) processing, as the cns values in between, and between repetitive dives are "logical", so the OSTC does not seem confused. All this reminds me of a commit 9b5d3b60abca206bcb. My bet is a bug is somewhere in that file. 

As I have multiple dives in my logbook with cns over 300%, I never really cared about this value :-)

sfuchs79, 2017-11-08:
@janmulder Thanks for looking into this so quickly and thanks for the confirmation that it may be that I'm not completely crazy ;-)
One more important remark from my side: I believe this issue could be not limited to CNS but also other values (second best example is NDL) could be affected.

sfuchs79, 2017-11-08:
I did an interesting test:
Two times downloaded dives from my OSTC to a fresh and empty log. Both times I selected 3 out of many logged dives from the OSTC.
1. Selected 3 arbitrary dives which have other dives between them and the other 2 selected.
Result: Dive 1 and 2 show incorrect CNS value in first sample. But the cns value is not an value from one of the other two dives.

2. Selected 3 successive dives
Result: Dive 1 and 2 show an incorrect CNS value in first sample. Dive 1 incorrect value is the final value from dive 2. Dive two incorrect value is the final value from dive 3.
BTW: Between all these dives I have >>12h surface interval. So CNS should be "desaturated".

[Tests.zip](https://github.com/Subsurface-divelog/subsurface/files/1455106/Tests.zip)

Maybe this helps to narrow down the issue.

janmulder, 2017-11-08:
CNS and NDL are fundamentally different. Initial CNS is in the dive header, so that one might be a bug (as it is currently probably not picked up from there). NDL is a product of current remaining deco obligation and current depth, so at the surface, it is a useless number (and it is obviously not stored in a dive header).

sfuchs79, 2017-11-08:
I think we are not talking about cns in the dive header here but cns (and ndl) in the samples.
Honestly my poor guess is that we are somewhere violating the boundary of the "sample" array writing data of one sample from dive A into the data for dive B.

janmulder, 2017-11-08:
But the issue is, that the OSTC does not report a cns (and ndl) instantly at the first sample, so there is (flawed) logic, for the initial part of the samples (which is also responsible for carrying cns data from one dive to the other) to set these values. See core/libdivecomputer.c:

```
// The previous sample gets some sticky values
// that may have been around from before, even
// if there was no new data
		if (sample) {
			sample->in_deco = in_deco;
			sample->ndl.seconds = ndl;
			sample->stoptime.seconds = stoptime;
			sample->stopdepth.mm = stopdepth;
			sample->setpoint.mbar = po2;
			sample->cns = cns;
		}
```

sfuchs79, 2017-11-08:
Ok, I see. But isn't it that we then should reset cns to 0 in the same way you already fixed it for po2 value before?

sfuchs79, 2017-11-08:
And what about the last sample? Why do we have the CNS=0 there in the resulting XML?
My expectation would be that the last sample has still the max CNS value of the dive and therefore no new CNS info should be exported in the XML. The same thing I sometimes see for stoptime, ndl, po2.

sfuchs79, 2017-11-08:
Hmm.. maybe we see two things here while I was first thinking it's only one item:
1. The missing reset for the CNS. This corrupts the first sample.
2. We maybe receive a special last sample from the OSTC which doesn't hold valid info but we treat it as a normal sample. Why I say this: Today it looks af if we still receive a NDL of 240min for the second last sample and then a NDL of 0min for the last sample. And that today makes us write "ndl=0min" to the last sample in the XML which is completely useless and also a little bit misleading. Ok, that was just a theory now!

janmulder, 2017-11-09:
Ok, getting closer to a fix for the CNS weirdness. To do things right, we need to fix libdivecomputer first, and as it needs a change that IMHO should go in the upstream from Jef, we need to fix it there first and than merge it back to our branch.  This fix uses the data from the header to get the first sample right. Just resetting it to 0 on our side is just wrong (I did not try this, but this might work in a reasonable way). 

So, currently, only taking care of the initial CNS, and not trying to mix-up other stuff like NDL or last sample.

janmulder, 2017-11-09:
And in hindsight. No involvement of Jef. Almost all changes could go into our branch.

janmulder, 2017-11-15:
Ok, my earlier attempt to fix the initial CNS for OSTC was very validly refused as too complex, and expanding an interface in an unnecessary way. The much easier and cleaner fix has now landed in our branch of libdivecomputer (thanks Linus). 

With #812 currently open, there seems not much reason to keep this issue open, and I propose close. @sfuchs79 agree?

sfuchs79, 2017-11-15:
@janmulder 
I did a new build with updated libdc and #812 today and a new import from my OSTC2N a few minutes ago.
Can it be that my OSTC2N doesn't have initial CNS in the dive header?
I found in my dives one repetitive dive where there should be an initial CNS but I see no value in the first sample. What I see is a correct value in a sample @2min with an CNS>>1% which shows that there was some CNS carried over from the dive before correctly.
Maybe we can still clarify this.
Anyway for me the results are now absolutely ok and we can close this. Thanks!

janmulder, 2017-11-15:
1. check that you really linked with a libdc that has the fixes of (late) today. Normally, I compile from the debugger in qtcreator, so that does not rebuild (or pull) all the changed dependencies like libdb.
2. believe that the 2N runs hwOS and according to the HW specs, they have the same interface, so ...
3. Do not be confused that you will not see cns=0 on the first sample. It is not changed data
4. compile libdc with debug, and use the debugger to check.
5. or use dctool to pull the data from the 2N, email the dives to me, and I can take look.

janmulder, 2017-11-15:
and @sfuchs79 ... do you use the libdc simulator? very usefull for this.

sfuchs79, 2017-11-15:
> believe that the 2N runs hwOS and according to the HW specs, they have the same interface, so ...

Afaik NO hwOS on OSTC2N. It's an old one with the old type of firmware.
See:
http://heinrichsweikamp.com/service/firmwares

Did you test with your OSTC3? If it works with the OSTC3 then let's close this because also for me with the OSTC2 the results in terms of data in the XML is much, much better than before with our changes :-)

janmulder, 2017-11-15:
here is the interface doc (for 2N) from HW: https://bitbucket.org/heinrichsweikamp/ostc2_code/src/f8afe90796d6ed9abfb6e09d6b4626c10f05c82a/docu/ostc_interface_description.odt?at=default&fileviewer=file-view-default

Yes, tested with OSTC3

sfuchs79, 2017-11-15:
Interesting, thanks!
No, can't find a initial CNS neither in the short header with version 0x20 nor in the extended header with version 0x21. And you are correctly only doing the initial cns thing with version 0x22.

So no initial cns for me but I will neither start crying nor go and buy a new OSTC now ;-)

The most important thing for me is that we obviously did all these changes correctly.

commit 2d68214e (2015-09-17): Planner: show new icon to indicate that last point can't be deleted

Having the trash can disappear is somewhat intuitive, but I'd argue this
is even more so.

Fixes #789
closed 2017-11-16 by janmulder