subsurface/subsurface #794: Subsurface accesses Facebook every time it is started

dirkhh, 2017-11-09:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The logic for URL change shouldn't trigger the way I look at the code, but it clearly does. And because of that EVERY time you start Subsurface, you are accessing Facebook.
We have to stop that prior to the next release.

### Operating system:

All

### Subsurface version:

4.7.2 & current master

### Mentions:

@tcanabrava

bstoeger, 2017-11-09:
I think this happens:

Mainwindow generates a FacebookPlugin.
FacebookPlugin generates a FacebookConnectWidget.
If not logged in, FacebookConnectWidget calls facebookDisconnect().
facebookDisconnect() sets the URL and thus ultimately causes a HTTP access and an URL change (redirect of Facebook?).

When I comment out
`facebookWebView->setUrl(FacebookManager::instance()->connectUrl());`
in FacebookConnectWidget::facebookDisconnect().
I don't get a urlChanged event anymore. But I didn't check if there are no network accesses. I don't have a Facebook account, so can't be of much help...

bstoeger, 2017-11-10:
Here an untested quick fix. The facebook page is now loaded in the overriden exec() method. Disclaimer: no Facebook, don't understand the Qt object system.

[0001-Initialize-Facebook-page-only-when-dialog-is-shown.patch.txt](https://github.com/Subsurface-divelog/subsurface/files/1460828/0001-Initialize-Facebook-page-only-when-dialog-is-shown.patch.txt)

dirkhh, 2017-11-10:
Awesome - I’ll look this morning (still stuck in family duties)

bstoeger, 2017-11-10:
Let's perhaps wait for @tcanabrava for a proper fix. In the meantime, here an improved version, which uses the show-event instead of hijacking the exec() method.
[0001-Initialize-Facebook-page-only-when-dialog-is-shown.patch.txt](https://github.com/Subsurface-divelog/subsurface/files/1462159/0001-Initialize-Facebook-page-only-when-dialog-is-shown.patch.txt)

dirkhh, 2017-11-10:
@tcanabrava has been very busy with real life lately, not sure we can wait for him to do his own fix. Why don't you submit this as a pull request which makes it MUCH easier for me to test it?
At first glance it looks good.

bstoeger, 2017-11-10:
Made a PR: #798

bstoeger, 2017-11-10:
Please note that this introduces a functional change: the URL is now set every time the dialog is opened and not just once at application startup. Therefore, repeated invocations (in logged-in and logged-out state) should be tested.

janmulder, 2017-11-13:
Fix is merged => close

closed 2017-11-13 by janmulder