subsurface/subsurface #802: Depth > 20 is rounded on manually entered dives

PeterZaal, 2017-11-12:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When adding a dive manually, the dive depth is rounded to an integer value when a value with decimals greater than 20 is entered.
E.g. dive depth 15.6m is saved as is, while a dive depth of 42.8m is rounded to 43m (on pressing Apply changes).

### Operating system:
Windows 10

### Subsurface version:
Subsurface 4.7.4

### Steps to reproduce:
Add dive, enter 22.6 in the Depth field. Press Apply changes. The value is changed into 23m.

### Current behavior:
See above.

### Expected behavior:
The value should be rounded to 1 decimal, same as with values with decimals under 20.

janmulder, 2017-11-12:
On 10-12-15, @dirkhh wrote on the ML: 

> It's a matter of perceived precision (and certainly debatable). Showing 102.2m seems silly - actually, in general showing "10cm" resolution on depth is silly - is that your left arm, your right arm, or your toes? Yet not showing a decimal for single digits (6m, 7m) causes people to tell you that you are over-simplifying. So we compromise and show a decimal down to 20m (IIRC) and no decimal after that. I agree that this looks inconsistent - but I'm not sure what a more reasonable solution would look like.

I do not think that the ideas have changed since then.

PeterZaal, 2017-11-12:
Fortunately the depth recorded in the XML does have the decimals as entered. So it's a matter of display.

Still, I like to debate: as Dirk said it is inconsistent for sure. Why 20? Why not 30 (as known as a common recreational depth limit), or whatever other number. I feel that if I enter 36.7 is should display that. If I don't like digits, I will enter 37. Who is Subsurface to decide that it should display otherwise?
I don't think showing 10cm resolution is silly. For example some of us like to do a 6m deco stop as perfect as possible. And it doesn't matter if it's on my left arm, right arm or toes, I know where I wear my computer and it's consistent for me and I like to see the information as recorded or entered.

dirkhh, 2017-11-12:
And this is an open source project. Fork it, create the version that does what you want, and be happy. The developers who run this project have decided that this is the compromise that we consider OK and personally I see no reason to change it.

closed 2017-11-12 by dirkhh

PeterZaal, 2017-11-12:
Although I am a developer, I am a noob at GIT, C++ etc and not able to setup a build environment and all, so unfortunately that's no option for me.
I was hoping my reasoning would provide a new look into this, and sometimes opinions can change over time, right? But I guess I failed at that, oh well, can't have it all ;)

atdotde, 2017-11-12:
Plus the 10cm precision can be total fake: My Suunto Vyper/Vytec computers for example log depth in integer multiples of 1 foot. So the accuracy is at best 30cm. And the sensor has jitter.

dirkhh, 2017-11-12:
and as someone who almost always has multiple dive computers with him... two dive computers on the SAME WRIST will not only show different depth (by as much as a meter) but also show different differences at different depth (so DC1 is shallower at shallow depth, but possibly deeper at deeper depth) and different jitter.
And also, in most places (except for a secluded mountain lake, maybe), wave and current action on the surface will make the depth oscillate by more than 10cm.
Anyway, the issue is closed