subsurface/subsurface #844: AppImages for non-continuous tags contain apparently wrong update information

TheAssassin, 2017-11-21:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
The AppImages published as separate tags contain update information pointing to `continuous`. Hence, they will update to future releases of the `continuous` tag, not to the one they're currently published to.

The file has been downloaded from the "Continuous build (testUploadtoolPRComment)" release.

### Operating system:
irrelevant

### Subsurface version:
Subsurface-4.7.4-66-gd51bb52d6158-x86_64.AppImage

### Steps to reproduce:
-/-

### Current behavior:
See the following example:

```
> ./appimageupdatetool-163-d6834d6-x86_64.AppImage -d Subsurface-4.7.4-66-gd51bb52d6158-x86_64.AppImage
Parsing file: Subsurface-4.7.4-66-gd51bb52d6158-x86_64.AppImage
AppImage type: 2
Raw update information: gh-releases-zsync|Subsurface-divelog|subsurface|continuous|Subsurface*-x86_64.AppImage.zsync
Update information type: ZSync via GitHub Releases
```

### Expected behavior:
The update information should contain the right tag name.

### Additional information:
The only reason to have this tag contain update information pointing to `continous` would be if the tag/branch was being abandoned, to make people switch to another "update channel" for future updates. If this is the case, please let me know (and please add that to the release notes of that specific release). At the moment, it looks like this is not the case.

### Mentions:
CC @dirkk @probonopd

dirkhh, 2017-11-21:
I haven't looked at the magic that goes on in the tooling there. I agree that it would make sense to have the update information to be based on the same tag - but I'm not sure where the tools get that tag from, so not sure what we'd have to change in our Travis CI setup to have it do the right thing.

Ideally you'd be able to give the "update channel" as a command line argument and overwrite what's there. Let's say you downloaded an AppImage in "continuous-Feature-A" and test a few iterations in that branch and now that feature gets merged. It would be nice to just be able to give the user the option to switch back to the "continuous". And of course vice-versa. I have the "continuous" AppImage here and now want to test something in a feature branch...

TheAssassin, 2017-11-21:
@dirkhh oh, I see, Subsurface builds make use of the linuxdeployqt auto detection magic update information generation thingy.  @probonopd is there a way to override `continuous`? Possibly using the same logic used in uploadtool to determine the release name?

probonopd, 2017-11-21:
You can manually pass any update information you like into `appimagetool` (which is internally used by `linuxdeployqt`) but experience has shown that it is too cumbersome (and difficult) for people to add update information, so that most people did not make use of this feature. Hence, `linuxdeployqt` instructs `appimagetool` to make an educated guess. When I implemented the logic I did this under the assumption that there are only 2 "channels", namely _continuous_ and  _tagged builds = releases_ (=everything else). Obviously with the added flexibility contributed by @dirkhh this doesn't hold anymore and we need to adjust the logic a bit. I will need to make the "educated guess" a bit more "educated".

Let's make sure we have the same thing in mind:

* Tagged builds should get 'continuous' (a reserved GitHub keyword that resolves to the latest tagged GitHub Release not tagged as "pre-release") in the update information, 

* Untagged builds for which `$UPLOADTOOL_SUFFIX` is set and is not empty should get 'continuous-$UPLOADTOOL_SUFFIX' in the update information because uploadtool will create a tag `continuous-$UPLOADTOOL_SUFFIX` for them

* Untagged builds should get 'continuous' in the update information because uploadtool will create a tag `continuous` for them

Can you confirm that this is the logic we want to achieve @dirkhh @TheAssassin?

TheAssassin, 2017-11-21:
@probonopd I'd rather introduce a more generic variable (or, at least with a `APPIMAGE` prefix), and then `export UPLOADTOOL_SUFFIX="$APPIMAGE_GITHUB_UPDATE_CHANNEL"` (it's not that concise, but at least precise).

Maybe we should move the discussion to the linuxdeployqt repository.

probonopd, 2017-11-21:
@TheAssassin no, since "educated guess" should mean "no extra work", i.e., no additional environment variables over those that have already been introduced. Who wants more fine-grained control can always supply custom update information manually (but almost no one ever does).

probonopd, 2017-11-21:
In my opinion `$UPLOADTOOL_SUFFIX` does well enough as a "namespace" so as not to clash with other environment variables. What I want to avoid is that the user needs to think about an _extra_ environment variable for the purpose of making the "educated guess" work, as that should happen seamlessly without the user ever having to think about. (That's what I've learnt from Apple: Have all the command line switches under the hood, but choose defaults so that 99.9% of the users will never even have to know they exist.)

dirkhh, 2017-11-22:
Sorry to have dropped out of the discussion for a moment - work was killing me the last two days.
I kind of agree with @TheAssassin - the uploadtool is completely independent from the AppImage tooling, so it would make sense to use an APPIMAGE_ prefix for such a variable.
And as a user of both tools, I don't mind the duplication - it's not like it's a lot of work to get that right. But it does seem more consistent.

probonopd, 2017-11-22:
> the uploadtool is completely independent from the AppImage tooling

Sorry but I have to disagree here: If we introduce (without need) a second variable, then the user has to think about two instead of one variable, and worse, needs to ensure that the content of those two variables is consistent, because otherwise the update information embedded inside the AppImage will not match to the file on the server, resulting in breakage. Hence I still think one variable should be enough (whatever we name it).

Apart from the name and number of variables, do you think the logic that I had proposed above is correct and what we want?

dirkhh, 2017-11-22:
Yay, let's fight over variable names. It's ON! :-)
Seriously though - I understand the concern about multiple variables. But the challenge is this:
uploadtool has truly nothing to do with AppImages. I use it to upload my Windows installer and the Mac .app.zip file and the Android .apk. So having uploadtool use an `APPIMAGE_` variable would be truly weird.
And the AppImage infrastructure is of course completely independent from the uploadtool. My guess is that most people don't even use it in the context of Travis CI. So having it use an `UPLOADTOOL_` variable would be rather confusing.
So what is that concept that we are trying to implement here. It's an update channel name, correct? And that's true in both instances. Would `UPDATE_CHANNEL_NAME` be a reasonable variable name for both use cases? I think so.

To your other question... your writeup above... how do I reference other comments... oh bother...
This:

> Let's make sure we have the same thing in mind:
>
> - Tagged builds should get 'continuous' (a reserved GitHub keyword that resolves to the latest tagged GitHub Release not tagged as "pre-release") in [...]

I think the first bullet talks about `latest` being reserved? And that should be the channel?
Other than that I agree. But as I mentioned somewhere - I'd love the ability to switch in flight between channels.

probonopd, 2017-11-22:
Correct, that was a typo, first bullet must be `latest`; edited above. Thanks for catching this!

probonopd, 2017-11-22:
>  So having uploadtool use an APPIMAGE_ variable would be truly weird.

Which is why I suggested to use the already existing `$UPLOADTOOL_SUFFIX` variable when making the "educated guess" about what update information should be embedded.

> I'd love the ability to switch in flight between channels.

You can always do this by supplying a different string to `appimagetool -u`.

glance-, 2019-11-10:
This is probably outdated now when we run on github actions and use it to provide the binaries.

commit 1982bdb9 (2015-03-23): Implement handler for cancel button of Save As dialog

When an user opened the "Save as" dialog and pressed the
cancel button a null string was returned. Therefore the
file_save_as function returned an error which was lately
shown when the file_save function was called.
Now the function checks if the cancel/exit button was
pressed and returns.

Fixes #844

Reported-by: longjohnsilver
closed 2019-11-10 by glance-