subsurface/subsurface #846: Get rid of the stupid MentionBot

janmulder, 2017-11-21:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Simple, the MentionBot does not work. It almost never fires, and when it fires it produces BS (like Robert has to review his own stuff). IMHO: just disable all this. While the idea is good, it does not work.

### Mentions:
euh .... @neolit123 ?

neolit123, 2017-11-21:
talk to @dirkhh

dirkhh, 2017-11-21:
I agree with @janmulder on both counts. It's a great idea. But it just doesn't seem to work well enough to be worth it. I'll disable it.

closed 2017-11-21 by dirkhh

dirkhh, 2017-11-21:
Hook is disabled, see https://github.com/Subsurface-divelog/subsurface/commit/32a7cde180a342aef13b1ffcbe53a088ab4dc51c

commit 57374fb9 (2015-03-15): Profile: call clearHandlers() in setAddState()

Points (handles) from the last dive ADD operation remain on
the profile unless we clear them.

fixes #846