subsurface/subsurface #849: Planner crash on recreational mode

DennisArreborg, 2017-11-23:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When changing from Bühlman deco to recreational mode subsurface goes Not Responding

### Operating system:
WIN10

### Subsurface version:
subsurface 4.7.4

### Steps to reproduce:
Open planner
default dive is 15 m 20 min. Change to 30 m 20 min.
If in recreational mode, program stops. If not in recreational mode, program will stop uåon change to recreational mode
[image]

### Current behavior:
Crash

### Expected behavior:
No Crash

### Additional information:

### Mentions:

sfuchs79, 2017-11-24:
It took me some time but now I at least can reproduce this.
I had to change at least one of the two ascend rates for shallow depths (6m to surface and 50% to 6m) from my default values (3m/min and 1m/min) to higher values like 10m/min in the above screenshot to trigger the freeze.

Unfortunately I'm up to now not able to retrieve any useful debugging information.
What happens is rather a freeze than a crash and attaching gdb didn't give interesting info. But I don't have much experience in using gdb - what other command than "bt" could give useful info after one attached gdb to the frozen process?

atdotde, 2017-11-24:
You really want to use the debugger through Qt Creator. There you can single step through the code, see variables, set breakpoints etc etc all this much easier than in the command line interface of gdb.

sfuchs79, 2017-11-26:
I wanted to look at this again with the result that there is a chance that this issue disappeared with the latest changes in the planner. Or I'm once again too stupid to test with the correct setup....
@DennisArreborg 
Could you please check the latest Travis Continuous Build from here:
https://github.com/Subsurface-divelog/subsurface/releases

DennisArreborg, 2017-11-27:
It runs smooth here as well!

closed 2017-11-27 by DennisArreborg