subsurface/subsurface #878: Planner variations - list of issues

sfuchs79, 2017-11-28:
### Describe the issue:

- [x] Bug
- [x] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

@atdotde 
I'm going to open an issue to collect issues and feature requests in the current planner variations implementation. Just to make one thing clear: I'm doing this because I like the feature a lot and would like to improve it :-)
Here is what I have up to now:
- Plan update issues described here #874 
- Discrepancy in runtime described here #874 
- Possible mistake in calculating variations:
Additional deco stops are not considered.
If I look at this code here:
https://github.com/Subsurface-divelog/subsurface/blob/master/qt-models/diveplannermodel.cpp#L981
I would say you are ignoring possible additional stops in mid and max. This leads to underestimated delta time.
- Calculating average of +- results but displaying it as + results.
My understanding is that today you calculate results for +-1m and +-1min, then you average these separately but then you print it as "+ x.xx /m + x.xx /min". Is this approach good at all? Maybe focus on +Xm / +Xmin only?
- When printing result for +1min for the user it looks like this is a delta to total runtime. But in reality today it is just a delta for decostop time. I would prefer to print the real delta for runtime by simply adding the bottom time delta (today +-1min hard coded) as well.
- Honestly now that I start to like the feature I want to be able to configure the delta depth/bottom time. Would be ok for me to only use one UI element (combobox) for both, linking it together like "+1m/+1min", "+2m/2min", "+3m/3min" or two separate UI elements.
- Preparation for imperial units (ft) completely missing.

atdotde, 2017-11-28:
Lots of good points. The first two, I hope, are already addressed in the original PR. Regarding missing stops, I have to think about this when I am less tired than now. Will think this through tomorrow.

Regarding 4, 5, 6: Let me give a bit of philosophical background: This feature is intended to be a simple helper if you want to do something along the lines of ratio deco/deco on the fly: You basically have a plan but want to have some flexibility under water to do updates if your bottom segment changes. Motivated by @RickMWalsh's post https://thetheoreticaldiver.org/wordpress/index.php/2017/11/02/ratio-deco-is-it-nonsense/

But, since we are a planner, we don't have to do this relative to some fixed dive but we can do it relative to a dive we are actually planning. So, what I really intend to compute is the partial derivative of the stoptime with respect to depth of last segment and duration of last segment. So the diver knows how much stop time to add or scratch off when varying depth or time. To deliver compact information, we don't do it per stop (although we compute it to second precision) but rather the sum and the diver is supposed to distribute that sum reasonable over stops.

As we do Delta t instead of dt, there are left and right derivative (and we compute the symmetric average). When you turn on the corresponding compiler option you can have both printed. In my testing it turned out, the difference is typically at most a few seconds (so the second derivative is quite small), much less than the precision of this crude measure, so there is no point in listing plus and minus separately.

Anyway, the diver is supposed to multiply this number by the actual difference in minutes or meters (both can be positive or negative) to get to the actual time to add/subtract. And of course this approximation gets worse if the number of meters/minutes is more than a few. But it is good enough. (BTW, this is also why I list the variations including seconds even though stops are rounded to full minutes: Those seconds can become minutes when you multiply). As I expect the diver to do that in her head, I don't think there is a point to offer explicit other differences (the result of the multiplication would be good enough).

Whether the number shown is actually the difference in stop time or total runtime (i.e. if we add the minute you suggest for the change in bottom time) is a matter of taste. I guess it the same question as if people do their stops based on runtime or stop time for the individual stop (which we both allow to be shown or hidden). I don't know.

Adding imperial unites should be easy. Will do that tomorrow.

Writing all this I realise I probably also need to write a section on this for the manual...

atdotde, 2017-11-28:
Imperial units done over there...

sfuchs79, 2017-11-28:
Also need to go to bed today... so only quick answers and then let's continue this during the next days.
1)) Unfortunately with the latest change the UI is again less reactive/slower. A poor guess: For there overall concept, may there be the need to kill a running thread as soon as another change in the UI is done?

4-6)) I now understand the idea much better, thanks! I also did some tests and it became clear to me that this simplification totally makes sense.
But one point still open: Delta stoptime vs. delta runtime may be a matter of taste. But, if we print it as:
>Runtime: xxmin VARIATIONS

then I really expect delta runtime.
If we output delta stoptime I would write
>Runtime: xxmin
>Stoptime variation: VARIATIONS

Honestly I do prefer the runtime variations.

7)) You implemented depth variations for 1ft now? I doubt that this is really useful, thats only ~30cm?! Maybe the imperial units users may like 3ft? 5ft? 10ft?

PR #902 (2017-12-02): Planner variations - simpler deco stop time calculation and variables use for variations

### Describe the pull request:

- [ ] Bug fix
- [x] Functional change
- [ ] New feature
- [x] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

A few change proposals for the planner variations to be discussed with @atdotde 

BTW:
I really appreciate this added string "Stop times" in the planner notes. Now this is ok for me because it is clear that we are NOT printing runtime variations.

### Changes made:

### Related issues: [... cut]
closed 2017-12-02 by atdotde