subsurface/subsurface #898: Dive picture tab: Show all pictures from trip / Zoom picture thumbnails

sfuchs79, 2017-12-01:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

These are two new features requests for the dive picture tab:
1. When selecting a dive trip it would be nice to display all pictures from all dives in the dive trip. Today, before commit https://github.com/Subsurface-divelog/subsurface/commit/0fc4bd8976e5459b84307b595049f081ed8c2c78 only the pictures from last dive where displayed. With the referred commit I currently disabled the dive picture tab when a trip is selected. This would then need partial revert.
When displaying all the pictures from multiple dives at the same time it would be great to show a kind of separator between pictures from different dives. But this is no must have for a first version ;-)
2. Enable the possibility to zoom the picture thumbnail size between e.g. 30% and 300% (100% = size today). I tried to to this but failed. But I have the very first step to do this here:
https://github.com/sfuchs79/subsurface/commits/feature_dive_picture_zoom
In the code I also put some comments which may help as a starting point.

### Mentions:

Looking for someone more experienced than me to finalize this.
@neolit123 @tcanabrava @bstoeger @atdotde others...

sfuchs79, 2017-12-01:
Hint how to add separators in QListWidget:
https://stackoverflow.com/questions/24671636/qt-qlistwidget-separator-line-after-particuler-items

sfuchs79, 2017-12-14:
@bstoeger 
May I ask you also for a suggestion on this because you now already know this part of the code well:

I still would like to try to implement the zoom feature but I'm already struggling with introducing an additional int variable like "scale" with values like 100(%) or 200(%) for the scaleImage function here:
https://github.com/sfuchs79/subsurface/blob/feature_dive_picture_zoom/qt-models/divepicturemodel.cpp#L13
Main reason is that calling this function via QtConcurrent::blockingMap(pictures, scaleImages); doesn't allow to pass any additional variables. Which way to go? Global variable? If yes where and how to define it and use it?

Then a second question is how to handle the thing with the thumbnailCache and different scale factors. My first idea is to only use the cache for 100% scale (picture size as today) and always bypass the cache if pictures are zoomed to s.th. smaller/larger than 100%. Plus when changing dive selection always reset scale to 100%.
Otherwise we would need to cache multiple copies of each picture with different scale factors plus store the scale factor for each copy -> faster but more difficult and larger cache file.

atdotde, 2017-12-14:
The way to give additional parameters to blocking map is to apply it to a different list: Instead of using a list of pictures make it a list of pairs: In the pair is the picture and the scale factor.

Regarding the caching: Please keep in mind that you don't want to slow down the UI my a lot. I guess the critical operation is having the picture tab open and scrolling through the divelist using cursor keys.  You definitely don't want to reload the images. But please do a bit of testing to make sure the rescaling is not too bad (I guess it is not as long as the pictures in original size are in memory).

bstoeger, 2017-12-15:
If I understand correctly, you want all pictures have the same zoom level, right? Then there is a rather simple way. It may prove to be rather unpopular though, so you didn't hear it from me. ;)
```
void scaleImages(PictureEntry &entry, int scale)
{
...
}
...
QtConcurrent::blockingMap(pictures, [scale](PictureEntry &p) { scaleImages(p, scale); });
```

Concerning caching of thumbnails, how about resizing to the maximum supported resolution and then always scale it down to the actual display resolution? Rescaling small images is pretty fast...

bstoeger, 2017-12-17:
@sfuchs79: Are you actively working on this or should I have a go at it? I think this could be done rather easily.

sfuchs79, 2017-12-17:
@bstoeger: Due to lack of time not one line of code written on my side so far ;-(
I would be really happy if you implement this. I'll be available for testing.

sfuchs79, 2017-12-19:
Closed by #956

closed 2017-12-19 by sfuchs79