subsurface/subsurface #917: Subsurface AppImage on Gentoo throws libgssapi_krb5.so.2: cannot open shared object file: No such file or directory

zdavatz, 2017-12-06:
Hi

I just tried to install Subsurface using AppImage.

Then I get the following error:
```
./Subsurface-4.7.4-x86_64.AppImage: error while loading shared libraries: libgssapi_krb5.so.2: cannot open shared object file: No such file or directory
```
According to your description, AppImage should just work.

I am on Gentoo Linux, 64bit Kernel 4.14. wie FUSE compiled in.

Looking forward to your reply.

dirkhh, 2017-12-06:
No idea. I don't think anyone ever tried this on Gentoo.
@probonopd - have you seen this?

probonopd, 2017-12-06:
`usr/lib/libcurl.so.4` and `usr/lib/libcups.so.2` are pulling in Kerberos dependencies. Most people do not use Kerberos, so we might think about disabling it, hence removing the dependency on Kerberos.

For the meantime, I have [removed](https://github.com/probonopd/linuxdeployqt/commit/ff5cd2102bce70746a28446839e48a309e463fc9) `libgssapi_krb5.so.2` from the blacklist in `linuxdeployqt` so that it gets bundled into the AppImage, so the next builds should fix this.

zdavatz, 2017-12-06:
Ok, cool, thank you. Let me know when I can test again. BTW, which C++ buildframework do you use for distributing Subsurface binaries?

probonopd, 2017-12-06:
@zdavatz CMake, as can be see in https://github.com/Subsurface-divelog/subsurface/blob/master/scripts/linux/travisbuild.sh.

probonopd, 2018-01-13:
@TheAssassin wrote [here](https://github.com/desktop/dugite-native/issues/42#issuecomment-352253495):

> Since libcurl is built to support a variety of protocols, it can be annoying to have to bundle all those libraries as well. Also, as part of the series https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues, the CA chain is located in different places on all platforms. libcurl solves the issue by having the user specify the location on configure time. To fix this issue and also reduce the amount of dependencies, I started to build my own packages on OBS, based on the Debian ones, patched to support only HTTP(S) and search for the certificate location on runtime.
> 
> Check out https://build.opensuse.org/package/show/home:TheAssassin:AppImageLibraries/curl-httponly. Use like https://github.com/AppImage/AppImageUpdate/blob/rewrite/resources/bootstrap-container.sh#L8-L10. The packages should upgrade fine on Debian based systems. The package has been forked from the latest available unstable package.

elbaulp, 2019-09-08:
@zdavatz How did you installed AppImage on Gentoo? I can not find any documentation about it.

probonopd, 2019-09-08:
@elbaulp applications shipped in AppImage format do not need to be installed. Just make executable and run.

https://docs.appimage.org/introduction/quickstart.html#ref-how-to-run-appimage

elbaulp, 2019-09-08:
Silly of me... I read that section of the docs, but still though I had to install something in order to run it...

Thanks

TheAssassin, 2019-09-08:
@elbaulp if you think the section needs some clarification with regards to that, please feel free to open an issue and/or send a PR.

elbaulp, 2019-09-08:
@TheAssassin Nop, its perfectly clear, was my fault :-)

commit dbf2868d (2015-08-19): Restore selection after renumbering dives.

After renumbering any number of selected dives, the selected
dives would still be selected, but this was not visible from
the divelist view. Clicking on subsequent dives would deselect
them while the GUI shows them as selected, any further action
like delete would not be done on the visible selection, but
on the invisible dive->selected state, leading to apparently
very weird behaviour.

Fixes #917
closed 2019-11-10 by glance-