subsurface/subsurface #922: Subsurface does not honour UDDF <divemode> tag

bluppfisk, 2017-12-08:
### Describe the issue:

- [ ] Bug
- [ X ] Change request
- [ ] New feature request
- [ X ] Discussion request

### Issue long description:
Probably because of how Subsurface is built, it does not seem to honour the UDDF format's <divemode> tag, which is defined at <waypoint> level and can be either opencircuit, closedcircuit or apnoe (or the correct English term apnea since the soon to be released UDDF 3.2.2).

Irrespective of what divemode type is set, Subsurface will always default to 'OC' in its interface. I would love to see it switch to 'Free' if <divemode type="apnea" />

### Operating system:
Ubuntu 16.04

### Subsurface version:
Subsurface 4

### Steps to reproduce:
Import an UDDF with <divemode type="apnea" /> (or 'apnoe') tags at waypoint level (sample: http://worldofnonging.com/deepblu-tools/done/backup_f1f1f57762.uddf) into Subsurface

### Current behavior:
Observe the dive type not changing to 'free', but rather remaining at 'OC'

### Expected behavior:
Dive type should be set to 'free'.

### Additional information:
There is more to it than this and it may prove a bit complicated to solve. The UDDF format allows for a dive to consist of several dive modes, including opencircuit and apnea. This is why the UDDF format sets the dive mode at waypoint level. Subsurface, however, considers dive mode a setting at dive log level.

### Mentions:

mturkia, 2017-12-10:
I have just implemented support for importing the divemode. This will choose the mode from the first waypoint that includes this information. This will naturally ignore the case that one switches to different mode during the dive. Such a case might be when a rebreather diver switches to bailout gas during the dive. However, even in this case (IMO) the dive should still be considered a rebreather dive.

atdotde, 2017-12-10:
The subsurface Convention was that any dive that has a rebreather segment (possibly only indicated by a non-zero setpoint) is a rebreather dive.

bluppfisk, 2017-12-10:
Great! Looking forward to testing it out :)

commit 38cc8bdc (2017-12-10): Support dctype/divemode on UDDF import

This will grab the divemode information from the first waypoint within
the dive that includes this info and uses it for dctype.

Note that our dive type is not as fine grained as the one in UDDF. The
latter allows one to switch divemode during the dive when we do not.
However, one could argue that i.e. a rebreather dive should be a
rebreather dive even if one switches to bailout gas at some point...

Fixes #922
closed 2017-12-10 by mturkia