subsurface/subsurface #923: Target runtime ooo deco time in planner

atdotde, 2017-12-08:
This is an idea I got from the latest version of pasta deco: Have the option to let the planner figure out the bottom time for a given target total runtime ("I have to be back for my favourite TV show")  or target total deco time ("I get bored after 2:30").

### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2026-10-18:
Hi @atdotde,

Worth flagging upfront: this isn't a simple UI change, it inverts the planner's normal forward calculation (bottom time -> runtime/deco) into a search/solve problem (target runtime/deco -> bottom time), likely needing an iterative search given the non-linear relationship involved. That's a capability:high, dedicated-design-effort scope, not a quick addition.

Given that, is this still something you'd like prioritised? It's a genuinely useful planning convenience, just want to set the right expectations on effort before committing to it.

notionparallax, 2026-11-10:
This is similar to something I have in my planner (personal toy, not for general use) https://diveplanner-demo-page.streamlit.app/ and https://github.com/notionparallax/dive_planner

It doesn't do a "be back on the boat in 1 hour" approach, although it could. It does a "given the scenarios you want to plan for, and the gasses that you have, what's the maximum bottom time you can reasonably have". It does it through an iterative search.

Oddly, when I need to be back on the boat at a specific time, I turn the smart bit off, and just click down on the bottom time spinner until I surface at the desired time.