subsurface/subsurface #924: Too specific dependencies for open Suse leap

atdotde, 2017-12-08:
I had a user report on Facebook that our rpm packages for openSUSE Leap 43.2 (he probably meant 42.3) require qt in version 5.9.2 but he has 5.9.3 and says we should only require 5.9

I have no idea what we actually do need and what we should require, just reporting. 

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2017-12-08:
I don't think we are doing that directly. That's an OBS artifact. Similar to the Fedora issue in the user forum

mikeller, 2026-10-17:
Hi @atdotde,

This report is from 2017 and describes what looks like an OBS packaging artefact specific to an openSUSE Leap version from that era, openSUSE Leap has had many releases since, and we have no further detail or corroborating reports. We're closing this as stale.

Please reopen with your current openSUSE Leap version and the specific dependency conflict if you're still hitting this on a current build.

closed 2026-10-17 by mikeller