subsurface/subsurface #937: wrong volume calculation

28173, 2017-12-12:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When selecting the 12l 232 bar tank and putting in a start/end pressure of 150/120 the resullt of the gas volume at standard presure should be 30x12=360 liter. the system culcalutes 341 liter
### Operating system:
 windows 10
 laptop
no

### Subsurface version:
 the newest
 yes
 official release

### Steps to reproduce:

put in tank 12l 232 bar and start pressure 150 bar end pressure 120 bar
### Current behavior:

calculates 341 liter
### Expected behavior:

360 liter
### Additional information:

### Mentions:

janmulder, 2017-12-12:
A small paragraph from the user manual: 

22.1. Subsurface appears to miscalculate gas consumption and SAC

Question: I dived with a 12.2 l tank, starting with 220 bar and ending with 100 bar, and I calculate a different SAC compared what Subsurface calculates. Is Subsurface miscalculating?

Answer: Not really. What happens is that Subsurface actually calculates gas consumption differently - and better - than you expect. In particular, it takes the incompressibility of the gas into account. Traditionally, gas consumption and SAC should be:
consumption = tank size x (start pressure - end pressure)

and that’s true for an ideal gas, and it’s what you get taught in dive theory. But an "ideal gas" doesn’t actually exist, and real gases actually don’t compress linearly with pressure. Also, you are missing the fact that one atmosphere of pressure isn’t actually one bar. So the real calculation is:

consumption = (amount_of_air_at_beginning - amount_of_air_at_end)

where the amount of air is not just "tank size times pressure in bar". It’s a combination of: "take compressibility into account" (which is a fairly small issue under 220 bar - you’ll see more differences when you do high-pressure tanks with 300bar) and "convert bar to atm" (which is the majority of your discrepancy). Remember: one ATM is ~1.013 bar, so without the compressibility, your gas use is:

12.2*((220-100)/1.013)

which is about 1445, not 1464. So there was 19 l too much in your simple calculation that ignored the difference between 1 bar and one ATM. The compressibility does show up above 200 bar, and takes that 1445 down about eight litres more, so you really did use only about 1437 l of air at surface pressure.

So be happy: your SAC really is better than your calculations indicated. Or be sad: your cylinder contains less air than you thought it did. And as mentioned, the "contains less air than you thought it did" really starts becoming much more noticeable at high pressure. A 400 bar really does not contain twice as much air as a 200 bar one. At lower pressures, air acts pretty much like an ideal gas.

closed 2017-12-12 by janmulder

28173, 2017-12-12:
Hi Jan,

I agree on the scientific approache, air is far away from an ideal gas and
indeed there is a conversion between Bar towards ATM. However, this not
really common practice for the recreational diver.

Even when I take the error of calculation bar vs atm into account the
difference is only 5 liter instead of 19. Based on your way of calculation
:12*((150-120)/1.013)= 355.4
That is why I raised it as an incident.

Regards,
Leo

dirkhh, 2017-12-12:
You didn't read what Jan wrote, right? It's not ATM vs bar that creates the error, it's the compressibility of air.
So what would you like your software to show you: an incorrect value that "feels right", or the best approximation of the correct value that we can give based on the information that we have (because we don't have temperature information, and temperature of course also figures into this calculation).
So no, the result from Subsurface is indeed "more correct" than what you suggest.

commit 05f4f9b5 (2015-10-05): Cmake: ensure marbledata is actually copied into the build directory

This can be done regarless of whether Marble is enabled or not. No harm
done.

Fixes #937