subsurface/subsurface #949: mobile: edit dive date does not save

janmulder, 2017-12-14:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:
What version of Subsurface are you running?: latest master
Does the same happen on another Subsurface version? probably yes. code is untouched for long time
Are you using official release, test build, or compiled yourself? self compiled
Provide Git hash if your are building Subsurface yourself.  375490dfc9dddb01b

### Steps to reproduce:
mobile: edit dive date, and save

### Current behavior:
No errors, but no save.

### Expected behavior:
saved change and divelist re-sorted

janmulder, 2017-12-14:
Hmm. Even a developer can be confused. I tested this by just editing an existing date to (for example) one month earlier. And did not look to the log/console output ... which shows "unable to parse date 11/01/17 11:46 A.M. with the given format  MMM d, yyyy h:mm AP". The given format comes from the preferences, and this is not honored by the mobile app (another problem?). Then, reading the code ... oh boy ... the comments are "what a pain", "stupid Qt" and "WTF" :-) I seemed to have stept on a true minefield.

Back to thinking how to proceed here ...

janmulder, 2017-12-15:
Just to document another curiosity. Manually add a dive. Notice that the date field is filled in with the current date/time formatted according to the preferences. Save this (sparsely filled) dive. And see that the console prints a unable to parse date message. But in the dive list the dive shows up with a correct date.

dirkhh, 2017-12-15:
Hey at least we are consistent in being inconsistent.
:man_facepalming: 

Are you looking into that or do you want me to add it to my list?

janmulder, 2017-12-15:
Its low on my list ... editing a dive date is something that I hard ever do, and that holds probably for a lot of people, as this issue has been unreported from a long time.

That the date format not coming from the preferences on mobile, is higher on my list. As I have my mobile in English, and have US dates as well due to that ... that I do not like.

janmulder, 2017-12-16:
@dirkhh: I see on the device, that in the Subsurface.conf file, there is no data for the [Language] block, so on mobile, the date/time setting is always coming from the device. Is this intentional, or just (poor) implementation?

dirkhh, 2017-12-17:
Back when we started Subsurface-mobile we felt that it would be smart to just use whatever language the device is running and not add too many config options.

janmulder, 2017-12-17:
I now remember the discussion over minimal config options in the mobile app, and agree on this. The described bug is fixed without changing anything to this preferences strategy.

janmulder, 2017-12-17:
But ... keep thinking over the settings/preferences stuff. We have (on linux and Android, and probably other plaforms too) Subsurface.conf, independent of cloudstorage, but we also have a 00-Subsurface sitting in the cloud account (so shared between devices, attached to that account). That file has some settings data, but it is unclear to me, what functionality is intended with it.

closed 2017-12-17 by dirkhh

dirkhh, 2017-12-17:
We have a few settings we want to share between desktop and mobile. Those go into 00-Subsurface

I will NOT claim that this is well designed, well thought through, consistently handled, etc.
As we all know, a lot of this is a hack on top of a kludge made to quickly fix a problem without fully thinking through the consequences.

So always feel welcome to challenge assumptions here. My comment on all this is usually to reflect on my / our thinking that got us to where we are. Sometimes I feel fairly strongly that those assumptions are correct (I don't want a ton of settings options on mobile, for example), in other cases this is just for historic reasons and changing it would be the right thing to do.

E.g. in this case... we could indeed write the language / unit prefs into 00-Subsurface because it's a reasonable assumption (IMHO) that the user wants this consistently everywhere...

PR #955 (2017-12-17): mobile: use short date format in edit mode

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
See issue #949. In the dive list, dates are shown in short format, and when we start to edit an existing dive, the date field on the edit page is shown in sort format. However, when adding a new dive, the initial date shows up in long (normal) format. This in not only inconsistent, but also introduces the parsing problem, as descri [... cut]
commit b59778ab (2017-12-17): mobile: use short date format in edit mode

See issue #949. In the dive list, dates are shown in short format, and
when we start to edit an existing dive, the date field on the edit
page is shown in sort format. However, when adding a new dive, the
initial date shows up in long (normal) format. This in not only
inconsistent, but also introduces the parsing problem, as described
in the mentioned issue.

This can be fixed by using short formatted dates in all cases in
the mobile app. As the screen real estate is precious on mobile,
this seems the most logical choice.

Fixed: #949