subsurface/subsurface #969: UI: Filter on buddies broken

janmulder, 2017-12-21:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Issue reported on out google forum by yrevawerd@gmail.com.

> Installed 4.7.5 on Windows 7
> 
> Go to Log>Filter Dive list
> 
> Tried to filter Persons, for example checked an individual with 10 dives listed only 2 showed up, checked other individuals some all dives show up,  some they don't. Could not determine what was the criteria that made the dives show.
> 
> Checked the other categories- Tags, Location and Suits all 3 of these seem to be working properly.

Some initial analysis. In the filter list, all dives with a specific buddy are counted. Also in case of multiple buddies. Filtered dives only show dives that have exactly the one selected buddy. 

### Operating system:
What OS are you running, including OS version and the language that you are running in: WIndows 7, and reproduces on Linux, so likely not OS related

### Subsurface version:
What version of Subsurface are you running? 4.7.5
Does the same happen on another Subsurface version? currently unknown
Are you using official release, test build, or compiled yourself? official release

### Steps to reproduce:
Just filter on buddy, and compare the counts.

### Current behavior:
See above.

### Expected behavior:
Show all dives that the buddy participates in (as it the count)

### Mentions:
reported by yrevawerd@gmail.com

janmulder, 2017-12-21:
And bisected this to:

6343515fedbc43be4fd2cb3f1b3fea384e362c59 is the first bad commit
commit 6343515fedbc43be4fd2cb3f1b3fea384e362c59
Author: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
Date:   Sat Nov 25 18:26:00 2017 +0100

    Use equality instead of substring comparison in suits and buddy filter
    
    This commit is a continuation of commit 739b27427cfb5119eebe214c984843cd5d155620,
    in which a substring comparison was replaced by equality comparison to
    avoid confusing UI behavior of the filter interface.
    
    The suit and buddy filters were plagued by the same problem, so change
    their code in analogy.
    
    Fixes #551 (in conjunction with commit dd2466f51899aae406dc8c13904787710f30ec1c).
    
    Signed-off-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>

bstoeger, 2017-12-21:
Thanks, will fix..,

PR #970 (2017-12-21): Fix BuddyFilterModel
Commit 6343515fedbc43be4fd2cb3f1b3fea384e362c59 introduced equality
instead of substring comparison for filters. This broke the buddy
filter in the case of more than one buddy, because in such a case
the buddy list is a comma-separated string.

Fix this by splitting the buddy string, trimming the individual
strings and search in the list.

Fixes #969

Reported-by: <yrevawerd@gmail.com>

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

# [... cut]
commit 935fb3c3 (2017-12-21): Fix BuddyFilterModel

Commit 6343515fedbc43be4fd2cb3f1b3fea384e362c59 introduced equality
instead of substring comparison for filters. This broke the buddy
filter in the case of more than one buddy, because in such a case
the buddy list is a comma-separated string.

Fix this by splitting the buddy string, trimming the individual
strings and search in the list.

Fixes #969

Reported-by: <yrevawerd@gmail.com>
closed 2017-12-21 by neolit123