subsurface/subsurface #971: Wrong pO2 values displayed on CCR dives

Bocio, 2017-12-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Looking to one of my CCR dives (SF2 with a Petrel 2 Divecan controller) I realized that Po2 values reported by Subsurface at the 6 meter stop are way higher than those recorded and displayed by Shearwater Desktop.
I'm 100% sure that Shearwater Desktop shows correct values because I perfectly remember O2 values I was seeing at 6 meter: I was never never never over 1.6 for all the time.
According to Subsurface profile I was all the time at 6m over 1.6 with spikes at 1.75.

Now I realized that all of my dives report wrong pO2 values.

Both Jef and Anton confirmed the problem. After several analysys they didn't come out with a solution.
Maybe it's some calibration constant specific of the Divecan controller non properly used. 
I could not get any feedback from other users with a Petrel Divecan controller.

I chose a point in time (122' 50")and I attach three screenshots:

- Shearwater (which display only the AVG pO2)
- Subsurface with single pO2 sensor display
- Subsurface with AVG pO2 sensor display

[image]
[image]
[image]

I attach here raw dive data downloaded with dctool:
[dive.0001.bin.zip](https://github.com/Subsurface-divelog/subsurface/files/1582092/dive.0001.bin.zip)

Here you can find a long discussion with Jef and Anton about the bug and their findings:

http://lists.subsurface-divelog.org/pipermail/subsurface/2017-December/031101.html

### Operating system:

Win 7 64 bit

Yes is OS unrelated

### Subsurface version:

4.7.4

yes

Official release

### Steps to reproduce:

### Current behavior:

http://lists.subsurface-divelog.org/pipermail/subsurface/2017-December/031101.html
### Expected behavior:

http://lists.subsurface-divelog.org/pipermail/subsurface/2017-December/031101.html

### Additional information:

### Mentions:

@glance-

atdotde, 2017-12-22:
Do we have the data from the Shearwater app as numbers rather than a plot? Have those for all (or many) points in time should be sufficient to figure out how to do the calibration.

Bocio, 2017-12-22:
Yes,

I forgot to add them.
Both Shearwater and Subsurface exports fo that dive.

Thanks

[sheawater-and-subsurface-log-dive-data.zip](https://github.com/Subsurface-divelog/subsurface/files/1582189/sheawater-and-subsurface-log-dive-data.zip)

glance-, 2017-12-22:
My suggestion is to detect the bad cal factors, (ie when they are 2100 sharp) and just parse out the accurate average ppO2 over all the sensors, instead of trying to compute the individual O2 sensor values.

atdotde, 2017-12-22:
I meant, could you attach a file that has the data for the green line in the shearwater program (which is supposed to never go above 1.6bar)?

Bocio, 2017-12-22:
Hi Robert,
The above files is all I can share (all i have actually).
I uploaded:
- Raw Petrel data download from dctool
- Subsurface XML file (dive-shearwater-12-11-2017.ssrf)
- Shearwater XML file (dive-subsurface-12-11-2017.xml)

I thought that the Shearwater XML file has all the sensor samplings inside. i realized I wrongly named the shearwater and subsurface files. SSRF is a subsurface file. XML is a Shearwater file.

Sorry

atdotde, 2017-12-22:
I see. Will play with it over the holidays.

Bocio, 2017-12-22:
Thank you Robert

glance-, 2017-12-22:
Jef has already done the linear regression:
```If I do a simple linear regression (on your dive.0001.bin data)
between the millivolt values and the stored avgppo2, then I find these
calibration values for each sensor:

Sensor 0: 1861
Sensor 1: 1765
Sensor 2: 1874```

atdotde, 2017-12-22:
@glance- yes this is the plan. Also this dive shows that one sensor differs from the other two versus the end of the dive. Thus there might be some voting logic involved as well.

atdotde, 2017-12-22:
I spent an evening looking at this data with mathematica. Here is a PDF of that notebook. 
[shearwaterregresseion.pdf](https://github.com/Subsurface-divelog/subsurface/files/1583675/shearwaterregresseion.pdf) The .nb file is also available on request.

In the end I agree with Jef's regression values for the calibrations. We need to understand how to read those from the dive computer download (they must be in there as the Shearwater software is using them)

Bocio, 2017-12-23:
No Robert.

We already discussed it.
No sensor was voted out (sensor #2). I had no alarm on my display nor on
Shearwater Desktop.
A sensor voted out it's not something you don't notice or care about it.

Bye

PS
I will upload other dives with the same problem.

davide@mobile

glance-, 2017-12-25:
> In the end I agree with Jef's regression values for the calibrations. We need to understand how to read those from the dive computer download (they must be in there as the Shearwater software is using them)

I'm not sure they are in there. The Shearwater software never displays individual ppO2 value, only mV values and the stored average ppO2.

My money are on that they are stored in the sensor board, and it transmits ppO2 and mV values to the computer, and the computer does the voting/averaging and stores the raw mV values and the average ppO2.

atdotde, 2017-12-25:
Oh. I was under the impression that we only get the mV values for the sensors and that the average pO2 comes only from the Shearwater program not from the dive computer. @Bocio, does the Shearwater software show pO2 measurements of individual sensors?

Bocio, 2018-05-24:
Hi guys,
After more than one year I realize I never replied to @atdotde question. It was Xmas time and I missed it.

I will try to revive this old bug which is still opened in a different way. I will copy paste some discussions from Subsurface mailing list.

@jefdriesen wrote:

> Sorry for the lack of response, but for the last few weeks, I was just 
> too busy with non-libdivecomputer related things.
> 
> Anyway, the problem is as follows. The shearwater devices record two 
> different things:
> 
>     1. The average/voted ppO2.
> 
>     2. The value from each O2 sensor. Unfortunately this value is not the 
> ppO2 value, but the raw millivolt measurement from the sensor. In order 
> to convert this value to a ppO2 value, we need to take into account the 
> calibration values.
> 
> Last time I checked, shearwater desktop only shows the average/voted 
> ppO2 (#1). It doesn't show the individual ppO2 from each sensor (#2) at 
> all (e.g. no millivolt values nor the converted ppO2).
> 
> Now, originally libdivecomputer only reported the average/vote ppO2. 
> Later on we figured out how the sensor calibration worked, and this was 
> replaced with the ppO2 value from the individual sensors. But afterwards 
> we also discovered that some devices (like the one from Davide) don't 
> seem to store the calibration values correctly, and leave them at their 
> default values (2100). I have no idea why this happens. Anyway, because 
> applying those calibration values produces incorrect ppO2 values, I 
> added some code to detect this case and disable the ppO2 from the 
> sensors. Since the average/voted ppO2 was already disabled earlier, that 
> also means you won't get any ppO2 values at all anymore.
> 
> For those bogus default calibration values, there is not much we can do 
> about that. Without the (correct) calibration values, we simply can't 
> calculate the ppO2 values. So unless the calibration values are for some 
> reason stored elsewhere, and we just don't know about this, we're simply 
> stuck here. Shearwater desktop doesn't show this info, so it's difficult 
> to tell what's going on.
> 
> The only part that we can do something about, is restoring the 
> average/voted ppO2 again. That's already on my todo list. But right now 
> the libdivecomputer api doesn't have a way to indicate the type of the 
> ppO2 value (e.g. average/voted or from an individual sensor), so that 
> would cause confusing. So this will require some (backwards 
> incompatible) changes, and that's why it's not done yet.

The relevant part is:

Now, originally libdivecomputer only reported the average/vote ppO2. 
> Later on we figured out how the sensor calibration worked, and this was 
> replaced with the ppO2 value from the individual sensors. But afterwards 
> we also discovered that some devices (like the one from Davide) don't 
> seem to store the calibration values correctly, and leave them at their 
> default values (2100). I have no idea why this happens. Anyway, because 
> applying those calibration values produces incorrect ppO2 values, I 
> added some code to detect this case and disable the ppO2 from the 
> sensors. Since the average/voted ppO2 was already disabled earlier, that 
> also means you won't get any ppO2 values at all anymore.

So now, in some cases, we have no pO2 samples anymore on Subsurface CCR profile.
This happens when into the libdivecomputer download log we find:

> WARNING: Disabled all O2 sensors due to a default calibration value. [in /home/travis/build/Subsurface-divelog/subsurface/libdivecomputer/src/shearwater_predator_parser.c:503 (shearwater_predator_parser_cache)]

Current Shearwater desktop profile traces individual sensor mV values and average pO2:

[image]
[... 24 lines cut ...]
-                               if (callback && (parser->calibrated & 0x01)) callback (DC_SAMPLE_PPO2, sample, userdata);
-
-                               sample.ppo2 = data[offset + 14] * parser->calibration[1];
-                               if (callback && (parser->calibrated & 0x02)) callback (DC_SAMPLE_PPO2, sample, userdata);
-
-                               sample.ppo2 = data[offset + 15] * parser->calibration[2];
-                               if (callback && (parser->calibrated & 0x04)) callback (DC_SAMPLE_PPO2, sample, userdata);
-#endif
+                               if (!parser->calibrated & 0x07) {
+                                       sample.ppo2 = data[offset + 6] / 100.0;
+                                       if (callback) callback (DC_SAMPLE_PPO2, sample, userdata);
+                               } else {
+                                       sample.ppo2 = data[offset + 12] * parser->calibration[0];
+                                       if (callback && (parser->calibrated & 0x01)) callback (DC_SAMPLE_PPO2, sample, userdata);
+
+                                       sample.ppo2 = data[offset + 14] * parser->calibration[1];
+                                       if (callback && (parser->calibrated & 0x02)) callback (DC_SAMPLE_PPO2, sample, userdata);
+
+                                       sample.ppo2 = data[offset + 15] * parser->calibration[2];
+                                       if (callback && (parser->calibrated & 0x04)) callback (DC_SAMPLE_PPO2, sample, userdata);
+                               }
                        }

                        // Setpoint
```

Bocio, 2018-06-11:
I tried to temporary overcome the problem downloading dives with Shearwater desktop and then importing them in Subsurface via XML.
Even in this case there are errors: 

- no pO2 samples
- average pO2 completely wrong: graph reports a bottom pO2 of 1,92 bar

[image]
[image]

Patrick-BS, 2025-05-02:
Hi Guys, since today i used the Android app from the playstore. 

Now i switched to the apk file from the website and i have this issue too. Before, there was no individual sensor data displayed, but the average o2 was correct.  Is this issue really still open? Appversion: 6.0.5365.

mikeller, 2025-05-03:
> Hi Guys, since today i used the Android app from the playstore. Now i switched to the apk file from the website and i have this issue too.

Yes, the Play store version is old and outdated, we are not publishing it there any more due to the massive amount of maintenance work required to keep up with the ever changing version / compliance requirements.

> Update: Logdata from Subsurface XML shows, that the dc suppield its own calculated value. Is it possible to switch to this one for deco calculation etc?

Yes, this is intentional, Subsurface uses the raw data in the log as much as possible to do its own decompression calculation. You can already get the output of the dive computer's decompression calculation in the form of the dive computer supplied deco ceiling.
As your example shows, Shearwater computers - which are currently the only ones where we are able to extract the dive computer calculated 'ppO2' value - seem to be using a quite strong low pass filter to smooth out the ppO2. This explains the considerable difference to the actual measured sensor values. This may make sense for the live display of the expected time to surface on the dive computer, as it stops it from jumping all over the place as the ppO2 changes, but it is not ideal when the aim is to get an accurate idea of the tissue pressures post dive.
What is your aim when you are asking for the ppO2 values supplied by the dive computer to be used?

Patrick-BS, 2025-05-03:
> What is your aim when you are asking for the ppO2 values supplied by the dive computer to be used?
I just want to see the correct data in the profile, because the ppO2's calculated by Subsurface are wrong. Maybe because of incorrect calibration data.

Explanation:
The dive profile we see here is from my CCR1 course. So we did some drills, and at the end, the task was to push the o2 @6m to 1.6ppO2. Nobody got byond 1.50 and the learning was, that, as you are only "adding" o2, you can not really achieve 100%. Even doing an O2-fush, the loop still will contain some N2 (additional you are off-gassing).

Looking at the export with the old Subsurface version, this is perfectly visible, scraping 1,5 at the end:
[image]

Now, downloaded with the new version of the app this is very different, achieving impossible values >1,6ppO2:
[image]

Disabling the 3 sensors, showing only average, does not help:
[image]

The difference seems, that in the old import, the dc data was used:
[image]

and in the new one, it is the sensordata, conveted by subsurcae, with some calibration values:
[image]

Here is the original calibration data.
[image]

And i think here is key. Maybe some numerical issue, converting from O2-millivolts to [bar/V]?

Patrick-BS, 2025-05-03:
Addition:
If i convert the calibration values manualle: **1 / 0,052** I get **=19.23** not 21.00.

jefdriesen, 2025-05-03:
The sensor calibration values in your dives are all set to the default values (2100), which means the calibration data isn't available and thus we can't convert the millivolt values to ppO2 values. Subsurface ignores that and uses those calibration values anyway, and therefore you end up with ppO2 values that are incorrect. The code to handle this was removed with commit https://github.com/subsurface/libdc/commit/6e65428fc83ddd29ce1bc8b6abea6b7c2a1042c5.

Patrick-BS, 2025-05-03:
So my first question
>Is this issue really still open?
is answered with "no - but yes, again" xD

mikeller, 2025-05-03:
@Patrick-BS:

Is the device that you are pulling the log from a 'normal' Shearwater dive computer, or is this a Shearwater CCR controller that is connected to a CCR over 'DiveCAN'?

Patrick-BS, 2025-05-03:
@mikeller 
Indeed, it is a JJ CCR Controller. Shearwater Petrel 3 connected via Dive-CAN.

mikeller, 2025-05-03:
@Patrick-BS:

Ah, that makes sense then.
Unfortunately there does not seem to be a good way to get all of the information that CCR users would like to get from Shearwater 'DiveCAN' controllers - it looks like the controller does not have the sensor calibration information as the calibration is done in the CCR.
For this reason, libdivecomputer would throw away all sensor information  for these devices and randomly for some other Shearwater devices, based on how they were calibrated, thus making it impossible for owners of these controllers to use dive logs to get an idea of the condition of their sensors. This was broken in https://github.com/subsurface/libdc/commit/3e36cad5eeab3b3e96f1b522ac95928673408f0b, and it was reported as a problem by users and then fixed in Subsurface in https://github.com/subsurface/libdc/pull/59.
The behaviour that you are seeing with older versions of Subsurface showing the dive computer reported, smoothed ppO2 value is caused by a different, not really related bug that caused Subsurface to incorrectly import the dive computer reported ppO2 value on Shearwater dive computes as the first 'sensor' value. This resulted in all users of non-'DiveCAN' Shearwater dive computers seeing a mix of this ppO2 and the actual ppO2 reported by the first 2 sensors in their dive logs. This was fixed in https://github.com/subsurface/subsurface/pull/4419.
Unfortunately it does not help that there does not seem to be a way to differentiate between Shearwater dive computers and Shearwater 'DiveCAN' CCR controllers based on the log data downloaded from them. But I think that a good solution for this will meet both of these user requirements:
- show individual sensor values to allow insights into the condition / functioning of the sensors;
- show an accurate ppO2.

Patrick-BS, 2025-05-03:
Maybe a setting in the tech-settings?

 "Always use dc po2 values as averedged po2 instead of the ones, calculated by subsurface"

mikeller, 2025-05-03:
> Maybe a setting in the tech-settings?
> 
> "Always use dc po2 values as averedged po2 instead of the ones, calculated by subsurface"

And how will users with this setting enabled be able to see how their sensors behave, and if there is a problem with their sensors - like a sensor being voted out for half of the dive because its value is too far away from the average?

Also, it has to be noted that at the moment only Shearwater dive computers are reporting a smoothed out theoretical 'ppO2 value' - so this will not work as expected for most users.

Patrick-BS, 2025-05-03:
Why not show the sensors simultaneously?  All Sensors + Average

And the average is calculated by subsurface or taken from dc.

mikeller, 2025-05-03:
> Why not show the sensors simultaneously? All Sensors + Average
> 
> And the average is calculated by subsurface or taken from dc.

That's an idea. I'll ponder this.

mikeller, 2025-05-20:
@Patrick-BS: Can you please grab one of the versions linked in https://github.com/subsurface/subsurface/pull/4524, and check if this imports the ppO2 with the correct scaling for you?

Patrick-BS, 2025-05-21:
Looks good! 
Thoughts:
-Would be nice to se the adjusted calibartion values in the Additional Data
-At some dives, it looks a tiny little bit too low, though

Here the updated profile of the same dive as[ here.](https://github.com/subsurface/subsurface/issues/971#issuecomment-2848570323)

[image]

Patrick-BS, 2025-05-21:
Here for example. The DC values are much more floating around 0.7, whereas the calculated sensorvalues stay below 0.7 mostly.
But maybe i am too picky :)

[image]

mikeller, 2025-05-21:
Hi @Patrick-BS, thanks for testing!

> Looks good! Thoughts:
> -Would be nice to se the adjusted calibration values in the additional data tab

It should show up there as 'Estimated DiveCAN calibration [bar / V]' - it's going to be the same value for all sensors.

> -At some dives, it looks a tiny little bit too low, though

Yes, this is always only going to be an estimate of the actual calibration, as there are other factors like smoothing of the dive computer calculated ppO2 that we don't know. And for practicality's sake the estimate is only done based on the first sample of the dive that has ppO2 information.

> Here for example. The DC values are much more floating around 0.7, whereas the calculated sensorvalues stay below 0.7 mostly. But maybe i am too picky :)

I think you are. 😉
This looks a lot like the dive computer calculated values have been smoothed.

Patrick-BS, 2025-05-22:
> It should show up there as 'Estimated DiveCAN calibration [bar / V]' - it's going to be the same value for all sensors.

Unforunately not:

[image]

mikeller, 2025-06-01:
@Patrick-BS: Can I get you to try again with the latest artifacts on #4524? After some more experimentation I have moved to using the average deviation between sensor and dive computer calculated values over the duration of the dive to calculate the estimate for the calibration - only using the first value of the dive resulted in incorrect values for some of the dives on my non-DiveCAN Shearwater Petrel that I analysed.
The Extra Info data is now showing as well.

Patrick-BS, 2025-06-06:
@mikeller tested #4524 from 3 days ago. Worked fine, guessed calibration seems to be more accurate and resulting estmate value is shown 👍

mikeller, 2025-06-06:
Fixed by https://github.com/subsurface/libdc/pull/81.

closed 2025-06-06 by mikeller