subsurface/subsurface #978: 4.7.5 just updated and map window no longer shows up

sikgerar, 2017-12-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I just updated to 4.7.5.
the map window does no longer show anything (grey filling).

### Operating system:

xubuntu 
uname -a
mycomputer 4.4.0-101-generic #124-Ubuntu SMP Fri Nov 10 18:29:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

### Subsurface version:

i used to run 4.6
I just updated to  4.7.5.

### Steps to reproduce:

update to 4.7.5
select view window-> all

### Current behavior:

no longer have a map window with a map to pick up locations or see locations

### Expected behavior:

see map

### Additional information:

### Mentions:

janmulder, 2017-12-24:
Can you run run subsurface from the console using command line option -v. That way we might see what is going on.

sikgerar, 2017-12-24:
its probably a module qtquick installed, based on the log.

```
Subsurface v4.7.5,
built with libdivecomputer v0.6.0-devel-Subsurface-branch (db70c581a621814feb881494c890ae8a49da709a)
built with Qt Version 5.5.1, runtime from Qt Version 5.5.1
built with libgit2 0.26.0
validateGL(): created OpenGLContext.
validateGL(): obtained QOpenGLFunctions.
validateGL(): detected OpenGL version 3.0.
added supported DC:  Suunto   Solution
added supported DC:  Suunto   Eon
added supported DC:  Suunto   Solution Alpha
added supported DC:  Suunto   Solution Nitrox
added supported DC:  Suunto   Spyder
added supported DC:  Suunto   Stinger
added supported DC:  Suunto   Mosquito
added supported DC:  Suunto   D3
added supported DC:  Suunto   Vyper
added supported DC:  Suunto   Vytec
added supported DC:  Suunto   Cobra
added supported DC:  Suunto   Gekko
added supported DC:  Suunto   Zoop
added supported DC:  Suunto   Vyper 2
added supported DC:  Suunto   Cobra 2
added supported DC:  Suunto   Vyper Air
added supported DC:  Suunto   Cobra 3
added supported DC:  Suunto   HelO2
added supported DC:  Suunto   D9
added supported DC:  Suunto   D6
added supported DC:  Suunto   D4
added supported DC:  Suunto   D4i
added supported DC:  Suunto   D6i
added supported DC:  Suunto   D9tx
added supported DC:  Suunto   DX
added supported DC:  Suunto   Vyper Novo
added supported DC:  Suunto   Zoop Novo
added supported DC:  Suunto   D4f
added supported DC:  Suunto   EON Steel
added supported DC:  Suunto   EON Core
added supported DC:  Uwatec   Aladin Air Twin
added supported DC:  Uwatec   Aladin Sport Plus
added supported DC:  Uwatec   Aladin Pro
added supported DC:  Uwatec   Aladin Air Z
added supported DC:  Uwatec   Aladin Air Z O2
added supported DC:  Uwatec   Aladin Air Z Nitrox
added supported DC:  Uwatec   Aladin Pro Ultra
added supported DC:  Uwatec   Memomouse
added supported DC:  Uwatec   Smart Pro
added supported DC:  Uwatec   Galileo Sol
added supported DC:  Uwatec   Galileo Luna
added supported DC:  Uwatec   Galileo Terra
added supported DC:  Uwatec   Aladin Tec
added supported DC:  Uwatec   Aladin Prime
added supported DC:  Uwatec   Aladin Tec 2G
added supported DC:  Uwatec   Aladin 2G
added supported DC:  Subgear   XP-10
added supported DC:  Uwatec   Smart Com
added supported DC:  Uwatec   Aladin 2G
added supported DC:  Uwatec   Aladin Tec 3G
added supported DC:  Uwatec   Aladin Sport
added supported DC:  Subgear   XP-3G
added supported DC:  Uwatec   Smart Tec
added supported DC:  Uwatec   Galileo Trimix
added supported DC:  Uwatec   Smart Z
added supported DC:  Subgear   XP Air
added supported DC:  Scubapro   Meridian
added supported DC:  Scubapro   Mantis
added supported DC:  Scubapro   Chromis
added supported DC:  Scubapro   Mantis 2
[... 155 lines cut ...]
added supported DC:  Ratio   iX3M Pro Tech+
added supported DC:  Ratio   iDive Free
added supported DC:  Ratio   iDive Easy
added supported DC:  Ratio   iDive Deep
added supported DC:  Ratio   iDive Tech+
added supported DC:  Cochran   Commander TM
added supported DC:  Cochran   Commander I
added supported DC:  Cochran   Commander II
added supported DC:  Cochran   EMC-14
added supported DC:  Cochran   EMC-16
added supported DC:  Cochran   EMC-20H
Plugins Directory:  QDir( "/usr/bin" , nameFilters = { "*" },  QDir::SortFlags( Name | IgnoreCase ) , QDir::Filters( Dirs|Files|Drives|AllEntries ) )
qrc:/MapWidget.qml:2:1: module "QtQuick" version 2.6 is not installed
QUrl("qrc:/MapWidget.qml") failed to load with status: 3
qrc:/MapWidgetError.qml:2:1: module "QtQuick" version 2.6 is not installed
loading dive data from ("/home/gerardo/.subsurface/gerardo.xml")
Set the current dive site: 0
Set the current dive site: 0

File locations:
...
```

shouldnt it be installed by default?
I had it working on 4.6

dirkhh, 2017-12-24:
Interesting. Qt 5.5.1 for both build and runtime, so we have consistent Qt.
We may have revved the QtQuick version too far in f3d978b8a5fb6511aa0c1c611cf9b0394fdc125b @janmulder - If we want to support Qt 5.5 we can't go past QtQuick 2.5

janmulder, 2017-12-24:
Yes, that commit to 2.6 is highly likely the reason ... but that said 5.5.1 was released in October 2015 ... So the big question is here ... do we really want to support 5.5 any longer? I bet there is a massive amount of other things like Bluetooth not working there.

sikgerar, 2017-12-24:
so I guess it will be fixed on a next release?
should I jump back to an older release to have it working? 
I will probably not do it right now, can wait until it is fixed since I already updated all my ddbb until next summer, but I would like to see it working again in the future.

<offtopic>
btw, is there any plans to do some more integration with google maps on the android/iphone apps?
I would love to see the same map with the diving sites layer on the android app..
</offtopic>

dirkhh, 2017-12-24:
@sikgerar the maps in Subsurface 4.7.5 are different from 4.6 - that's one of the major changes.
Except that you don't see them :(
I don't know what you mean by "the same map with the diving sites layer" - can you explain?

@janmulder we have to be careful how aggressively we abandon distros. It's nearly impossible for normal users to update their Qt Version.

Yes, it would be much better if the users updated their distros, but...

closed 2017-12-24 by dirkhh

sikgerar, 2017-12-25:
thank you guys for the explanation. I will wait for the new version to get the maps working.

@dirkhh it would be great to have a button on the app to open a google maps/earth window with ALL the saved locations, as you can do on the computer, and not just one location (as it can be done already on the app).
I spoke about a "layer" as I though you would think on how google maps treats groups of locations. just forget about that.

sikgerar, 2018-01-08:
fixed on 

Subsurface v4.7.6,
built with libdivecomputer v0.7.0-devel-Subsurface-branch (8ae735a4d70307ebe2a42d315697f02ce71dbe88)
built with Qt Version 5.5.1, runtime from Qt Version 5.5.1
built with libgit2 0.26.0

thank you guys

PR #979 (2017-12-24): mapwidget: revert QtQuick version to 2.0
In commit f3d978b8a5fb6 the QtQuick version was upgraded to 2.6 (from 2.0 for the mapwidget). This is, apparently, too aggressive, as there are still mainstream distributions that are still on Qt 5.5.1 for which 2.6 of QtQuick is too new. And as a sidenote: Qt 5.5.1 was released in October 2015.

So partially revert commit f3d978b8a5fb6.

Fixes: #978
commit 4c4222d6 (2017-12-24): mapwidget: revert QtQuick version to 2.0

In commit f3d978b8a5fb6 the QtQuick version was upgraded
to 2.6 (from 2.0 for the mapwidget). This is, apparently,
too agressive, as there are still mainstream distributions
that are still on Qt 5.5.1 for which 2.6 of QtQuick is too
new. And as a sidenote: Qt 5.5.1 was released in October
2015.

So partially revert commit f3d978b8a5fb6.

Fixes: #978