subsurface/subsurface #997: Newest AppImage does not run on recent Arch Linux box

janmulder, 2017-12-29:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
The Appimage does not start and produces the following error message. 

relocation error: /usr/lib/libk5crypto.so.3: symbol k5_os_mutex_destroy, version krb5support_0_MIT not defined in file libkrb5support.so.0 with link time reference

### Operating system:
Up to date Arch Linux box

### Subsurface version:
current master

### Steps to reproduce:
run and see the relocation error.

### Mentions:
@dirkhh, @probonopd

probonopd, 2017-12-29:
Did it work with older versions @janmulder?

I'd not be surprised if no one would actually _use_ Kerberos with Subsurface (happy to be proven wrong!). Maybe switching to a build of lib curl without Kerberos support (as @TheAssassin had suggested earlier iirc) would entirely remove this type of issue - and save some overhead.

janmulder, 2017-12-29:
Yes, I have an AppImage sitting around with Subsurface version 4.7.2-77, and that one runs fine on my current Arch box.

TheAssassin, 2017-12-29:
See https://github.com/AppImage/AppImageUpdate/blob/rewrite/resources/bootstrap-container.sh#L8-L10 on how to use my custom libcurl build.

marccollin, 2017-12-30:
work fine under tumbleweed

probonopd, 2017-12-30:
Any idea what might cause it to load `/usr/lib/libk5crypto.so.3` from the system in the first place?

janmulder, 2017-12-31:
This is what lddtree says:

>     libgit2.so.26 => /usr/lib/libgit2.so.26
>         libcurl.so.4 => /usr/lib/libcurl.so.4
>             libnghttp2.so.14 => /usr/lib/libnghttp2.so.14
>             libidn2.so.0 => /usr/lib/libidn2.so.0
>                 libunistring.so.2 => /usr/lib/libunistring.so.2
>             libpsl.so.5 => /usr/lib/libpsl.so.5
>             libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2
>                 libkrb5support.so.0 => /usr/lib/libkrb5support.so.0
>                 libkeyutils.so.1 => /usr/lib/libkeyutils.so.1
>                 libresolv.so.2 => /usr/lib/libresolv.so.2
>             libkrb5.so.3 => /usr/lib/libkrb5.so.3
>             libk5crypto.so.3 => /usr/lib/libk5crypto.so.3
>             libcom_err.so.2 => /usr/lib/libcom_err.so.2
>         libssl.so.1.1 => /usr/lib/libssl.so.1.1
>         libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1
>         librt.so.1 => /usr/lib/librt.so.1
>

probonopd, 2017-12-31:
OK, so like @TheAssassin suggested, using a `libcurl.so.4` without the Kerberos dependency would be a suitable way to reduce the footprint and solve the issue.

glance-, 2017-12-31:
I like the packaging of AppImages, but I don't like the lack of abi management. I would suggest looking at building the whole lot into the AppImage, and if one is a bit cleaver about how the dependant libraries are built I don't think it will end up that big either.

Maybe steal the relevant system libraries from alpine linux or some other slim distro, or maybe some fancy buildroot integration and build the whole lot our self.

probonopd, 2017-12-31:
@glance- would the result look somewhat like the [Open Build Service](https://build.opensuse.org/)? In this case, no need to reinvent the wheel...

dirkhh, 2017-12-31:
This isn't hard to do on Travis, either. We just need to build libcurl from source there.

janmulder, 2017-12-31:
I just build my first own AppImage :-) and obviously, that runs fine on my box.

I used the `scripts/linux/travisbuild.sh` for that, and surprisingly, I had to fix something in there to get it running. 

```
-mv appdir/usr/usr/local/Qt*/plugins/* appdir/usr/plugins # the usr/usr is not a typo - that's where it ends up
+mv appdir/usr/usr/lib/qt*/plugins/* appdir/usr/plugins # the usr/usr is not a typo - that's where it ends up
```

dirkhh, 2017-12-31:
Because you have a system provided Qt, not an "aftermarket" Qt that installs in /usr/local/Qt*

janmulder, 2018-09-04:
Issue is solved (in current master and probably long before). Closing.

commit 8813f13b (2016-01-28): QML UI: end edit mode when switching to different page

Fixes #997
closed 2018-09-04 by janmulder