subsurface/subsurface #1002: desktop: Bluetooth download from DC fails in common use case

janmulder, 2017-12-31:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
For some time, I downloaded from mobile, but back home, I like to download from the DC using the desktop application again. This is currently not working correctly. Simply go to the Import from DC screen, and as I always download this way, my BT button is default selected and the device mount point correctly filled with the following string `OSTC3#05106 (00:80:25:4A:0F:C3)`. This nicely formatted string comes from commit de81effb258c9c55af299ef (@bstoeger). Now click download and it fails (always). First I thought it had something to do with a almost end-of-live battery in my OSTC.

There is a workaround, and that points immediately to the reason of this bug. When going to the "remote Bluetooth device selection" screen first, selecting the DC and save it, and then back at the download screen, the download succeeds. Or just manually type in the BT address without the name padding.

Further investigation shows that all this is caused by passing the nicely formatted devname to libdc, and that obviously fails. The workaround route sets the devname to the BT address without the name, so that succeeds.

### Operating system:
will fail on Linux, Apple, Windows. 

### Subsurface version:
latest master

### Steps to reproduce:
see description

### Current behavior:
see description

### Expected behavior:
see description

### Additional information:
commit de81effb258c9c55af299ef is related. And I believe this is a high prio issue, as it is a very common use case, and an average user using BT/BLE download will not find the workaround instinctively.

### Mentions:
@bstoeger

atdotde, 2017-12-31:
Could this be related to 1c223b09c310d65ca554cae455cdc74fc33da60f ? At the time I tested this on iOS and Mac but it could be breaking Linux (at least I didn't test).

janmulder, 2017-12-31:
Could be, not sure. Will try this by reverting.

bstoeger, 2017-12-31:
Give me a few minutes, I'm working on a fix.

janmulder, 2017-12-31:
@atdotde: unrelated to 1c223b0.

commit 520f505d (2016-02-03): Fix Liquivision import crash

When we detect a redundant DC we free the memory reserved for the model.
Thus we need to malloc that memory here.

Fixes #1002
commit 3c022d86 (2017-12-31): Get Bluetooth device nam from preferences if dialog wasn't shown

If the BT dialog hasn't been shown, the device name was taken from
the text field, which contained a formatted string. The device open
would then fail.

Fixes #1002
closed 2017-12-31 by dirkhh