subsurface/subsurface #1013: WebserviceID handling mobile vs. desktop

janmulder, 2018-01-04:
### Describe the issue:

- [X] Bug (maybe)
- [ ] Change request
- [ ] New feature request
- [X] Discussion request

### Issue long description:
The handling of the WebserviceID is different in the mobile app and the desktop. On mobile, the WebserviceID is linked to the cloud email address/username, and there is no capability to enter it manually (at least in the UI). When switching cloud account (ok, not a very normal use case, but development and testing, pretty common for me), the WebserviceID is pulled from the server based on the email address. On the desktop, however, there is a UI to enter the WebserviceID, and a button to ask for local storage. But what I do not like is that the WebserviceID is **not** pulled from the server at cloud account switch.

I'm writing this issue, as there is more to this. We still have the GPS only logger (companion app) around that does have the WebserviceID visible/editable in the UI. Further, I'm not sure that all WebserviceID's are tied to an email address, as the WebserviceID was there before the cloud storage. So Apple users might have WebserviceID's that are not tied to email/cloud account (as they are forced to use the companion app).

While fixing this on the desktop end is not that complex (automatic population on the WebserviceID based on email, like is done on mobile), but unsure that this the way on. 

### Mentions:
@dirkhh

dirkhh, 2018-01-04:
I thought I had added the code to the desktop to grab the ID from the backend based on the cloud storage email if the ID hasn't been entered. I did not check this in the code, so I may imagine this.
Once we are comfortable that the GPS tracking in Subsurface-mobile works well enough in both iOS and Android, we should simply drop the older apps from the respective stores and get rid of the confusion. And permanently hide the WebserviceID from the user

janmulder, 2018-01-06:
@dirkhh: yes, that code to grab the ID from the backend based on the cloud storage email if the ID hasn't been entered was in the the desktop code. But it did not get triggered as I had an ID of one of my test cloud accounts filled in in preferences. And as I never asked to store that ID locally, I did not think of removing it there first (as I did not realize it came from prefs).

PR #1020 (2018-01-06): Do not save WebserviceID in preferences until asked for
Do not save the WebserviceID in the preferences until we ask for this. That is simply wrong.

This is definitely not a full fix for all the weirdness that is going in related to the WebserviceID but saving the ID even if we do not ask for it, adds to possible confusion.

To always automatically pull in the ID from the server based on cloud credentials, just leave the ID field empty in the network preferences.

Fixes: #1013 (well ... a tiny part of this mess)
commit 97127e16 (2018-01-06): Do not save WebserviceID in preferences until asked for

Do not save the WebserviceID in the preferences until we ask
for this. That is simply wrong.

This is definitely not a full fix for all the weirdness that is
going in related to the WebserviceID but saving the ID even
if we do not ask for it, adds to possible confusion.

To always automatically pull in the ID from the server based
on cloud credentials, just leave the ID field empty in the
network preferences.

Fixes: #1013 (well ... a tiny part of this mess)
closed 2018-01-06 by dirkhh