subsurface/subsurface #1031: [FEATURE REQUEST] dive, show, plan modes in the APP

sikgerar, 2018-01-08:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

The app is cool as it can be used while diving and as a showroom for friends. 
however, there are functions with no use in this case.
there is no way to differentiate between modes and options keep messing it up visually.
the solution would be to add, on the App, differentiated modes for "dive", "show", or "plan".

### Operating system:

App, android or iphone.

### Subsurface version:

N/A

### Steps to reproduce:

N/A

### Current behavior:

N/A

### Expected behavior:

some options should be accessible only in specific modes.
ie: gps functions should be available in "dive" mode. 
however, a map with the dive locations should only be available in "show" mode.
planning of a dive could (maybe) done in "plan" mode, without messing around with the easy-to-use gui the app has right now.

### Additional information:

N/A

### Mentions:

N/A

janmulder, 2018-01-08:
Let me summarize:
1) map, like in the old companion app (showing all dive sites with a gps marker) => maybe not to difficult to implement any more ... @neolit123, what is your take on this?
2) mode based UI for mobile => well as an open source project, feel free to send in patches for this, I do not see the current interface as cluttered or nonlogical, so no support from my side on this.

sikgerar, 2018-01-08:
@janmulder 
about 2.- it is not that it is nonlogical, but it would be simpler to have a "dive" mode where only a set of options are available, separated from a "show" mode where map, pictures and stuff can be accessed.

I am not a coder myself, just a subsurface user, so I am just proposing what I feel could be good stuff missing from the app/software. sorry for not being able to help more..

neolit123, 2018-01-10:
@janmulder @sikgerar
i don't understand the feature request and 1).
having a inline map widget for the mobile version of Subsurface is planned but currently not available.

janmulder, 2018-01-10:
@neolit123. you understand 1) correctly. OP asks (among others) for a map in the mobile app. Ie. more that the current link for 1 marker to the google maps. But some overview, like we had in the old companion app.

And tempted to close this issue, as 1) is on your list, and 2) is a no-go according to me.

janmulder, 2018-01-10:
See #1037.

closed 2018-01-10 by janmulder

janmulder, 2018-01-10:
rant-alert: As long as almost no developers are working on mobile, we (or better I), try to get answers from the Kirigami guys, that do not get answered, PR's for Kirigami are sitting for 6 weeks idle, or get merged under different user, and finally that programming in Kirigami/QML is more a fine art instead of a science/skill, I tend not to accept vague and random issues from random github uses.

commit abb3b035 (2016-03-23): Don't start Add or Plan while accessing cloud storage

This avoids a race condition where we get confused about our internal
state.

Fixes #1031
commit c332bf2c (2016-03-23): Don't start Add or Plan while accessing cloud storage

This avoids a race condition where we get confused about our internal
state.

Fixes #1031