subsurface/subsurface #1041: Git storage progress bar text does not fit widget

atdotde, 2018-01-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

A QProgressBar widget is supposed to grow automatically to fit the text label it contains.
Except it doesn't. This becomes more obvious when running Subsurface in German where
the translated strings are longer but it is present in the English version as well. I tried fix this by turning setLabelText into setLabel but that did not help.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2018-01-12:
@atdotde 

we have a list of `QProgressBar` usages in the tree, but those are fixed in a layout and don't seem to cause troubles:

```
..desktop-widgets> grep QProgress * -rnI
configuredivecomputerdialog.cpp:15:#include <QProgressDialog>
configuredivecomputerdialog.cpp:348:    QProgressDialog *dialog = new QProgressD
ialog("Updating firmware", "", 0, 100);
configuredivecomputerdialog.ui:3948:    <widget class="QProgressBar" name="progr
essBar">
diveshareexportdialog.ui:186:       <widget class="QProgressBar" name="progressB
ar">
downloadfromdivecomputer.ui:186:        <widget class="QProgressBar" name="progr
essBar">
mainwindow.cpp:69:QProgressDialog *progressDialog = NULL;
mainwindow.cpp:2028:    progressDialog = new QProgressDialog(tr("Contacting clou
d service..."), tr("Cancel"), 0, 100, this);
mainwindow.h:15:#include <QProgressDialog>
printdialog.cpp:7:#include <QProgressBar>
printdialog.cpp:94:     progressBar = new QProgressBar();
printdialog.h:12:class QProgressBar;
printdialog.h:26:       QProgressBar *progressBar;
webservices.ui:59:    <widget class="QProgressBar" name="progressBar">
```

instead of `QProgressBar` do you mean the `QProgressDialog` in `mainwindow.cpp`?

edit: oops, i guess the title partially confirms that `Git storage progress bar text does not fit widget`.

neolit123, 2018-01-12:
@atdotde on what OS you are reproducing the issue? on Windows 7 i get a correct resize:

[image]

janmulder, 2018-01-12:
Things also seem to work (even in German :-)) on my Linux box. That said, as my network is fast, I have definitely no time to read the messages, but can see that they are not truncated, and resize between the different messages.

atdotde, 2018-01-12:
I noticed this today on my office iMac with the big screen where at some point I had increased the font size in the preferences. Now, on my laptop, everything did fit in the box but there the font is much smaller. Thus I artificially increased the font size in the preferences, and here you go: 
[image]

janmulder, 2018-01-12:
Well, not sure this "artificially increased" trick shows the issue. That whole screen looks like shit, not only the dialog we are talking about here.

atdotde, 2018-01-12:
No, that was just coming from an ugly font that I had selected:

[image]

janmulder, 2018-01-13:
Yes, that convinces me more ... but as we not seem to be able to reproduce on Windows or Linux I am very tempted to call it a bug outside of our code base. Could be Qt or even Os/windowmanager related. 

As it seem to be centered correctly, it could be something related to font metrics. Hmm, that said, we do mess around with font metrics ... and just found the intriguing function:

```
QFont defaultModelFont()
{
	QFont font;
//	font.setPointSizeF(font.pointSizeF() * 0.8);
	return font;
}
```

But do they resize at all? Or are they to small for every string?

neolit123, 2018-01-13:
@atdotde 

what Qt version?
are you building Subsurface yourself on that mac?

here is a patch to try:
```
diff --git a/desktop-widgets/mainwindow.cpp b/desktop-widgets/mainwindow.cpp
index 5cf3dc27f..b0220402d 100644
--- a/desktop-widgets/mainwindow.cpp
+++ b/desktop-widgets/mainwindow.cpp
@@ -13,6 +13,7 @@
 #include <QShortcut>
 #include <QToolBar>
 #include <QStatusBar>
+#include <QFontMetrics>
 
 #include "core/version.h"
 #include "desktop-widgets/divelistview.h"
@@ -77,6 +78,8 @@ extern "C" int updateProgress(const char *text)
 		qDebug() << "git storage:" << text;
 	if (progressDialog) {
 		progressDialog->setLabelText(text);
+		const int width = QFontMetrics(progressDialog->font()).width(progressDialog->labelText()) + 100;
+		progressDialog->resize(width, progressDialog->height());
 		progressDialog->setValue(++progressCounter);
 		if (progressCounter == 100)
 			progressCounter = 0; // yes this is silly, but we really don't know how long it will take

```

works for me on Windows. it does no harm if the auto-resize works, but it could take care of the non-resizing issue.

atdotde, 2018-01-13:
It seems to actually be a Qt problem, I can reproduce this in a a small example program: 
[image]

Could you please try this on Linux to see if this is a Mac only problem?  
[progresdialogoverlap.zip](https://github.com/Subsurface-divelog/subsurface/files/1628458/progresdialogoverlap.zip)

neolit123, 2018-01-13:
@atdotde @janmulder said that he cannot reproduce it on Windows and Linux. is this on Qt 5.9+ ?

also, we need to test the above patch.

atdotde, 2018-01-13:
@neolit123 we overlapped. I came up with the same workaround as #1045 which solves the problem for me. The problem exists both for off the shelf 5.8 and 5.10.0. In my question about other OSs I was referring to my example program.

PR #1045 (2018-01-13): Resize progress bar width to fit all text
At least on Mac with larger font sizes part of the label
text of the git access progress bar is cut off (even though
it should automatically resize). This patch adds explicit
resize.

Fixes #1041

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:
commit de49f248 (2018-01-13): Resize progress bar width to fit all text

At least on Mac with larger font sizes part of the label
text of the git access progress bar is cut off (even though
it should automatically resize). This patch adds explicit
resize.

Fixes #1041
closed 2018-01-13 by neolit123