subsurface/subsurface #1046: CI: Linux build fails pulling linuxdeployqt

neolit123, 2018-01-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The CI Linux build fails at this step:
`wget -q -c https://github.com/probonopd/linuxdeployqt/releases/download/continuous/linuxdeployqt-continuous-x86_64.AppImage`

the file doesn't seem to exists anymore.

https://travis-ci.org/Subsurface-divelog/subsurface/jobs/328484842

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@probonopd @dirkhh

probonopd, 2018-01-13:
We are on it. cc @TheAssassin

TheAssassin, 2018-01-13:
There's quite a bunch of issues within the build process of linuxdeployqt leading to situations like this. Working on it. Documenting all the issues at the moment, and the list is getting longer...

neolit123, 2018-01-13:
maybe we should stay on a `linuxdeployqt-continuous-x86_64.AppImage` that works for us and only update when necessary, @dirkhh ?

TheAssassin, 2018-01-13:
We'll be working on our deployment infrastructure. There was a bug in another tool, uploadtool, which has been fixed, to prevent faulty builds from deleting the release on GitHub in the future. Also we're thinking of keeping archives of our builds etc. to quickly be able to restore binaries if we have to. But if you have further suggestions, feel free to report them to us in the linuxdeployqt repository. Thanks.

neolit123, 2018-01-13:
thanks for the fix

closed 2018-01-13 by neolit123