subsurface/subsurface #1048: Forbidden characters in cloud password

janmulder, 2018-01-15:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Issue just not for forget this. Fill in a password in the preferences with a forbidden character.  A red error is shown, but the password is saved in the preferences anyway.

### Operating system:
most likely all desktop (mobile not tested)

### Subsurface version:
master

### Current behavior:
See above

### Expected behavior:
When error, do not save the change. Further, the list of allowed characters for the cloud password is very restrictive. Not sure this has a reason relates to the official cloud server (@dirkhh ?)

mturkia, 2018-01-15:
IIRC this restriction was set due to some characters being encoded (or not) during transit, and messing up the authentication. Rather than looking more deeply into the authentication or dealing with individual user reports, this restriction was put in place as a quick fix.

dirkhh, 2018-01-16:
Yes, there were numerous problems with libraries trying to be clever and quote some characters and not others which lead to all kinds of weird problems. Since back then I was the only one working on all this I made my life simple by allowing only a restrictive set of characters. Of course all the research shows that adding crap to your passwords doesn't make them any safer, and they become impossible to type in on mobile... so I think in hindsight this wasn't a bad decision, either.
What is bad is what you describe here - rejecting a password but still storing it.

closed 2018-01-18 by janmulder

sveeke, 2018-10-31:
> Yes, there were numerous problems with libraries trying to be clever and quote some characters and not others which lead to all kinds of weird problems. Since back then I was the only one working on [...]

I actually disagree with you on this. "Adding crap to you passwords" (== more entropy) does indeed make passwords stronger. I don't know where you have read that the opposite is shown by all research, because it doesn't even make sense. While not a complete source, you may find "Random passwords" in https://en.wikipedia.org/wiki/Password_strength an interesting read.

Maybe you mean that the salted hashed credentials in the server's database are all equally secure in the end? Maybe so, but with a strong algorithm and salt there shouldn't be a problem, even if the database gets hacked.

Then my reason for posting here. I just started diving (I got my Open Water a couple of days ago) and was searching for a dive log application. I remembered that Linus also is (was?) involved with a dive application, and as a long time user of open source software I wanted to use this.

Normally I don't use cloud services (I host everything myself), but in this case I thought: why not. But then I couldn't use any of my standard passwords (no memorized ones, but also none of my password manager) since the app has this weird password policy of not allowing all characters in passwords.

Of course I will manage by making another profile in my password manager specifically for Subsurface, but I think it's not best practice to limit a user's ability to choose their password. Users can decide for themselves if they find certain passwords "impossible to type" or what length and what characters they should use. I don't think us IT people should limit that, especially not when more and more people become aware of the need for strong(er) passwords and get taught practices/behaviour which may consist of special characters.

Well I think I babbled long enough about this. I hope it will be possible to use special characters in passwords. By the way, this might be the wrong place for this 'request' since this issue is more about saving the password while showing an error. If I need to make a separate issue, let me know :).

dirkhh, 2018-10-31:
I always wonder if it is worth to respond to comments like this...
On the question whether special characters add to password security there has indeed been quite a bit of research. A very readable summary is here https://resources.infosecinstitute.com/password-security-complexity-vs-length 
Yes, per character (in a fixed length world) entropy goes up (but only slightly - we allow 66 different characters... adding 10 or 15 additional characters really doesn't make a substantial difference). But in the end, length always trumps character set.
But you clearly didn't come here to discuss actual password security (given that you say

> I couldn't use any of my standard passwords

which indicates that security never was your issue).
Teaching people to add special characters to password is at best misleading. Teaching people to never reuse passwords and to use longer passwords is much better.

What does all this mean for Subsurface?
Well, this is an open source project that offers a free cloud service (which means free to you, I actually have to pay for the server and the bandwidth and the storage) to its users. We (the developers) and I (the maintainer of both the project overall and that cloud server) prefer to spend our time doing things that are fun (like diving, or writing code), not answering support emails. Mis-typed passwords are actually a rather substantial amount of the support emails that I get. And that's without allowing special characters. I assume it would be worse if I allowed them.

So... no, this will not change. What we have works. You are of course free not to use it.

dirkhh, 2018-10-31:
> Supporting ASCII (+~768 bits per char) or even extended ASCII (+~1800 bits per char) would mean a substantial increase in entropy for each character used.

I mean no disrespect, but I think statements like this make it pretty clear that there are some misunderstandings here. Which gets me back to why I hesitate to engage in these discussions. Sure, I can argue about the mathematical definition of entropy and what it all means with you. But what's the point? No, there are no 1800 bits of entropy in a single character of "extended ASCII". But even if I were to get you to understand that, it wouldn't change any of the rest of the issues we simply disagree about.

I'll disengage here. Final comment: backhanded statements like "I value open source software more than all the shortcomings of Subsurface" are a good reminder why I shouldn't have responded in the first place. I whole-heartedly invite you not to use Subsurface and not to subject yourself to the shortcomings of our software. I'll help you finding a different, better package which lives up to your expectation by disabling your cloud account.

atdotde, 2018-10-31:
> We (the developers) and I (the maintainer of both the project overall and that cloud server) prefer to spend our time doing things that are fun (like diving, or writing code), not answering support [...]
Fair point. Although there may be no relation between the amount of supported characters and an increase in support mail, I understand your position.

I think we had more characters in the past and those caused problems as various instances between client and server caused enough confusion (like quoting or having different characters under the " key) meaning users that couldn't get to their data and were thus frustrated and complained that it was decided it's not worth the hassle and restrict ourselves to characters that are not ambiguous. 

That said, the Subsurface cloud storage is git under the hood. In the manual, it is described how you can set up a local directory as git repository (or by running subsurface with -v option figure out where it keeps its own) and then manually sync with your own remote git repository (that you can secure in any way you like i.e. with a retina scan and a one-time pad). It just doesn't work with the mobile app that has the subsurface cloud hard coded. 

Just one thing: With that you are on your own. Trust me, it works. But you have to set it up by yourself, subsurface developers don't have time to help you figure out how to do that. We provide a simple solution that is supposed to work out of the box (and if it doesn't we are happy to help out) but yes, it has restrictions. 

PS: Subsurface being open source, you even have the option of changing the hard coded server in the sources and compile your own mobile version.

sveeke, 2018-10-31:
> I think we had more characters in the past and those caused problems as various instances between client and server caused enough confusion (like quoting or having different characters under the " [...]

Thanks for your answer. This is a good argument. If more characters result in more user problems (and thus more support/time from Subsurface) then it makes sense not to support it.

> That said, the Subsurface cloud storage is git under the hood. In the manual, it is described how you can set up a local directory as git repository (or by running subsurface with -v option figure [...]

This sounds straight forward enough. Thanks for the pointer. Especially nice for people that like to keep everything under their own control/influence. Recompiling apps (and their updates) is a bit more of a hassle though. But running your own git instance for subsurface is of course no feature normal diving people (non-technies) would consider/want, so obviously it isn't a feature.

Thanks for the pointer and explanation! Much appreciated.

sveeke, 2018-10-31:
> I mean no disrespect, but I think statements like this make it pretty clear that there are some misunderstandings here. Which gets me back to why I hesitate to engage in these discussions. Sure, I [...]

I also mean no disrespect to you, just in case my posts come across like that. I can express myself poorly in English (parts were done with google translate). And I think we would still disagree indeed.

> I'll disengage here. Final comment: backhanded statements like "I value open source software more than all the shortcomings of Subsurface" are a good reminder why I shouldn't have responded in the [...]

Interesting. It may be my lack of good English, but do you mean people should not use software projects you contribute to when they find it has shortcomings? And then you disable their account? I don't understand your reasoning behind that. Every piece of software has shortcomings and in the end it's a individual's balance/consideration (can't find the right word) between those and the good parts. More people I spoke to when searching for 'the best' dive app use Subsurface on iOS/Android primarily or only because it's open source. What's wrong with that? That's a good thing since it means it has unique properties that sets it apart from other dive apps. But that doesn't mean it doesn't have shortcomings... It was sincerely meant as a positive statement about Subsurface.

PR #1049 (2018-01-18): Fix preferences handling with invalid passwords 
### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
This fixes 2 problems related to entering passwords with illegal characters in it:

1. Do not save an invalid password in the preferences, but keep the old one.
2. On password change, check both old and new password for format validity instead of pushing an invalid password to the server that has to ignore it.

Fixes: #1048

# [... cut]
commit d28f48a7 (2018-01-18): Fix preferences handling with invalid passwords

This fixes 2 problems related to entering passwords with
illegal characters in it:
1) Do not save an invalid password in the preferences, but keep the old
one.
2) On password change, check both old and new password for format
validity instead of pushing an invalid password to the server that
has to ignore it.

Fixes: #1048