subsurface/subsurface #1063: Planner: Differentiate between updating everything and updating notes only

sfuchs79, 2018-01-20:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Today in the planner I see two somehow strange effects:
- If I change maximum bottom pO2 setting no update of planner notes is triggered. This is in some way incorrect because changing max bottom pO2 may create or remove some pO2 warnings in the planner notes. The correct results today only become visible if one also changes another setting which triggers full recalculation of everything (diveplan, profile, notes,...)
- If I today change one of the two input parameters for minimum gas calculation, everything (diveplan, profile, notes,...) is recalculated. In this case this is not necessary because minimum gas parameters don't affect diveplan or profile but only some results (printed text) in the planner notes.

I think we should have two functions/code parts/slots/whatever - one doing full recalculation, one updating the notes only. And we should always trigger the one really needed.

Wrote this in the hope that @atdotde can tell us immediately how easy/difficult this is :-)

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@atdotde @RickMWalsh

atdotde, 2018-01-20:
Not updating the notes upon changing the max pO2 is clearly a bug. The easy fix would be to simply trigger a replan. And probably the price is not so high in terms of lag as you are not going to enter different values is quick succession.

OTOH, updating only the notes is probably not too hard either: I would add a slot to the diveplanner model (since that has the dive plan that is needed for the notes) that simply calls add_plan_to_notes() and updates the notes in the dive and in the window. @sfuchs79, are you going to try it or do you want me to give it a shot?

That said, I can think of situations where updating the notes is not enough when changing max pO2: We allow the user to put a "*" in the O2 column in the cylinder table meaning to pick the best mix. That of course depends on max pO2 and that would trigger a full replan. So we could either take care of that separately or ignore it and always replan.

sfuchs79, 2018-01-21:
At least right now I was too stupid to find the missing link to do the simple fix (trigger full replan on bottompo2 and decopo2 change).
Maybe you could give it a try?

mikeller, 2026-10-25:
Hi, [PR #5068](https://github.com/subsurface/subsurface/pull/5068) addresses the planner-notes problem when the bottom pO₂ limit changes. A suitable desktop test artifact was not available when I checked; the [Linux AppImage build](https://github.com/subsurface/subsurface/actions/runs/40146981729) was still running. Please wait for a downloadable artifact from this PR before testing, rather than using a released build.

Once one is available, could you change the maximum bottom pO₂ in the planner and check whether any pO₂ warning in the notes updates immediately, both when the new limit makes the value invalid and when it makes it valid again? If you use an automatic best-mix cylinder, please check that its mix still updates too. The separate request to avoid recalculating the plan and profile when *minimum-gas* inputs change is not addressed by this PR. Please report the platform and any case where the notes remain stale. The artifact will be an unmerged test build; do not use its plan for an actual dive.

closed 2026-10-26 by mikeller