subsurface/subsurface #1064: Printing planned dives with long notes from divelist view - notes don't fit

sfuchs79, 2018-01-20:
### Describe the issue:

- [x] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When trying to print a planned dive with very long dive notes (runtime plus all the extra info we meanwhile have) from the divelist view with e.g. template "One Dive" the notes will never fit.
It would be great to either change the "One Dive" template or to create a new template to support printing a dive on multiple pages and to fit all the info.
Could be best to always start dive notes on the second page if complete dive notes do not fit on space left on first page.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@willemferguson @atdotde

willemferguson, 2018-01-21:
I am working on the print templates. Af far as I am aware grantlee does not provide for multiple-page output. Try the attached template and tell how far you get with it? If the notes become very long, it reduces the font size down to 9 pt, depending on how long the notes are.
[One Dive.html.tar.gz](https://github.com/Subsurface-divelog/subsurface/files/1650025/One.Dive.html.tar.gz)

sfuchs79, 2018-01-21:
I was not able to see any difference between the "old" One Dive and this version.
Beside this I tried to set font size to 9 pt but still long notes won't fit.
I doubt for this we really need a template which prints on two pages.

mikeller, 2026-10-17:
Hi @sfuchs79,

To close the loop on this: long dive-plan notes not fitting in print output is a known limitation of Grantlee, the templating engine the print feature is built on, it's not a simple layout bug we can quickly patch, and would need a more substantial print-template rework to address properly. Given that and the time since this was reported, we're closing it as a known limitation rather than a trackable bug.

Please reopen if you'd like to make the case for prioritising a print-template rework, or if you've found notes being cut off in a way that looks like an actual regression rather than this known constraint.

closed 2026-10-17 by mikeller

dirkhh, 2026-10-17:
Besides this being obviously an **ANCIENT** issue - the answer is incorrect. We have stopped using Grantlee ages ago. So the AI agent closing the issues should be updated to check if statements about dependencies are correct before making errors like this :)