subsurface/subsurface #1070: Problem plotting dive planner points when gas partial pressure graph is enabled

willemferguson, 2018-01-23:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Within the dive planner, take the default dive plan when the planner is activated. There is only one dive planner point indicated, at 15m and 20 min. Activate or deactivate the O2 partial pressure graph. Because of the PO2 graph, the vertical space available for the dive profile decreases and the profile now has to fit in less vertical space. This works ok. However, the dive planner point (or points) is not re-plotted with the rest of the profile. The result is that the white dive planner point hangs separately from the dive profile graph, usually in the area meant for the PO2 graph.

### Operating system:

Ubuntu 16.04

### Subsurface version:
 4.7.6-125 up to 4.7.6-139

test build

g18c54301

### Steps to reproduce:

### Current behavior:

### Expected behavior:

The dive planner point(s) should re-plot with the dive plan profile.

### Additional information:

### Mentions:

janmulder, 2018-01-23:
Interestingly, I can not reproduce this. The waypoint stays attached to the profile when switching bottom stuff (like the po2 graph) on and off.

willemferguson, 2018-01-26:
I can repeat this bug in Fedora as well. Do it as follows:
1) Create a deco dive, say to 50m.
2) Switch on the display of the dive ceiling (topmost profile tool button). Dive planner points move as they should when the O2 graph is switched on and off.
3) Switch off the display of dive ceiling.
4) Now repeat the process of switching the O2 graph on and off.
5) The dive planner points are left behind as originally described. See attached image.
[image]

PR #1085 (2018-01-31): profile: replot handles while planning
### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
The settingsChanged() function of the profile widget tries to be clever and tries to prevent not needed replots of the profile. I'm not sure this is very relevant in the first place as change of settings are almost instantly, with or without replot. It appears that replot is always executed when there are calculated ceilings visible. Bu [... cut]
commit 20dc6a31 (2018-01-31): profile: replot handles while planning

The settingsChanged() function of the profile widget tries to be
clever and tries to prevent not needed replots of the profile.
I'm not sure this is very relevant in the first place as change
of settings are almost instantly, with or without replot. It
appears that replot is always executed when there are
calculated ceilings visible. But without calculated ceilings
a replot is "optimized out". This does, however, introduce dangling
divehandles in PLAN/EDIT and ADD mode of the profile, when
any setting that influences the y-axis is changed.

The fix is tr [... cut]
closed 2018-01-31 by janmulder