subsurface/subsurface #1078: Sigdev crash when exiting from dive planner

willemferguson, 2018-01-26:
Crash on non-intended exit from dive planner

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

V4.7.6 on Ubuntu 16.04
To leave the dive planner after doing some planning, one either has to cancel the dive plan or save it. If one by accident closes the application by clicking the close window icon on the window frame, the program crashes:
_subsurface: Fatal IO error 2 (No such file or directory) on X server :0.
Segmentation fault (core dumped)_
This is not ideal at all. There used to be a dialog box with a message that said "Save your data or cancel the dive plan" or something similar.

### Expected behavior:

Provide a dialog box with a warning or exit gracefully.

### Additional information:

### Mentions:

atdotde, 2018-01-26:
Does this always happen? I cannot reproduce a crash upon closing the main window on my Mac (although the app just closes without a dialogue as long as there are no changes to the divelist. Otherwise there is the usual "Do you want to save?" modal dialogue coming up).

willemferguson, 2018-01-26:
Its is absolutely consistent, since at least V4.7.5.
Apologies for being a trouble maker.

neolit123, 2018-01-26:
@willemferguson hi, can you proved a crash log?

for that you can use `gdb`:

```
sudo apt-get install gdb

$ gdb subsurface
# run subsurface inside the debugger
gdb> r
# make it crash
# then generate a back trace
gdb> bt 30
# share the results
```

willemferguson, 2018-01-27:
I spent some time looking at this.
This happens in Ubuntu, not in Fedora. Since my Ubuntu is quite old (16.04), this is possibly an OS and installation issue. I will hopefully upgrade my OS when 18.04 is available. In general, what happens on Ubuntu is that from hitting the "Quit" button it takes about 7 seconds for the program to close down. The problem I reported is associated with this slow close-down of the software. So, when I start Subsurface, go into the planner and then hit the Close Window icon, the message in the image below appears:
[image]

If I do not do anything and wait for 7 seconds, the following trace appears:
(gdb) r
Starting program: /home/willem/src/subsurface/build/subsurface 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe6767700 (LWP 3668)]
[New Thread 0x7fffdc0cc700 (LWP 3669)]
[New Thread 0x7fffdb8cb700 (LWP 3670)]
[New Thread 0x7fffdb0ca700 (LWP 3671)]
[New Thread 0x7fffd9036700 (LWP 3673)]
[New Thread 0x7fffd8835700 (LWP 3674)]
[Thread 0x7fffd9036700 (LWP 3673) exited]
[New Thread 0x7fffd9036700 (LWP 3676)]
[New Thread 0x7fffc0087700 (LWP 3677)]
[New Thread 0x7fffbde92700 (LWP 3680)]Gry1!u5:

[New Thread 0x7fffbd691700 (LWP 3681)]
[New Thread 0x7fffbce90700 (LWP 3682)]
[New Thread 0x7fffaffff700 (LWP 3683)]
[New Thread 0x7fffaf7fe700 (LWP 3684)]
[New Thread 0x7fffaeffd700 (LWP 3685)]
[New Thread 0x7fffae7fc700 (LWP 3686)]
[New Thread 0x7fffadffb700 (LWP 3687)]
[New Thread 0x7fffacffa700 (LWP 3688)]
[New Thread 0x7fff93fff700 (LWP 3689)]
[Thread 0x7fffd8835700 (LWP 3674) exited]
[Thread 0x7fff93fff700 (LWP 3689) exited]
[Thread 0x7fffacffa700 (LWP 3688) exited]
[Thread 0x7fffd9036700 (LWP 3676) exited]
[Thread 0x7fffc0087700 (LWP 3677) exited]
[Thread 0x7fffadffb700 (LWP 3687) exited]
[Thread 0x7fffbd691700 (LWP 3681) exited]
[Thread 0x7fffae7fc700 (LWP 3686) exited]
[Thread 0x7fffaffff700 (LWP 3683) exited]
[Thread 0x7fffbde92700 (LWP 3680) exited]
[Thread 0x7fffaeffd700 (LWP 3685) exited]
[Thread 0x7fffbce90700 (LWP 3682) exited]
[Thread 0x7fffaf7fe700 (LWP 3684) exited]
[Thread 0x7fffe6767700 (LWP 3668) exited]
[Thread 0x7fffdb8cb700 (LWP 3670) exited]
[Thread 0x7fffdc0cc700 (LWP 3669) exited]
[Thread 0x7ffff7e2d900 (LWP 3664) exited]
[Inferior 1 (process 3664) exited normally]
(gdb) bt 30
No stack.

That means a normal exit.

If , on the other hand, I select "Force quit", then the following trace appears. This is the option I generally take when a window hangs.
(gdb) r
Starting program: /home/willem/src/subsurface/build/subsurface 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe6767700 (LWP 4074)]
[New Thread 0x7fffdc0cc700 (LWP 4075)]
[New Thread 0x7fffdb8cb700 (LWP 4076)]
[New Thread 0x7fffdb0ca700 (LWP 4077)]
[New Thread 0x7fffd9036700 (LWP 4079)]
[New Thread 0x7fffd8835700 (LWP 4080)]
[Thread 0x7fffd9036700 (LWP 4079) exited]
[New Thread 0x7fffd9036700 (LWP 4082)]
[New Thread 0x7fffc0087700 (LWP 4083)]
[New Thread 0x7fffbde92700 (LWP 4086)]
[New Thread 0x7fffbd691700 (LWP 4087)]
[New Thread 0x7fffbce90700 (LWP 4088)]
[New Thread 0x7fffaffff700 (LWP 4089)]
[... 25 lines cut ...]
#0  0x00007ffff543f468 in QAbstractEventDispatcherPrivate::allocateTimerId() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff543fb49 in QAbstractEventDispatcher::registerTimer(int, Qt::TimerType, QObject*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff544020d in QBasicTimer::start(int, QObject*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007ffff68ecd1d in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff5471d2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff54725df in QObject::destroyed(QObject*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff547967a in QObject::~QObject() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff53e70a9 in QAbstractItemModel::~QAbstractItemModel() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff4371ff8 in __run_exit_handlers (status=1, listp=0x7ffff46fc5f8 <__exit_funcs>, 
    run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#9  0x00007ffff4372045 in __GI_exit (status=<optimized out>) at exit.c:104
#10 0x00007fffe4a0cee1 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#11 0x00007ffff23cbcee in _XIOError () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#12 0x00007ffff23c9d4a in _XReply () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#13 0x00007fffc3d4a248 in ?? () from /usr/lib/nvidia-364/libGLX_nvidia.so.0
#14 0x00007fffc3d4ab75 in ?? () from /usr/lib/nvidia-364/libGLX_nvidia.so.0
#15 0x00007fffedadd7c3 in ?? () from /usr/lib/nvidia-364/libGLX.so.0
#16 0x00007fffedaddd61 in ?? () from /usr/lib/nvidia-364/libGLX.so.0
#17 0x00007ffff7fe90eb in ?? ()
#18 0x0000000012ec1fd0 in ?? ()

I am not sure that this would yield anything interesting from a Subsurface point of view. Would this have any information on what was executing while it took 7 s to close down?

atdotde, 2018-01-27:
This seems to be deep in Qt land (in the destructor of some Qt object). The only Subsurface specific things seems to be the error message

> subsurface: Fatal IO error 2 (No such file or directory) on X server :0.

but without knowing which filename it tried to access (maybe the settings file or the image cache) this does not really help.

BTW I find using the debugger from inside Qt Creator (where you have a GUI showing all the relevant things including local variables and where you are in the code) several orders of magnitude more convenient than the command line version of gdb.

Another thing you could try is to pause execution during the seven seconds and go down the stack trace until you find where in the subsurface code we are (i.e. not in Qt but in the subsurface source).

dirkhh, 2018-01-27:
If this only happens in Ubuntu and not in Fedora, there's also a risk that this is caused by Ubuntu's changes to Qt (in order to get their Mac style menu bar, plus some other various modifications).
That makes it harder to fix :-(

neolit123, 2018-01-28:
the stack trace suggests a bug in `qt`, so the `bug-not-in-subsurface` label seems appropriate.
have you tried a building with a different Qt version @willemferguson?

what output do you get when you run `subsurface -v`? i wonder if it's using the GPU driver.

PR #2807 (2020-05-03): desktop: ask user when quitting application while planning
We used to cancel a plan (or profile edit) when the user quit
the application while planning. This is inconsistent with
respect to closing or opening a different log, where the user
was asked for confirmation.

Thus, for consistency and to avoid loss of a planned dive,
use the okToClose() function in on_actionClose_triggered() of
MainWindow. As an added bonus, this saves a few SLOC.

Fixes #1078

### Describe the pull request:

- [ ] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Do [... cut]
commit 8f8457eb (2020-05-03): desktop: ask user when quitting application while planning

We used to cancel a plan (or profile edit) when the user quit
the application while planning. This is inconsistent with
respect to closing or opening a different log, where the user
was asked for confirmation.

Thus, for consistency and to avoid loss of a planned dive,
use the okToClose() function in on_actionClose_triggered() of
MainWindow. As an added bonus, this saves a few SLOC.

Fixes #1078
closed 2020-05-03 by dirkhh