subsurface/subsurface #1079: Reverse geo lookup icon not there

sfuchs79, 2018-01-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Reverse geo lookup icon (the small satellite) disappeared:

[image]

Can anyone confirm this?

### Operating system:

Windows 10

### Subsurface version:

Latest master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

janmulder, 2018-01-27:
Yes, can confirm on current master on Linux. Icon is not shown. And I think I saw this already when I fixed a crash in this area (in commit 435181aa19a4f).

dirkhh, 2018-01-27:
Can you bisect this and find out when it disappeared? A quick check here with 4.7.3 on Mac shows that the icon is there... but I don't think I'll be at the computer for enough time today to figure out when it disappeared (I need to head back to bed and rest)

sfuchs79, 2018-01-27:
Waah, bisecting still sucks on my slow machine. 3 children - was not able to afford the Ryzen I wanted to buy for Christmas. DDR4 memory too f**cking expensive at the moment.
But found a better approach: Drink half a bottle of nice Italian red wine, look at the code... and fix it. Yeah, it was not too difficult ;-)
Will submit a PR...

dirkhh, 2018-01-27:
You will forgive me for worrying about code after half a bottle of red wine (or while recovering from the flu - which is why I am not writing code right now and am hesitating to do in depth code reviews) ;-)

dirkhh, 2018-01-27:
Oh, and bisecting Subsurface is slow, even on blazing fast systems...

PR #1080 (2018-01-27): Use correct alias for satellite icon for revers geo lookup

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Bring back the satellite icon for the reverse geo lookup by using
the correct alias name.

Closes #1079 

### Changes made:

### Related issues:

### Additional information:

### Release note:

### Documentation change:
closed 2018-01-27 by dirkhh