subsurface/subsurface #1090: Salinity is acutally water density in dive planner

Schwaneberg, 2018-02-02:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [X] Discussion request

### Issue long description:
1) When I used the dive planner for the first time, I stumbled across the word "**Salinity**" (actually the word "**Salzgehalt**" in the German translation) and was confused to see a default value of 1kg/l! So, I believe this value should be defined as the "**water density**".
2) When I made my OWD last year (yes, I am a noob), I learned the textbook value of 1.025kg/l for sea water. However, this value cannot even be filled in the UI because of the limitation to 2 decimals. Should we add a third?
3) The value is restricted to a maximum of 1.05kg/l, but I read that water densities of up to about 1.25kg/l occur in hypersalin lakes like the dead sea. I think the limit should be raised accordingly.

### Additional information: 
There are publicly available salinity databases. There is a project called [UCNLSalinity](https://github.com/ucnl/UCNLSalinity) dealing with a global dataset created by the National Oceanic and Atmospheric Administration (US department). The project contains a big XML file that maps global coordinates to salinity expressed in [PSS](http://salinometry.com/pss-78/) (which can be converted to water density, if the temperature is known). It might be a bit overengineered, but I think it would be cool if Subsurface would utilize such a database.

### Operating system: Ubuntu 16.04

### Subsurface version: 4.7.6

atdotde, 2018-02-05:
I am not entirely sure about this. I guess the some reference to "salt" should be in the UI so users understand that this is in the end about the salinity (even though you are correct to point out that what we are asking for is the density). 

About the additional digit: Well, maybe yes, but of course this permille precision digit it totally bogus and does not influence the calculation in a meaningful way. After all, all depths are in the end pressures (as this is what dive computers measure) and nobody except maybe in some free diving record attempts measures depth using a rope of a defined length. And it is really the pressure that matters for the decompression not the depth. All this sweet water/salt water thing is only how the numbers are presented to the user (as humans prefer to measure depths in meters or feet rather than bar/psi) but its only cosmetics.

The only thing that matter is that the dive computer and Subsurface have the same idea how to translate between depth and ambient pressure (as the communication is also in terms of depths). So, if somebody for (what I would consider stupid reasons) decided to set their computer to sea water mode Subsurface needs to be able to "undo" this. So from this point of view it is explicitly wrong to use "real" data from a database as long as the dive computer doesn't. Plus then the user would have to add a geolocation to a planned dive (rather than a number that is totally save to leave it as it is). And did anybody mention "over engineering"? And a C# library?!?

I would say: Feel free to add a digit and a combo box (it is actually a good idea to have reasonable defaults).

Schwaneberg, 2018-02-05:
> I would say: Feel free to add a digit and a combo box (it is actually a good idea to have reasonable defaults).

Thank your for your response. I will _replace_ the current field by a combo box.

>  And did anybody mention "over engineering"? And a C# library?!?

I was referring to the salinity database integration as a feature. I agree with you that the actual value isn't that important.

glance-, 2018-02-05:
I created that field way back in time, because we tend to use "sonar depth", and not "dc reported depth".

Not that it really matters, a quick check says that it varies 3 minutes on a typical 70m dive if your're in fresh(1.0) or salt water(1.03).

dirkhh, 2018-02-05:
Hold off here for a moment. No, we will not add a third digit to our salinity / density value. That's just ridiculous. Salt water density distribution isn't remotely close to being consistent to a permille level. Also, typical pressure sensors don't have better than 0.1% accuracy (which would be required for this change to make any sense at all).
This is fake precision and we shouldn't start falling into that trap. On a 50m dive this is a 25cm difference. Your wrist position has a bigger impact on your depth than that.
So NAK on this idea.

Schwaneberg, 2018-02-05:
Ok, I got that there is no fanbase for a third digit, but what about the proposed combo box containing the presets "fresh water", "sea water" and (perhaps) "custom"?

dirkhh, 2018-02-05:
I'm ok with that

sfuchs79, 2018-02-05:
If we go for a combobox for salinity I would recommend to add a 4th option which is the EN13319 value of 1,020kg/l. Some (even many?) dive computers use this value as a default.

Schwaneberg, 2018-02-05:
Or we just choose 1,02kg/l as our sea water default. If I'd be a dive computer manufacturer, I would grab such values from standards whenever possible - if that is an argument ;-)

closed 2018-02-11 by Schwaneberg