subsurface/subsurface #1113: Integrate AppImageUpdate Qt UI to implement self-updating for Linux releases of Subsurface

TheAssassin, 2018-02-13:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
AppImageUpdate is in the process of receiving a reusable Qt UI which Qt apps can integrate with very little effort to implement a self-update feature.

See https://github.com/AppImage/AppImageUpdate/pull/67 for the WIP branch.

When `show()`ing the widget, it automatically starts an update process using libappimageupdate in the background, so the AppImage needs to contain update information to make use of this feature.
The widget is a modal, so the rest of the app can't be used. When the update has finished, the user can either close the widget and continue to work with the existing version of the app, or run the new AppImage.
Furthermore, there's an update check available, for applications which don't have their own mechanisms. It can also be used in combination with an existing method to make sure the new AppImages have been deployed, in case the update notification is sent before the new files are available on the servers.

There's a few remaining points which require a real application for testing:
  - Allow overriding "re-run" button behavior, or provide mechanism to hook into it. This should allow apps to close "gracefully", e.g., to ask the user whether to save the current document, etc.
  - (doesn't need a real application, actually) Implement update cancellation in AppImageUpdate. Rollback is easy, since libzsync2 creates a separate file which is finally moved into place, but when cancelling an update, the files should be cleaned up properly. (Current behavior: throws unhandled `std::runtime_error`)

Subsurface seems like the perfect candidate to test this very new feature with.

What do you think about adding self-updating to Subsurface? We can assist with the implementation, but I think it's pretty straightforward already.

### Mentions:
@dirkhh @probonopd

probonopd, 2019-06-06:
@anthony-jr has made a Qt plugin that can update an AppImage without even touching the application's source code. It can override an existing menu entry, in the case of Subsurface "Help -> Check for updates".

[image]

Check it out:
https://github.com/TheFutureShell/QtUpdateBridges/issues/2#issuecomment-499597102

glance-, 2019-11-10:
As said in #770 I really like this and I'm all for implementing it.

antony-jr, 2019-11-11:
> As said in #770 I really like this and I'm all for implementing it.

Would love to help integrate the [official library](https://github.com/AppImage/AppImageUpdate) into Subsurface. Might need a custom GUI instead of the default one made by @TheAssassin just for the aesthetics. 

> @antony-jr has made a Qt plugin that can update an AppImage without even touching the application's source code. It can override an existing menu entry, in the case of Subsurface "Help -> Check for updates".

Note that it overrides the Qt XCB plugin  to load the other plugin and then override menu entries. *Kind of hack* so if you consider using this please review it further. I will always recommend using the official library. **Only bonus for using this is that it depends only on Qt base and its single threaded.**( so I think it's decent as a library and plugin)

mikeller, 2026-10-18:
Hi @TheAssassin,

This thread (and the related #770) has an unusual amount of already-prototyped groundwork from you, probonopd and antony-jr, a working Qt plugin demonstrating binary-delta self-updates. Given it's now been a few years since that prototyping, is the approach (and the libappimageupdate API it relies on) still current/viable? If so, we'd treat the remaining work as mostly packaging/build-pipeline integration rather than fresh design, keen to know if you're still interested in seeing this land in Subsurface's own AppImage build.