subsurface/subsurface #1116: Current available gas calculations are misleading for multiple bottles.

arosl, 2018-02-15:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

Available gas calculations are misleading if you have more than one bottle you plan to use on the bottom phase.

### Issue long description:

Gas calculation when planing a dive with multiple bottom stages. I regularly do cave dives that require one or more bottom stages, but only one deco tank. (shallow and long dives). The only way to get the gass planing correctly in Subsurface is to add the size of the bottom stages together with the back gass. 

I suggest a change where bottles with the same gas are added together in the gas calculation, in stead of showing this:

Gas consumption (based on SAC 16|13ℓ/min):
7258ℓ/316bar of EAN32 (51ℓ/3bar in planned ascent)
 — Warning: this is more gas than available in the specified cylinder!
0ℓ/0bar of EAN32
0ℓ/0bar of EAN32
80ℓ/14bar of oxygen (80ℓ/15bar in planned ascent)

The gas in the AL80 bottles with EAN32 should have been used first. (its common practice in tech/cave diving to use (bottom) stage cylinders first, as this gas can not easily be shared with a buddy as you only have one second stage on it)

### Operating system:

macOS 10.13.3

### Subsurface version:

4.7.7

It happened before I upgraded as well.

official release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

sfuchs79, 2018-02-17:
Just for clarification - you want to do s.th. like this?!:
(Sorry for the German - I was too lazy to change localization)
[image]

arosl, 2018-02-18:
@sfuchs79 , thank you for pointing that out. I had totally missed that the field "used gas" actually referred to the specific cylinder. It doesn't really help me, as I want to know if I have enough total gas for the planned dive. If I manually have to select how many minutes for each gas it becomes to cumbersome. Automatically group similar gasses together makes more sense to me, and for my personal experience I can't come up with a common senario that would be a problem for the dive planing. But I guess there is a reason behind this  implementation, so this might be closed as a user error. I still think it would be a major improvement to to have the option to group the same gasses together from a usability perspective.

mikeller, 2026-10-18:
Hi @arosl,

The existing per-cylinder "used gas" assignment field only partially covers this (as you confirmed with sfuchs79), grouping same-gas cylinders together for a combined availability check would need a change to the planner's consumption-calculation/display logic specifically. Still interested in this for your cave/technical multi-bottle setups? Confirming that'll help us prioritise it.

arosl, 2026-10-22:
HI, Mikeller, 

Yes your correct assuming its for cave/tech. I must admit 99% of all my dives the last 6-7 years have been pSCR or mCCR so probably why I never followed up on this. 

But I think the logic of grouping similar gasses together to a total volume makes more sense than handling them individually. So for me personally it's not a priority as I would use a RB for my own dives. But most people cave diving still is on open circuit, and that would make there planing easier. So for people getting serious about their dive planing it would be a nice upgrade using bottom stages first from a total volume of similar gasses.

mikeller, 2026-10-22:
Hi @arosl,

Separately from the open-circuit gas-planning request in this issue, your pSCR experience may be useful for an Android planner improvement we are still assessing. As discussed recently in #5022, the mobile app does not currently expose the pSCR dilution-ratio and O₂ metabolic-rate settings that are available on desktop.

If you use Subsurface on Android and would be willing to help, it would be useful to know:

- which pSCR unit and Android device/version you use;
- which pSCR planner settings you need to inspect or change in normal use; and
- whether you would be willing to check a future test build for the expected workflow and settings, without relying on it for dive planning.

There is no test build to try yet, but feedback from someone actively diving pSCR would help keep the work grounded in real use.

arosl, 2026-10-22:
Im sorry, I totally captured by Apple 🍎 and use the iPhone version as my way of syncing my dives of the handset, and the macOS version to actually look at the logs, so I have never seen the Android planner. 

Also my RBK is my backup RB at this point and my dives the last two years have mostly been on the Fathom Gemini. I still have many hours on a pSCR but it's not something i actively dive every week. I however happy to contribute any knowledge i might have about pSCR diving regardless.

arosl, 2026-10-22:
For planning purposes Ke is the better option as that is the combined RMV/Vo2 value that is actually what matters for your o2 drop. 

pSCR divers usually find out there Ke values from dives with a o2 sensor when they get their training and use this for planning so that the o2 drop not will be dangerously high for their gasses. And then adjust their gasses according to this p2 drop.

mikeller, 2026-10-22:
Thanks, that is a useful clarification. Using Ke established from sensor-observed dives during training better describes the practical planning workflow than treating O₂ metabolic rate as an isolated value.

This is closely related to the pSCR settings and planner discussion in #3400, which already covers the interaction between SAC/RMV, O₂ metabolic rate, and Ke. To keep this issue focused on the open-circuit multi-cylinder gas-planning request, could we continue the pSCR discussion there?

In particular, your experience with how Ke is established and checked in normal planning would help us assess both desktop and mobile planner support. We do not have a concrete change to propose yet, and would not change pSCR defaults or calculation inputs without suitable review.

pabdakine, 2026-10-24:
In the OC cave scenario, the user can effectively group bottles by just defining one bottle in the planner and focus on the total amount displayed (disregarding the pressure value). Take at least 1.5x that amount (2-way profile) or at least 3x (1-way profile) distributed among various bottles. No changes to Subsurface are needed.