subsurface/subsurface #1119: Decimal separator comma vs. dot

sfuchs79, 2018-02-17:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

All over the Subsurface UI when printing values we use a dot "." as decimal separator. There is no config option for this. Many of our strings containing values are generated by sprintf which uses the dot as default.
**I think this is OK and I'm NOT the one asking for any change here!!! ;-)**
But if you look a little bit closer, mainly in the planner, then you recognize that some QT UI elements (mainly QSpinBoxes) change the decimal separator to a comma "," if e.g. German language is selected. And at this moment we have a mixture with dots and commas. That became most obvious to me when testing @Schwaneberg salinity feature.

Shall we maybe prohibit the QT UI elements to switch to a comma for the sake of consistency? As I was saying I'm not the one asking to have a comma everywhere also because I think it would be a lot of work to change all our sprintf to support a comma.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @atdotde

Schwaneberg, 2018-02-17:
I believe the clean way would be to use localized decimal separators everywhere.

atdotde, 2018-02-17:
I am torn on this issue. In principle I like localisation, but localising the decimal separator has already caused me so much pain. The greatest offender is MS Excel. They even put "," in CSV files (I am obviously not talking about the field separator). When you read those with another program that is not aware of this you get ill-formed numbers. And importing data into Excel with the wrong decimal separator makes that field a string (rather than a number) and in Excel it is (almost) impossible to convert that back to a floating point number (and this works in both directions). Argh.

That said, I would have expected sprintf() to localise its numerical output....

Schwaneberg, 2018-02-17:
That is also true. Excle is a good example as it is really totally broken in this regard. Therefore, I recommend only to localize the numbers as they are displayed in the GUI - no internal stuff, no CSV export behavior.
I know that many (German) people are confused by numbers like 1.00 and it is not getting better if you mix both expressions.

sfuchs79, 2018-02-17:
:-)
Well, I saw this coming...
Let me repeat: "I'm not the one asking for..." ;-)

Honestly, it would be pretty much work to teach all the sprintf in the C domain to "automatically" either use a comma or a dot, isn't it?

atdotde, 2018-02-17:
Quick as I am on the phone: locales should already do that for you: https://stackoverflow.com/questions/7684014/c-printf-with-f-but-localized-for-the-users-country

We just need to make sure this does not happen when writing to files.

dirkhh, 2018-02-17:
Oh, my favorite topic... NOT.
Yes, we should follow the locale and since Germans are using the comma as symbol for the decimal point, it is good practice to show that. I know that at some point we were fairly consistent in doing that, but since I never use Subsurface in German I have no idea what the current state might be.
We have a TON of German users - so this certainly is a change request that will affect many of our users.
My biggest fear is unintended side effects and breakage that we might introduce in data stored locally or in the cloud. That said, yes, if anyone has the energy to work on this, you have my encouragement.

Schwaneberg, 2018-02-17:
And this is not only a German problem. [Many countries](https://en.wikipedia.org/wiki/Decimal_separator) around the globe use commas as decimal separators.

sfuchs79, 2018-02-17:
Maybe this is really much easier than I expected...
I have s.th. here:
https://github.com/sfuchs79/subsurface/tree/feature_decimal_separator

Not everything covered, some things not working, almost nothing tested, no idea how dangerous it could be...
Any hints?

What in particular doesn't work is the:
`setlocale(LC_NUMERIC, prefs.locale.lang_locale);`
And I don't know why.

dirkhh, 2018-02-18:
Is that a Qt question? Or what exactly doesn't work?
I'd recommend asking such questions on the mailing list, lots of the developers aren't subscribed here

sfuchs79, 2018-02-18:
@dirkhh 
The issue was that I had to figure out what type of language string (like "de_De" or "German" or whatever) I have to pass to setlocale()
I finally found a solution which works under Windows which is passing strings like "German" or "English" and I can extract them from prefs.locale.language
@atdotde @Schwaneberg 
So with the hints from both of you I have a version now which switches correctly between using dots and commas for the decimal separator. May still be that I missed some values. I will give this some more testing even before submitting a PR...
I updated my branch mentioned above.
But already now any feedback is welcome!

Schwaneberg, 2018-02-18:
Hi! I just checked your branch and it looks pretty good to me - but I got some remarks:

- The x-axis of the He, N2, O2 plots is not yet localized
- That "Teilen auf"/facebook menu item does not open (but I don't see why)
- In profile-widget/diveprofileitem.cpp line 807 you removed a blank space between %1 and %2. The plotted value "pressure" is an integer, so no localization required.

PS: I tested CSV and XML export and I confirm that they are not affected.

Schwaneberg, 2018-02-18:
You mentioned in your commit messages why you changed the gui and line 807. I missed this before. Sorry.

sfuchs79, 2018-02-18:
@Schwaneberg 
Thanks for testing!

> The x-axis of the He, N2, O2 plots is not yet localized

Now also done. This one was a little bit more tricky... but just a little bit.
And I bet this was not the last value I missed.

I once also really have to grep the whole code to check if not any import/export function is using some of the Qt helper functions which I changed.

dirkhh, 2018-02-18:
Looking forward to a pull request - then we can get this broader testing

sfuchs79, 2018-02-18:
Ok, why not. Testing is what we need for this...

I meanwhile searched the code for any calls to the changed functions. I up to now found nothing obviously harmful. What I found is that the HTML export is using some of the changed Qt helpers. I already hesitated if this could be bad but then it became clear to me that it even will be nice because then in the HTML export e.g. a German user will also have the comma instead of the dot.
But what I saw is that in the HTML export there are some more values to change. But that's just work - not for today but I can add this during the next days.

sfuchs79, 2018-03-05:
Closed via #1122 and #1134

closed 2018-03-05 by sfuchs79