subsurface/subsurface #1123: Move friendly name before UUID in Subsurface-mobile

arosl, 2018-02-19:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When using the iOS version of Subsurface-mobile on an iPhone the friendly name (Perdix) is not visible in portrait orientation on an iPhone 7. Moving the "friendly name" in front of the UUID would increase the usability of the app on small screens. 

Workaround is to use the iPhone in landscape mode.

### Operating system:

iOS 11.2.5 (English US, with Norwegian time and date localisation)

Iphone 7

### Subsurface version:

Subsurface-mobile 2.0(4.7.4.280)

Official release from App Store.

### Steps to reproduce:

1. Try to sync an Perdix in portrait mode on an iPhone 7.

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2018-02-19:
Hmm, I thought I had already implemented that in #1c223b09c3

janmulder, 2018-02-19:
Yes, implemented but not in the respective app store versions (yet), as we are behind with that.

janiversen, 2018-06-22:
@janmulder When do we close a bug ?

I got confused by this one looking for outstanding iOS bugs, it is implemented and have been available in several test flight versions. Other issues are closed as soon as the fix is merged on master.

dirkhh, 2018-06-22:
I typically close bugs once they are in master.
I am unusually late with releasing new apps - I hope to get the new v2.1 out within a week (FINALLY)

closed 2018-06-22 by dirkhh