subsurface/subsurface #1133: Failed to parse

NHscuba, 2018-02-23:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I open Subsurface it gives me an error message of Failed to parse 'C:\Users\Scott\AppData\Roaming\Subsurface\Scott.xml'

This is new - I used subsurface as recently as a couple weeks ago and had no issue. I can create a new logbook and it is fine but I've lost data if I do that. The Scott.xml DOES contain the data (when viewed as text) but Subsurface can't read it. 

### Operating system:

Windows 10, 64-bit
### Subsurface version:

Subsurface 4.7.7, official release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

[ScottXML_subsurface.pdf](https://github.com/Subsurface-divelog/subsurface/files/1752713/ScottXML_subsurface.pdf)

### Mentions:

janmulder, 2018-02-23:
There is some weird data in this XML:

 ```
<sample time='0:00 min' depth='4.999 m' temp='21.111 C' pressure0='564.68 bar' pressure1='564.68 bar' ndl='99:00 min'
 <sample time='0:00 min' depth='2.195 m' temp='0.0 C' pressure0='564.68 bar' pressure0='564.68 bar' ndl='99:00 min'
 <sample time='0:00 min' depth='2.56 m' temp='5.556 C' pressure0='564.68 bar' pressure0='564.68 bar' ndl='99:00 min'
```
The parsing fails on a redefine of `pressure0 ` in the bottom 2 lines. Manually changing that data (just rename the second `pressure0 ` to `pressure1 `) makes the data readable again.

This said. The big question is here ... where does that data come from? It is not only the double defined pressure, but 564 bar (8100 psi) seems not correct as well.

NHscuba, 2018-02-23:
That did the trick! I have no idea about the weird data but my dives are viewable again in Subsurface - thank you!

sfuchs79, 2018-02-23:
@janmulder 
Did you also recognize that the offending lines with the weird "pressure" entries are only present for dives with:
`<divecomputer model='Shearwater Petrel/Perdix'>`

@NHscuba 
Could you please re-import dives from the Petrel/Perdix in the same way you did before and see if the issue reoccurs?

janmulder, 2018-02-23:
@sfuchs79. Saw that, but did not realize this ... and there was a recent libdivecomputer commit reverted related to the Shearwater parser.

@NHscuba: and when the problem reoccurs ... could you again import and select the 2 options "Save libdivecomputer logfile" and "Save libdivecomputer dumpfile", and post the resulting output files here?

NHscuba, 2018-02-24:
I tried to to save it as asked but got the following error message. The logbook DID update though...
[image]

janmulder, 2018-02-25:
@NHscuba: that error can happen sometimes using Bluetooth interfacing. There is so much that can go wrong, in a seemingly non deterministic way, that my way of dealing with is, is just try to download again.  

But, I am puzzled by your remark 

> The logbook DID update though.

So, are you saying that the import with the 2 libdivecomputer boxes checked did update your logbook? And did this introduce the bogus data again?

NHscuba, 2018-02-26:
I updated the logbook as normal. There was no bogus data introduced. Then,
when I also clicked the 'dump file' box, that error message appeared. I
tried a few times to get the dump file but it wouldn't let me.

I'm not sure what happened with the bogus data but everything looks good
now. Thanks again!

closed 2018-02-27 by janmulder