subsurface/subsurface #1155: Erroneous `snprintf`s

bstoeger, 2018-03-08:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

Note: I sent a mail to the mailing list, but it was lost in the ether, so I'm posting this here.

In the files `core/plannernotes.c`, `core/statistics.c` and `core/uemis-downloader.c`, we have code of the kind
```
snprintf(buffer + len, sz_buffer - len, ...);
```

I think this is wrong because the second argument to snprintf is `size_t`, i.e. unsigned. For `len > sz_buffer`, this will happily write past the end of the buffer. This is not a pressing issue, since the buffers seem to be amply sized. Nevertheless, let's fix this (or switch to the non-bound checked `sprintf`).

Ad-hoc, I see three rather easy to implement fixes:

1)
```
int sane_snprintf(char *str, int size, const char *fmt, ...)
{
        va_list ap;
        va_start(ap, fmt);
        int res = vsnprintf(str, max(0, size), fmt, ap);
        va_end(ap);
        return res;
}
```
2) Use `struct membuffer` and `put_format()`.
3) Change to `QString`:
```
res += QString::asprintf(...);
```

Concerning the amount of work, I'd say it's 1 < 2 < 3 (from least to most work  needed). Personally, I feel that nevertheless 3 would be the best option, even though I dislike `QString` (COW, UTF-16). This would remove back-and-forth conversion QString -> C-string -> QString and wouldn't change the C-logic at all. Moreover, for my tastes, it's the easiest-to-read version.

### Operating system:

OS independent

### Subsurface version:

current master (4.7.7)

### Steps to reproduce:

### Current behavior:

Some `snprintf` instances do no bound checking.
 
### Expected behavior:

All `snprintf` instances should do bound checking.
 
### Additional information:

### Mentions:

@dirkhh

atdotde, 2018-03-08:
Using QString violates our rule to avoid calling C++ from C (which we already violate by all the stuff in qthelper.cpp). Such would require a good justification.

In any case, if our string outgrows the buffer, we are screwed. So, there is also a 0th possibility: throwing in a number of 

assert(len <= sz_buffer);

and act if we ever hit that condition. Until then, this is compiled out in production code.

bstoeger, 2018-03-08:
I believe that truncating a string is better than either a buffer-overflow or a hard abort. Best thing would be to not rely on fixed buffer sizes (2 or 3).

glance-, 2018-03-08:
membuffer's for the win.

commit a4834e19 (2018-04-11): Cleanup: fix "unsave" snprintf()s in plannernotes.c

The planner notes were constructed using a sequence of
  len += snprintf(buf, buflen - len, ...);
calls. This will fail once len > buflen, because the second parameter
of snprintf is unsigned. Note that snprintf returns the number of
bytes that would have been written if it weren't truncated.

Fix this by using membuffer with put_format()/put_string() and
asprintf_loc().

Fixes #1155.
closed 2018-04-11 by neolit123