subsurface/subsurface #1158: Changes to XSLT when exporting dive profile as *.CSV

p-neves, 2018-03-11:
### Describe the issue:

- [ ] Bug
- [X ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'd like the following changes to the XSLT for when exporting a dive profile as *.CSV:
    - Export heart rate (create new field - e.g. "sample.heartrate")
    - Use the same decimal separator for all fields (now. “sample.depth” uses “,” and “sample.temperature” uses “.” as separators)

### Operating system:

Arch Linux

### Subsurface version:

4.7.7

Self compiled

4.7.7-110-gcdba35530ba5

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

sfuchs79, 2018-03-11:
> - Use the same decimal separator for all fields (now. “sample.depth” uses “,” and “sample.temperature” uses “.” as separators)

That worries me and I can't reproduce it immediately. Which unit system did you use? Metric or imperial?

sfuchs79, 2018-03-11:
@p-neves 
Could you please test #1159 
Only quickly tested on my side. For the heartrate I have no test data at all.

I still can't explain the decimal separator issue and this still worries me...

PR #1159 (2018-03-19): Modify format of CSV dive list and dive profile export

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [x] New feature
- [x] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

In dive list CSV export:
- Move unit "min" from duration field to header
- Move unit "%" from O2 and He fields to header

In dive profile CSV export:
- Add "min" to sample time header
- Add column for heart rate

Add two missing format-number functions for mean depth in dive list
export and depth in dive profile export [... cut]
closed 2018-03-19 by dirkhh