subsurface/subsurface #1162: Display of heartrate in profile - bug in layout

sfuchs79, 2018-03-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

There are min. two serious bugs in the display of the heartrate in the profile:
- The y-axis tics are totaly wrong:
[image]
- When heatmap is on it's covering the heartrate graph:
[image]

### Operating system:

Windows 10

### Subsurface version:

Latest master 4.7.7-something

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@p-neves

p-neves, 2018-03-12:
I can confirm the same happens in Linux with latest master

jbygdell, 2018-03-14:
Fully known issue.. based on the fact that the scaling of the different components in the profile is not dynamic.
Displaying all components at the same time results in a very messy profile.
Building a code that can handle every single combination is rather tedious with a lot of if statements that need to work together. And since only 1 or 2 DC can actually deliver HR data the priority has been very low on this.

sfuchs79, 2018-03-14:
@jbygdell 
I agree with you that displaying everything at the same time may not be useful at all.
But nevertheless we could improve it maybe simple by switching heatmap off when switching heartrate on and vice versa.
And the tics are easy to fix. Maybe let's just change from step value of 10 to 25 or 50.

jbygdell, 2018-03-15:
A better way to fix the set size is to set the y-axis scale from lowest value-5 to max value+5, not 0 to max rounded up as it is now.

PR #1172 (2018-03-19): Profile heartrate: Nicer min/max/tics and only enable either heartrate or heatmap

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Correct a bug in finding the minimum heartrate.

Use the minimum and maximum heartrate value to set min/max and
tic distance for the heartrate axis in the profile.

Only allow to enable maximum one of both items tissue heatmap or
heartrate in profile.
This is done by always switching off the  [... cut]
closed 2018-03-19 by dirkhh