subsurface/subsurface #1163: MXE Windows debug build crashes on startup

sfuchs79, 2018-03-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

My MXE Windows debug build crashes immediately on startup...

### Operating system:

Windows 10

### Subsurface version:

current master 4.7.7-something

### Steps to reproduce:

### Current behavior:

Registers:
eax=00000000 ebx=06c307b0 ecx=00000000 edx=00000000 esi=4ef02698 edi=4ef02858
eip=77af041c esp=53e7fd8c ebp=53e7ff80 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
77AF041C 00000054 4EF02698 00000010  ntdll.dll!_NtWaitForWorkViaWorkerFactory@20
77AB3B16 06C307B0 77798630 8EC80152  ntdll.dll!TppWorkerThread
77798654 06C307B0 8EEC6799 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
77AE4A77 FFFFFFFF 77B09EB7 00000000  ntdll.dll!__RtlUserThreadStart
77AE4A47 77AB3880 06C307B0 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=00000102 ebx=00000006 ecx=00000000 edx=00000000 esi=00000000 edi=00000000
eip=757b6a4c esp=6432df04 ebp=6432df70 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
757B6A4C 00000000 6432E05C FFFFFFFF  win32u.dll!_NtUserMsgWaitForMultipleObjectsEx@20
75DFD3EE 00000000 6432E05C FFFFFFFF  USER32.dll!RealMsgWaitForMultipleObjectsEx
75DBAF1D 00000000 6432E05C FFFFFFFF  USER32.dll!_MsgWaitForMultipleObjectsEx@20
075530F7 0000001F 6432FE2C 073551B2  Qt5Cored.dll!processEvents  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/kernel/qeventdispatcher_win.cpp @ 649]
074F241A 00000000 5ABF1538 5ABF1538  Qt5Cored.dll!processEvents  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/kernel/qeventloop.cpp @ 134]
0735108A 073551B2 5ABF1538 5F7806A8  Qt5Cored.dll!exec  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/thread/qthread.cpp @ 515]
22C69B9D 00000000 77798654 5ABF1538  Qt5Qmld.dll!run  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtdeclarative-i686-w64-mingw32.shared/qtdeclarative-opensource-src-5.9.2/src/qml/qml/ftw/qqmlthread.cpp @ 147]
77798654 5ABF1538 B9396799 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
77AE4A77 FFFFFFFF 77B09EB7 00000000  ntdll.dll!__RtlUserThreadStart
77AE4A47 073551B2 5ABF1538 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=00000000 ebx=00002710 ecx=00000000 edx=00000000 esi=725376e0 edi=ffff58ab
eip=77aeed3c esp=6486feb8 ebp=6486ff80 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000206

AddrPC   Params
77AEED3C 00000003 6486FF38 00000001  ntdll.dll!_NtWaitForMultipleObjects@20
72523676 00000000 77798630 B9A90152  WINMM.DLL!_timeThread@4
77798654 00000000 B98D6799 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
77AE4A77 FFFFFFFF 77B09EB7 00000000  ntdll.dll!__RtlUserThreadStart
77AE4A47 72523620 00000000 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=00000001 ebx=54b5e718 ecx=00000000 edx=00000000 esi=00000000 edi=71c60000
eip=77aee7fc esp=6e38ff50 ebp=6e38ff80 iopl=0         nv up ei pl nz ac po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000216

AddrPC   Params
[... 290 lines cut ...]
77AE7A04 00000003 77E8F3B0 FFFFFFFF  ntdll.dll!RtlExitUserProcess
77793BD2 00000003 04A0EC74 75FD6165  KERNEL32.DLL!_ExitProcessImplementation@4
75FD5BEE 00000003 D80F5DA5 4A7809F8  msvcrt.dll!___crtExitProcess
75FD6164 00000003 00000001 00000000  msvcrt.dll!doexit
75FD5B71 00000003 00010001 00000000  msvcrt.dll!__exit
75FCB349 04A0F310 4A7809F8 7CA24FA8  msvcrt.dll!_abort
073409EC 076440D8 04A0F048 54747F18  Qt5Cored.dll!qt_message_fatal  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/global/qlogging.cpp @ 1690]
073414BE 7F0B5CF0 7F0B5CF0 04A0F108  Qt5Cored.dll!fatal  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/global/qlogging.cpp @ 796]
6D0D329E 04A0F1B4 77ABF550 7CA5D6A0  libgcc_s_sjlj-1.dll!_Unwind_SjLj_Register  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-gcc-i686-w64-mingw32.shared/gcc-5.4.0.build_/i686-w64-mingw32.shared/libgcc/./gthr-default.h @ 628]
338A554E 7CA5D6A0 07648500 7D637610  Qt5Widgetsd.dll!findTextureWidgetsRecursively  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/widgets/kernel/qwidgetbackingstore.cpp @ 965]
77ABF550 4A7D0379 4A7D0360 0000004B  ntdll.dll!_RtlFreeHeap@12
4A78061B 04A0F108 00000001 49F82050  qwindowsd.dll!paintDevice  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/plugins/platforms/windows/qwindowsbackingstore.cpp @ 75]
0F4A0ED1 0000001F 04A0F8EC 00000000  Qt5Guid.dll!beginPaint  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/gui/painting/qbackingstore.cpp @ 184]
6D0D32F4 04A0F1B4 77AC13EE 7F0B5CF0  libgcc_s_sjlj-1.dll! ??   [/home/stefan/Entwicklung/Subsurface/mxe/tmp-gcc-i686-w64-mingw32.shared/gcc-5.4.0.build_/i686-w64-mingw32.shared/libgcc/./gthr-default.h @ 628]
338A554E 7F0B5CF0 04A0F8EC 00000001  Qt5Widgetsd.dll!findTextureWidgetsRecursively  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/widgets/kernel/qwidgetbackingstore.cpp @ 965]
77AC13EE 7F0B5CF0 0759EEAC 0763E4DC  ntdll.dll!RtlAllocateHeap
75FB7250 00000048 00000000 000004C3  msvcrt.dll!_malloc

### Expected behavior:

### Additional information:

### Mentions:

@neolit123

neolit123, 2018-03-12:
1) when did it start happening? new compiler, new subsurface SHA, new Qt version - what changed?
2) is that the same MXE setup as the official windows binaries?

can you provide more information for the last thread:
```
5.9.2/src/widgets/kernel/qwidgetbackingstore.cpp @ 965]
77AC13EE 7F0B5CF0 0759EEAC 0763E4DC ntdll.dll!RtlAllocateHeap
75FB7250 00000048 00000000 000004C3 msvcrt.dll!_malloc

?? <---------
```
possibly can be done using a breakpoint in main (`b main`) and stepping until it crashes somewhere in the UI thread. possibly around widget / QML initialization. the idea would be to check the last call in the subsurface object code.

given there is a `qt_message_fatal`, it should output an actual reason for the exception in `cmd.exe`.
is there any message?

since this is a C++ exception and not a `SIGSEGV` then `findAllTextureWidgetsRecursively()` is not overflowing the stack. what causes the exception - no idea. 

looking at some of the `5.11` changes, do you have a 100% working GPU driver on that Windows box?
also what happens if you start `subsurface.exe` from `cmd.exe`, but first do:
`set QT_QUICK_BACKEND=software`

neolit123, 2018-03-12:
let's try this out too.

hi @alpqr,
any idea what could cause a exception in `findTextureWidgetsRecursively()`.

sfuchs79, 2018-03-13:
> when did it start happening? new compiler, new subsurface SHA, new Qt version - what changed?

Very difficult to say because I didn't build the debug build for a long time.

> is that the same MXE setup as the official windows binaries?

I'm still on Qt 5.9.2

neolit123, 2018-03-13:
clean your build completely, if you haven't already tried.
if it still doesn't work match the MXE version of the official build.

could be a compiler issue.

sfuchs79, 2018-03-13:
When switching from release to debug our MXE build script does clean the build directory completely and does rebuild everything. I know this because I implemented it ;-)
@dirkhh is currently using which Qt and which gcc from MXE?
Rebuilding MXE - urgh... Still have no new machine but I'm close to order s.th.
(off topic: which AM4 micro atx board to choose? non-gaming, non-overclocking, office-use)

BTW: I up to now didn't start from cmd.exe to look for a possible error message - for sure still have to do this simple thing.

neolit123, 2018-03-13:
@sfuchs79 

a little search here:
https://api.travis-ci.org/v3/job/352730137/log.txt
and here:
https://github.com/mxe/mxe/blob/994ad47302e8811614b7eb49fc05234942b95b89/docs/build-matrix.html#L3332

tells that we are using Qt 5.9.2 with gcc 5.4.0 from MXE for the Travis MXE build (MXE head is at 994ad473).
so these versions match yours.

yes the `cmd.exe` output is needed because `qt_message_fatal` might tell something in the lines of "hey the software crashed because of reason XX". also don't forget about `set QT_QUICK_BACKEND=software` and check your GPU driver.

sfuchs79, 2018-03-14:
What I see on the cmd line is the following after the list of supported DCs, multiple identical lines:
`ASSERT: "!m_image.isNull()" in file qwindowsbackingstore.cpp, line 75`

The crash doesn't always show the exact same signature but it always happens at startup. This time I have this backtrace:
Registers:
eax=00000000 ebx=00000001 ecx=00000000 edx=00000000 esi=00000002 edi=00000002
eip=775ced3c esp=4b58fc68 ebp=4b58fdf8 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
775CED3C 00000002 4B58FE30 00000001  ntdll.dll!_NtWaitForMultipleObjects@20
750912A3 00000002 4B58FE30 00000000  KERNELBASE.dll!WaitForMultipleObjectsEx
7556EFDD 06B635D8 00007530 00007530  combase.dll!WaitCoalesced  [onecore\com\published\comutils\coalescedwait.cxx @ 70]
7556EDFD 7559EDF0 7559EDF0 4B58FF94  combase.dll!CRpcThread::WorkerLoop  [onecore\com\combase\dcomrem\threads.cxx @ 317]
7559EE0F 06B635D8 74688630 E41A1246  combase.dll!CRpcThreadCache::RpcWorkerThreadEntry  [onecore\com\combase\dcomrem\threads.cxx @ 72]
74688654 06B635D8 E7032D8C 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
775C4A77 FFFFFFFF 775E9EB8 00000000  ntdll.dll!__RtlUserThreadStart
775C4A47 7559EDF0 06B635D8 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=00000000 ebx=06b35178 ecx=00000000 edx=00000000 esi=06b74778 edi=06b74938
eip=775d041c esp=4b34fd8c ebp=4b34ff80 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
775D041C 0000030C 06B74778 00000010  ntdll.dll!_NtWaitForWorkViaWorkerFactory@20
77593B16 06B35178 74688630 E4761246  ntdll.dll!TppWorkerThread
74688654 06B35178 E76F2D8C 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
775C4A77 FFFFFFFF 775E9EB8 00000000  ntdll.dll!__RtlUserThreadStart
775C4A47 77593880 06B35178 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=00000000 ebx=4b6cf040 ecx=00000000 edx=00000000 esi=00000000 edi=00000454
eip=775ce7ac esp=4bbefe8c ebp=4bbefefc iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
775CE7AC 00000454 00000000 4BBEFED0  ntdll.dll!_NtWaitForSingleObject@12
7507EBF9 00000454 00000064 00000000  KERNELBASE.dll!WaitForSingleObjectEx
7507EB52 00000454 00000064 4BBEFF60  KERNELBASE.dll!_WaitForSingleObject@8
50EB0E7F 00000454 00000064 50EB1A70  nvoglv32.DLL!DrvValidateVersion
50D4D7CF 4B6CF040 50EB1A70 4BBEFF80  nvoglv32.DLL!0x9ad7cf
50D4D4A4 4B891FC0 06BB8680 4BBEFF94  nvoglv32.DLL!0x9ad4a4
50EB1A87 4B891FC0 74688630 E4FC1246  nvoglv32.DLL!DrvValidateVersion
74688654 4B891FC0 E7E52D8C 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
775C4A77 FFFFFFFF 775E9EB8 00000000  ntdll.dll!__RtlUserThreadStart
775C4A47 50EB1A70 4B891FC0 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=00000000 ebx=06b35178 ecx=00000000 edx=00000000 esi=06b6f478 edi=06b6f638
eip=775d041c esp=4b14fd8c ebp=4b14ff80 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
775D041C 0000030C 06B6F478 00000010  ntdll.dll!_NtWaitForWorkViaWorkerFactory@20
77593B16 06B35178 74688630 E4561246  ntdll.dll!TppWorkerThread
74688654 06B35178 E74F2D8C 00000000  KERNEL32.DLL!@BaseThreadInitThunk@12
775C4A77 FFFFFFFF 775E9EB8 00000000  ntdll.dll!__RtlUserThreadStart
775C4A47 77593880 06B35178 00000000  ntdll.dll!__RtlUserThreadStart@8

Registers:
eax=c0000225 ebx=00000000 ecx=00000000 edx=00000000 esi=00000003 edi=00000003
eip=775ced3c esp=4c26fd54 ebp=4c26fee4 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
775CED3C 00000003 06B99334 00000001  ntdll.dll!_NtWaitForMultipleObjects@20
750912A3 00000003 06B99334 00000000  KERNELBASE.dll!WaitForMultipleObjectsEx
75091188 00000003 06B99334 00000000  KERNELBASE.dll!_WaitForMultipleObjects@16
50EDBB45 4C26FF80 511AFE83 06B99318  nvoglv32.DLL!DrvPresentBuffers
[... 163 lines cut ...]

Registers:
eax=00000000 ebx=00000000 ecx=00000000 edx=00000000 esi=00000003 edi=776779e0
eip=775cea4c esp=049aeb34 ebp=049aec0c iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202

AddrPC   Params
775CEA4C FFFFFFFF 00000003 497809F8  ntdll.dll!_NtTerminateProcess@8
775C7A04 00000003 77E8F3B0 FFFFFFFF  ntdll.dll!RtlExitUserProcess
74683BD2 00000003 049AEC74 75296165  KERNEL32.DLL!_ExitProcessImplementation@4
75295BEE 00000003 AA866270 497809F8  msvcrt.dll!___crtExitProcess
75296164 00000003 00000001 00000000  msvcrt.dll!doexit
75295B71 00000003 00010001 00000000  msvcrt.dll!__exit
7528B349 049AF310 497809F8 7C489B60  msvcrt.dll!_abort
06C409EC 06F440D8 049AF048 4F6B4418  Qt5Cored.dll!qt_message_fatal  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/global/qlogging.cpp @ 1690]
06C414BE 7CFEF8F8 7CFEF8F8 049AF108  Qt5Cored.dll!fatal  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/corelib/global/qlogging.cpp @ 796]
6D0D329E 049AF1B4 7759F550 7C50F820  libgcc_s_sjlj-1.dll!_Unwind_SjLj_Register  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-gcc-i686-w64-mingw32.shared/gcc-5.4.0.build_/i686-w64-mingw32.shared/libgcc/./gthr-default.h @ 628]
[... cut ...]
6D0D32F4 049AF1B4 775A13EE 7CFEF8F8  libgcc_s_sjlj-1.dll! ??   [/home/stefan/Entwicklung/Subsurface/mxe/tmp-gcc-i686-w64-mingw32.shared/gcc-5.4.0.build_/i686-w64-mingw32.shared/libgcc/./gthr-default.h @ 628]
2FE3554E 7CFEF8F8 049AF8EC 00000001  Qt5Widgetsd.dll!findTextureWidgetsRecursively  [/home/stefan/Entwicklung/Subsurface/mxe/tmp-qtbase-i686-w64-mingw32.shared/qtbase-opensource-src-5.9.2/src/widgets/kernel/qwidgetbackingstore.cpp @ 965]
775A13EE 7CFEF8F8 06E9EEAC 06F3E4DC  ntdll.dll!RtlAllocateHeap
75277250 00000048 00000000 000004C3  msvcrt.dll!_malloc

sfuchs79, 2018-03-14:
Doing the `set QT_QUICK_BACKEND=software` doesn't seem to change anything.
My GPU driver is a NVidia driver comming with Windows 10:
[image]

neolit123, 2018-03-14:
> What I see on the cmd line is the following after the list of supported DCs, multiple identical lines:
ASSERT: "!m_image.isNull()" in file qwindowsbackingstore.cpp, line 75

seems like a bug in Qt. `findTextureWidgetsRecursively()` is not happy about some `NULL` `QImage`.
i don't think we can do much about this except debugging it in Qt.

i suggest to move away from this Qt version.

does the same happen for you with the official binaries? they seem to be based on the same MXE that you are using.

bstoeger, 2018-03-15:
> seems like a bug in Qt. findTextureWidgetsRecursively() is not happy about some NULL QImage.
i don't think we can do much about this except debugging it in Qt.

Could also be memory corruption. Perhaps try running it under `valgrind` or equivalent tool. Note that `m_image` is not a `QImage` but a `QWindowsNativeImage`, whatever the purpose of that is.

bstoeger, 2018-03-15:
Another thing to try: temporarily remove your divelog. This should tell us if it's something special about your divelog or a very general problem. Perhaps it's the video you added? Currently this gives
```
QImage::scaled: Image is a null image
```
which should be easily prevented.

neolit123, 2018-03-17:
this stack trace suggests something in the Windows event loop that exhaust the available memory.
have your tried your build on another Windows box?

sfuchs79, 2018-03-23:
I finally updated my hardware. The crash is gone. Maybe this was really an issue of insufficient memory for running the debug build.

@bstoeger 
I could try the video playback feature now again and try to reproduce the bug there but it would be nice if you could rebase your PRs (two of them I think) on current master.

closed 2018-03-23 by sfuchs79