subsurface/subsurface #1171: Unable to import from Suunto Eon Steel on Subsurface-4.7.7-x86_64.AppImage

aaddame, 2018-03-17:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Trying to import from Suunto Eon Steel on Subsurface-4.7.7-x86_64.AppImage I get the following error:
SubsurfacStarting download from  /dev/ttyS3
Starting the thread 0
[0.007894] ERROR: Failed to claim the usb interface (LIBUSB_ERROR_BUSY). [in ../../src/usbhid.c:430 (dc_usbhid_open)]
[0.008094] ERROR: unable to open device [in ../../src/suunto_eonsteel.c:776 (suunto_eonsteel_device_open)]
INFO: dc_deveice_open error value of -5
Finishing the thread Unable to open %s %s (%s) dives downloaded 0
Starting download from  /dev/ttyS3
Starting the thread 0
[0.005352] ERROR: Failed to claim the usb interface (LIBUSB_ERROR_BUSY). [in ../../src/usbhid.c:430 (dc_usbhid_open)]
[0.005486] ERROR: unable to open device [in ../../src/suunto_eonsteel.c:776 (suunto_eonsteel_device_open)]
INFO: dc_deveice_open error value of -5
Finishing the thread Unable to open %s %s (%s) dives downloaded 0
Starting download from  /dev/ttyS3
Starting the thread 0
[0.005307] ERROR: Failed to claim the usb interface (LIBUSB_ERROR_BUSY). [in ../../src/usbhid.c:430 (dc_usbhid_open)]
[0.005457] ERROR: unable to open device [in ../../src/suunto_eonsteel.c:776 (suunto_eonsteel_device_open)]
INFO: dc_deveice_open error value of -5
Finishing the thread Unable to open %s %s (%s) dives downloaded 0
Starting download from  /dev/ttyS3
Starting the thread 0
[0.006660] ERROR: Failed to claim the usb interface (LIBUSB_ERROR_BUSY). [in ../../src/usbhid.c:430 (dc_usbhid_open)]
[0.006754] ERROR: unable to open device [in ../../src/suunto_eonsteel.c:776 (suunto_eonsteel_device_open)]
INFO: dc_deveice_open error value of -5
Finishing the thread Unable to open %s %s (%s) dives downloaded 0

e-4.7.7-x86_64.AppImage gives following error trying to import from Suunto Eon Steel:

### Operating system:

NAME="Arch Linux"

Suunto Eon Steel

### Subsurface version:

Subsurface-4.7.7-x86_64.AppImage

does not happen on earlier versions

### Steps to reproduce:

Ctrl-d
### Current behavior:

Error on import from Suunto Eon Steel
### Expected behavior:

complete the import
### Additional information:

### Mentions:

aaddame, 2018-03-17:
> earlier versions show the following:
Starting download from  /dev/ttyS3
Starting the thread 0
[0.015548] ERROR: Usb read interrupt transfer failed (LIBUSB_ERROR_TIMEOUT). [in usbhid.c:525 (dc_usbhid_read)]
INFO: dc_deveice_open error value of 0
Finishing the thread  dives downloaded 1

aaddame, 2018-03-17:
added log from good import

aaddame, 2018-04-10:
still happnes on 4.7.8

glance-, 2019-11-10:
Somethings odd with your system, that hogs the hid device and keeps subsurface from talking to the device. Does this still happen or have it fixed itself?