subsurface/subsurface #1173: UI locks with both version check window and backup printing templates notification

Balandat, 2018-03-18:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
After installing 4.7.7., I got two dialog windows:
- "The following backup printing templates were created: ...."
- "Subsurface is checking every to weeks if a new version is available..."

Neither of them can be interacted with, the GUI completely locks. Only way out was to force-quit the program. After re-opening Subsurface, I was able to accept the backup printing dialog with "OK" before anything else showed. 

### Operating system:
Mac OS High Sierra 10.13.3 (17D102)

### Subsurface version:
4.7.7

### Steps to reproduce:
Open Subsurface after upgrade.

### Current behavior:
UI freezes, blue-shaded "OK" button in backup printing template notification isn't operational, version check question buttons "Decline" and "Accept" cannot be clicked.

### Expected behavior:
Can click at least one of the two dialogues.

### Additional information:

### Mentions:

neolit123, 2018-03-18:
hello, i've tried the latest master build of Subsurface and i cannot confirm the issue on Windows and Ubuntu.
this could be a window modality issue with our UI framework (Qt) on OSX when the two dialog box appear at the same time.

needs to be reproduced on OSX.

mikeller, 2026-10-17:
Hi @Balandat,

This is the same recurring startup-dialog-interaction class as a related report (#1499), concurrent modal dialogs (version check + backup template notification) interfering with each other at launch. It's a timing-dependent issue we've never been able to reliably pin down, and this report is now 8 years old with no further activity. We're closing this as stale.

Please reopen with a fresh repro (ideally a verbose log) if you're still seeing the UI lock with these dialogs on a current build.

closed 2026-10-17 by mikeller