subsurface/subsurface #1178: Getting "Unable to lookup Revision" when downloading from cloud storage on Windows 10

jivogt, 2018-03-24:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

jivogt, 2018-03-24:
"Unable to Lookup Revision" when attempting to download from cloud storage with Windows 10. Subsurface version 4.7.7.  I have uninstalled and reinstalled the application.

This is a new issue.  Cloud storage is working well from Android handset with version 2.0.3.

neolit123, 2018-03-26:
hello,
what happens if you go to this folder:
`C:\Users\<your user name>\AppData\Roaming\Subsurface`

and temporary rename the folder `cloudstorage` to something like `cloudstorage_` and then attempt to download again from the cloud (you can restore the folder anytime you want)?

we are using Git for the cloud storage, but i'm unable to tell what this error means exactly.

jivogt, 2018-04-04:
That worked!!!

Thanks,