subsurface/subsurface #1179: Dive picture thumbnails not loaded for dives with many pictures

sfuchs79, 2018-03-25:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

For dives with many dive pictures the picture thumbnails are not loaded.
@bstoeger 
My guess is that this may come from some changes you implemented recently but not necessarily the very latest updates but s.th. which may be a few weeks ago.
I may have revealed the issue on my side now because I upgraded my HW and started from a fresh Windows 10 and therefore also from a fresh Subsurface installation.

[image]

I see these two errors multiple times on stdout. First a long block of multiple lines of the first error, then a long block of multiple occurrences of the second error:

QImage::scaled: Image is a null image

Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

### Operating system:

Windows 10

### Subsurface version:

Latest master 4.7.7-something

Build with MXE against Qt 5.9.2 or Qt 5.10.1 (I checked both because I wanted to check if it's an issue with latest Qt)

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@bstoeger

sfuchs79, 2018-03-25:
@bstoeger 
I may have time to go thru the latest changes and/or bisect this later...

bstoeger, 2018-03-25:
> QImage::scaled: Image is a null image

This is expected and should be fixed in the video branch. It just means that currently there is no thumbnail.

> Qt has caught an exception thrown from an event handler.

Hmmm... My best bet would be resource exhaustion. I cannot reproduce this with a dive with ca 50 images. But if I add multiple videos at the same time, the same happens for me. It seems to generate hundreds of threads until the system just gives up and terminates the application.

> I may have time to go thru the latest changes and/or bisect this later...

That would be very helpful.

bstoeger, 2018-03-25:
I added a number of debug-messages in #1880 -> perhaps this gives us an idea of what's going wrong. But I think you must compile in debug-mode to get the messages.

sfuchs79, 2018-03-26:
@bstoeger 
Sorry, I have very limited time at the moment so I'm moving forward with debugging in very small steps. But I may have s.th. interesting now.

**First thing - test build configuration:**
Current master plus:
[...] some private stuff unrelated to the issue
pick c3584f8 Debug: sprinkle debug messages in thumbnailing code
What I want to say: No video stuff in the code...

**MXE/Qt:** 5.10.1
**Release build.**

My pictures are always on an SMB share - some pretty old LG NAS.

Results:
- For dives with small number of pictures: OK
- For dives with many pictures: Some thumbnails missing
- Interesting output on stdout, based on your debug code.
The ones with the error are exactly the ones with the mising thumnail.
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(5).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(8).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(7).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(6).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(11).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(9).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(10).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(14).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(12).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(13).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(15).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(17).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(16).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(18).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(19).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(20).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(21).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(16).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(22).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(18).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(23).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(19).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(24).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(21).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(25).jpg"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(17).JPG"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(26).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(5).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(27).jpg"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(10).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(28).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(15).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(29).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(6).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(30).jpg"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(11).jpg"
No thumbnail in cache for  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG"
Loaded picture  "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(7).jpg"
[... cut ...]
Write image to  "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/87c69528b32680c3569c2d69a59a2cb55e5b5788"
Write image to  "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/99b6c76021431b0dc3d46fb1addc28748739ff19"
Write image to  "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/cb7db8a0acfdd938e1b9d68b749b46b0e2088288"
Write image to  "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/0e8fdfb0d03f3916d5e7c768b7a009b07b149383"
Write image to  "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/783702b488ace135303ccd97fc356330eb7a8fce"

bstoeger, 2018-03-27:
Sorry, I don't get it. I see no correlation between filename and success/fail.

I see only one eyebrow-raiser:
```
Insufficient memory (case 4)
```
Something goes seriously wrong here. Can others reproduce this?

sfuchs79, 2018-03-28:
@bstoeger 
Yes, you're right. I also don't see a correlation between file name and issue.

We may have to think about different things: Number of possible open file handles exceeded, things like that,...??!

bstoeger, 2018-03-28:
Do you compile Qt from source? The only idea I have is replacing `Image::load(const QString &, const char *)` by something like:
```
bool QImage::load(const QString &fileName, const char* format)
{
    QImageReader reader(fileName, format);
    *this = reader.read();
    if (isNull())
        qInfo() << "Error loading image:" << (int)reader.error() << reader.errorString();
    return !isNull();
}
```
(untested, but you get the idea).

Well, another idea: Under Windows is there something like `strace`?

bstoeger, 2018-03-30:
I added an additional commit to #1180, which should report better errors if image downloading failed.

sfuchs79, 2018-04-01:
Very quick update based on current master (after @neolit123 merged two of @bstoeger PRs):

Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(6).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(22).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(7).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(23).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(16).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(24).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(19).JPG"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(25).jpg"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(17).JPG"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(26).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(10).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(27).jpg"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(9).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(28).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(15).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(29).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(5).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(30).jpg"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(8).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(11).jpg"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(32).JPG"
Error loading image "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG" 4 "Die Bilddaten konnten nicht gelesen werden"
Failed loading picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG"
"Translated filename: //nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG -> //nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(20).JPG"
Error loading image "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG" 4 "Die Bilddaten konnten nicht gelesen werden"
Failed loading picture from translated filename "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG"
QImage::scaled: Image is a null image
QImage::scaled: Image is a null image
QImage::scaled: Image is a null image
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(33).JPG"
Error loading image "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(32).JPG" 4 "Die Bilddaten konnten nicht gelesen werden"
Failed loading picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(32).JPG"
"Translated filename: //nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(32).JPG -> //nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(32).JPG"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(34).JPG"
Loaded picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(21).JPG"
No thumbnail in cache for "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(35).JPG"
Error loading image "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(34).JPG" 4 "Die Bilddaten konnten nicht gelesen werden"
Failed loading picture "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(34).JPG"
"Translated filename: //nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(34).JPG -> //nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(34).JPG"
[... cut ...]
Write image to "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/87c69528b32680c3569c2d69a59a2cb55e5b5788"
Write image to "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/c69fb05f08baf3410a7f7d516945cc23a775f499"
Write image to "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/274f6b21e7ec1d890d1523e5c87b6eb1ea8869df"
Write image to "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/0d24ec10791a90f2af8a0b2333486b26f234154f"
Write image to "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/0076e5bc2fde0ca13b6b66c4c78cb1c14bde9cb5"

sfuchs79, 2018-04-01:
@bstoeger 
This for me more and more looks like some kind of resource conflict. Would it be possible to limit the number of concurrent threads which are processing pictures to a specific number? I would like to try if this solves the issues.

sfuchs79, 2018-04-01:
Hmmm... the number of possible threads should be limited to a reasonable number:

The default maxThreadCount is QThread::idealThreadCount().

int QThread::idealThreadCount()

Returns the ideal number of threads that can be run on the system. This is done querying the number of processor cores, both real and logical, in the system. This function returns 1 if the number of processor cores could not be detected.

Stupid question:
Could it be the case that your code fails because the number of possible threads is exceeded? Does your code expect an infinite number of possible threads?

neolit123, 2018-04-01:
`pastebin.com` and text attachments are preferred over posting huge logs in github issue comments.
having a brief look at the issue, i think the code for thumbnails should do use a thread queue instead of spawning hundreds of separate threads.

bstoeger, 2018-04-01:
> i think the code for thumbnails should do use a thread queue instead of spawning hundreds of separate threads.

That's exactly what it does (`QtConcurrent::run`). I'm not aware that I changed the logic in master. Might be wrong though.

> Could it be the case that your code fails because the number of possible threads is exceeded? Does your code expect an infinite number of possible threads?

It shouldn't. But I serialized the whole thing in #1151 - that's why it is slower, so maybe try that.

But also note the actual error message:
```
Error loading image "//nas/Filebase/Bilder/Tauchen-UWR/Bodensee Überlingen 2017-01/Bodensee_Überlingen_2017-01_(31).JPG" 4 "Die Bilddaten konnten nicht gelesen werden"
```
That is `QImageReader::InvalidDataError` with the description:
> The image data was invalid, and QImageReader was unable to read an image from it. The can happen if the image file is damaged.

So it can read the file, but not parse it (perhaps a failure to run ImageMagick?). In combination with the out-of-memory errors you get, that's all very odd..

sfuchs79, 2018-04-02:
>>Could it be the case that your code fails because the number of possible threads is exceeded? Does your code expect an infinite number of possible threads?

>It shouldn't. But I serialized the whole thing in #1151 - that's why it is slower, so maybe try that.

Honestly I'm not a big fan of adding more and more changes now on top of each other and hope that at some point it gets better.
I think we need to go the other way and revert changes or bisect it to find the exact change which causes the issues. I'll try to do this during the next days.
Could you please help me to summarize the changes in the dive pictures code from the last few weeks?

And I still think we need minimum a second tester who can try to reproduce this under Windows. Currently I really don't believe that there is s.th. broken around my HW and/or my Windows installation.

sfuchs79, 2018-04-02:
Ok, I have first results from bisecting and I'm a little bit surprised:
The issue seems to go back to a change between V4.7.5 (still good) and v4.7.6 (already missing thumbnails).
Honestly also my fault that I didn't recognize that the issue is there for such a long time. I always focused on more recent changes. I will continue to bisect...

neolit123, 2018-04-02:
@sfuchs79 
i don't see any errors for a dive with around 200 images.

bstoeger, 2018-04-02:
I took the travis build of #799 (which is rebased on current master), generated a dummy dive and added 360 images. It took an annoyingly long time to get to the time-dialog (we'll have to fix that), but once I clicked on OK, importing was instantaneous.

Note that the images are outside of the dive-time, so they weren't automatically added to the profile - maybe that's the problem.

> i don't see any errors for a dive with around 200 images.

What if you (temporarilly) remove the thumbnail cache?

bstoeger, 2018-04-02:
> Ok, I have first results from bisecting and I'm a little bit surprised

Thanks. Just in case, did you also try an official build so that we can 100% exclude that something is funny with your build system?

neolit123, 2018-04-02:
> What if you (temporarilly) remove the thumbnail cache?

i do get some `No thumbnail in cache for...`
but that doesn't seem to affect the image loading.

sfuchs79, 2018-04-02:
@bstoeger 
Yes, I tested the official V4.7.7 build from Dirk to exclude also anything coming from a specific Qt version. Result was that also with V4.7.7 I'm missing some thumbnails.

Just to make one thing clear again - my current way of testing is: I have my dive log with various dives with small and large number of pictures. I delete the hash file before starting Subsurface. I start Subsurface with the mentioned dive log. I click the various dives and let Subsurface (re-)create the thumbnails.

**Now I have again a new result:
I was able to reproduce the issue with a V4.7.5 build as well.**

Ok, I'm close to giving up now - this is way too frustrating.

Maybe a last summary of what I think now:
- It is not @bstoeger code - or did you change s.th. before v4.7.5? I guess no.
- I don't think it's my build system - v4.7.7 from Dirk shows the issue as well.
- I don't think it's my HW or Windows.

What is left? Maybe really an issue in QImageReader?

Maybe as a last resort: Could you create a test code which does s.th. like 1-5 retries exactly at the point where I receive the error? I would like to learn if this changes s.th.

dirkhh, 2018-04-02:
I question the value of 200 or 300 pictures on a single dive, but that's just from a use case perspective, from a code perspective this should work for any number of pictures.

We can tell if there is no thumbnail available - what happens if we then try to load that thumbnail again? To me this looks like what we really need to figure out is why do some of these loads fail. Is it out of memory (huh?)? Is it a race condition? Is it a Qt bug?

We should be able to use an thread safe counter to count up the for every successfully loaded thumbnail, and then re-display once that matches the number of pictures that _should_ be on that dive.

Or am I missing the point of this investigation?

bstoeger, 2018-04-02:
> We can tell if there is no thumbnail available - what happens if we then try to load that thumbnail again?

That should be easy to test - switch to a different dive and then back.

> To me this looks like what we really need to figure out is why do some of these loads fail. Is it out of memory (huh?)? Is it a race condition? Is it a Qt bug?

The only thing we know for sure is that we get a `QImageReader::InvalidDataError` error out of the depths of Qt. So it's not directly IO related.

My first guess is some sort of memory or thread leak. Perhaps it needs a huge divelog such as Stefan's? @sfuchs79: Could you try to monitor the memory-consumption of the subsurface process and the numbers of threads started? Do you get this for the first dive or only after you selected a few other dives? Is it always the same pictures that fail?

My second guess would be memory corruption, but then I should be able to hit this.

sfuchs79, 2018-04-02:
>> We can tell if there is no thumbnail available - what happens if we then try to load that thumbnail again?

> That should be easy to test - switch to a different dive and then back.

That is a good point. In fact if I switch between different dives with a large number of pictures, after some iterations of switching dives I will have all thumbnails for all those dives loaded and displayed correctly.

dirkhh, 2018-04-02:
So that sounds like indeed having
a) a retry approach to the `QImageReader::InvalidDataError`
and
b) a counter and then doing a refresh of the profile pictures when all the thumbnails are loaded
would be a valid fix.

bstoeger, 2018-04-02:
> That is a good point. In fact if I switch between different dives with a large number of pictures, after some iterations of switching dives I will have all thumbnails for all those dives loaded and displayed correctly.

So if it is resource exhaustion, it's only temporary. This is becoming more and more obscure.

> So that sounds like indeed having
a) a retry approach to the QImageReader::InvalidDataError
and
b) a counter and then doing a refresh of the profile pictures when all the thumbnails are loaded
would be a valid fix.

Ugh. First I'd really like to understand the root-cause. What about a huge file that is really corrupt or not a supported format? It seems like a bad idea to attempt parsing it numerous times.

Maybe attempt removing all the `QtConcurrent` magic?

neolit123, 2018-04-02:
> Maybe attempt removing all the QtConcurrent magic?

+1. a single, dedicated `QThread` that does both the downloading and processing of an image queue seems like a very safe solution to me. the ease of use of things like `QtConcurrent::run()` can lead up to complications which are quite difficult to debug.

bstoeger, 2018-04-03:
@sfuchs79: Even if it's grasping at straws, it would be really interesting to see how #1151 behaves. It doesn't completely remove `QtConcurrent::run()` but is less aggressive about it. The code changed enough that it is worth a try. If the code shows exactly the same behavior, we might have a fundamental problem with multi-threading. Then we could try to construct a failing self-contained example.

Just don't try adding any videos - this gives some very strange behavior.

sfuchs79, 2018-04-03:
@bstoeger 
Very quick update with #1151 :
Makes it even worse but maybe helps to better understand the issue.
With a dive with a really large number of pictures I now even can make Subsurface freeze.

The last lines on stdout are many lines of:
`QImage: out of memory, returning null image`
and then finally a:
`terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc`

The Windows task manager says:
[image]

dirkhh, 2018-04-03:
OK, I'm more and more convinced that the concurrency here causes us to create a ton of temporary memory consumption when loading the thumbnails - which pushes us into OOM and would explain the behavior. So let's have a single thread that loads them one at a time and populates the profile as it gets them loaded.

bstoeger, 2018-04-03:
Yes, at least I can reproduce it in the way that my subsurface process grows to ~3 GB. It doesn't hurt my laptop at all, but that shouldn't happen. That raises the question why the image data is not freed? Something about different thread contexts? Unfortunately, I'll be too busy this week to find out.

sfuchs79, 2018-04-03:
Hey, don't blame my HW! ;-)
My guess would be that it's less a question of HW and available memory (I have 16GB here) but more a question of OS. Linux may be able to deal with this situation - Windows isn't. Or did you also test under Windows?
For me it's already a huge step forward that you are able to reproduce something. I'm now more optimistic again that this can be fixed.

bstoeger, 2018-04-03:
For fun I removed *all* `QtConcurrent::*` from #1151. And sure enough it still grows to ~3 GB, though sometimes goes back down to ~2 GB:
```
top - 22:43:59 up 5 days, 15:24,  9 users,  load average: 1.09, 1.14, 1.25
Tasks: 231 total,   1 running, 230 sleeping,   0 stopped,   0 zombie
%Cpu(s): 21.1 us,  5.2 sy,  0.0 ni, 72.9 id,  0.5 wa,  0.0 hi,  0.3 si,  0.0 st
KiB Mem :  3930660 total,   113328 free,  3464808 used,   352524 buff/cache
KiB Swap:  4082684 total,  1345884 free,  2736800 used.    87688 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                  
26783 bs        20   0 4167408 2.617g  44992 S  25.6 69.8   0:42.06 subsurface                                                                                           
```
It seems that at least #1151 is fundamentally broken.

bstoeger, 2018-04-03:
In #1151 I can drastically reduce memory consumption by commenting out the line
```
               //emit dataChanged(createIndex(i, 0), createIndex(i, 1));
```
in `qt-models/divepicturemodel.cpp`. Unforunately, this means that dive-profile pictures are not updated.

Then I can reduce memory consumption by another factor of two by removing all `QtConcurrent:*` constructs and not returning `QImage`s by value.

I don't get it _at all_.

bstoeger, 2018-04-05:
Ok, at least I understand the terrible behavior of #1151. Unfortunately it is not directly related to this bug.

The course of events is the following:
1) We schedule N pictures for thumbnailing.
2) For each thumbnailed picture a `dataChanged` signal is emitted to tell the dive-picture-widget to update the picture.
3) `dataChanged` is connected to `ProfileWidget2::plotPictures()`.
4) `ProfileWidget2::plotPictures()` repaints ALL pictures. Thus, we repaint not N, but N^2 pictures.
5) See #1189: the old pictures are not deleted right away, but with `deleteLater()`, meaning that the memory is kept until the main loop is not busy anymore. Thus, memory for N^2 thumbnails is allocated. This is a potential DOS attack.

So before continuing with #1151, 3) will have to be fixed: only repaint pictures that were actually changed! Unfortunately, this doesn't seem to be related to the bug described here. :(

dirkhh, 2018-04-09:
I'm trying to get a better handle on how to reproduce the problem.
So this is one dive with hundreds of pictures, I assume it's a really long dive?
And then run this on a machine with limited resources.
So creating a memory constrained VM might make it easier to reproduce?
Am I missing any other ingredients?

sfuchs79, 2018-04-09:
@bstoeger @dirkhh 
Sorry, only a quick update again:
- Issue still clearly there - I have the feeling nothing changed here with latest fixes
Again this error message multiple times:
Error loading image "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/b620c7e03729a4260f41f25c31c7b8f0db60cf13" 4 "Die Bilddaten konnten nicht gelesen werden"
QImage::scaled: Image is a null image
- Regarding the setup and sequence:
Yes for "Delete hashes file".
Dives: 1st dive ~10 pictures - ok. 2nd dive ~10 pictures ok. 3rd dive ~80 pictures errors start.

I have one idea which could help: I learned that the value of "maxZoom" (that's a const @bstoeger introduced for the dive picture zoom feature) has an effect on the severity of the issue. I have to admit that I have a "private" patch which changes "maxZoom" to "6". With "6" the issue is worse compared to the standard value of "3". And this also works in the other direction. If I set "maxZoom" to "1" the number of picture where I fail to load a thumbnail decreases (with same setup for the rest).

bstoeger, 2018-04-09:
> "C:/Users/Stefan/AppData/Local/Subsurface/Subsurface/cache/b620c7e03729a4260f41f25c31c7b8f0db60cf13" 4 "Die Bilddaten konnten nicht gelesen werden"

As stated above, this is `QImageReader::InvalidDataError` of `QImageReader` and it comes out of the depths of Qt.
@thiagomacieira: Is there a straight-forward way of getting more precise diagnostics?

> I have one idea which could help: I learned that the value of "maxZoom" (that's a const @bstoeger introduced for the dive picture zoom feature) has an effect on the severity of the issue. I have to [...]

Now this really does sound like a resource problem. I'm still wondering why it's only a temporary problem. You have to realize that there is a quadric relationship between size and memory consumption (i.e. one `maxZoom=6` thumbnail is 36 `maxZoom=1` tumbnails)!

So my conclusion would be to reduce the memory footprint.
Currently, for the selected dive(s), we have at least three thumbnails in-memory:
1) in `thumbnailCache`, with size `maxZoom`.
2) in `divePictureModel`, with size current `zoomLevel`, for divePicture tab.
3) in `divePictureModel`, with size default `zoomLevel`, for profile widget.

And who-knows how many copies/caches created by conversions `QImage`<->`QPixmap`, zooming, leaks, etc. From what I've seen, Qt does its best to copy image data only when needed, but who knows.

When I find time, I will attempt to get rid of 3), because the images in the Profile widget are scaled anyway. Perhaps also get rid of 2) at the cost of CPU time, when scrolling through the divePicture tab.

Number 1) will need refactoring. Do we really want to read hundreds (thousands?) of thumbnails into memory on startup? Perhaps this should be saved to disk on a per-dive basis and loaded on demand? If yes, how should we proceed with pictures that are added to different dives? Generate independent thumbnails?

bstoeger, 2018-05-01:
@sfuchs79: Could you perhaps test #1206? It should use much less memory and serializes all thumbnailing.

If this error still happens, I'm more and more inclined to think that this is a Qt-problem.

dirkhh, 2018-05-14:
According to @bstoeger this is addressed now

closed 2018-05-14 by dirkhh